lvgl/lvgl #5722: Search objects by id

MiSimon, 2024-02-22:
### Introduce the problem

Currently there is no way to search for an object by some sort of user defined id (like "_btn_exit_").
There should be a way to search objects by a user defined id, like the js function `document.getElementById`.

This has been discussed in a [forum thread](https://forum.lvgl.io/t/search-objects-by-name/14677/1).

### Proposal

At least 3 new public methods are required:
```
lv_obj_get_child_by_id // returns the first match
lv_display_get_obj_by_id // returns the first match
lv_obj_set_id // NULL clears the id
```

Internal this can be solved in multiple ways:

- by adding a new field name to _lv_obj_t_
- by adding a new list to _lv_global_t_ and store pairs of object pointers and names inside

One more thing to consider: The term _id_ is already in use by _OBJ_ID_, maybe the term _name_ should be used.

XuNeo, 2024-02-22:
The `OBJ_ID` has take this into consideration. The ID could be in any form, thus is left to user to define it, like `btn_exit`. This can be accomplished by disable `LV_USE_OBJ_ID_BUILTIN` and provide your version of 
```

/**
 * Assign an id to an object if not previously assigned
 * Set `LV_USE_OBJ_ID_BUILTIN` to 1 to use builtin method to generate object ID.
 * Otherwise, these functions including `lv_obj_[assign|free|stringify]_id` should be implemented externally.
 *
 * @param class_p   the class this obj belongs to. Note obj->class_p is the class currently being constructed.
 * @param obj   pointer to an object
 */
void lv_obj_assign_id(const lv_obj_class_t * class_p, lv_obj_t * obj);

/**
 * Free resources allocated by `lv_obj_assign_id`
 * @param obj   pointer to an object
 */
void lv_obj_free_id(lv_obj_t * obj);

/**
 * Format an object's id into a string.
 * @param obj   pointer to an object
 * @param buf   buffer to write the string into
 * @param len   length of the buffer
 */
const char * lv_obj_stringify_id(lv_obj_t * obj, char * buf, uint32_t len);

```

For `lv_obj_get_child_by_id`, it can be implemented similar to `lv_obj_dump_tree`.

The reason LVGL does not provide such function is that there are many ways to define and compare ID. It can be easily implemented in user code though.

MiSimon, 2024-02-23:
I thought about using the current `OBJ_ID` feature to implement this functionality, but it is missing some functions:
- `lv_obj_set_id`, a method to set an id for an existing object, `lv_obj_assign_id` is called at object creation inside lv_obj_constructor and doesn't allow the user to define an arbitrary id directly
- `lv_obj_compare_id`, LV_OBJ_ID lets the user define ids of any type, to search for a specific id a new method to compare ids would be required
- the methods to search for an id 

If this methods can be added a new feature is not required. But i don't know what the original purpose of `OBJ_ID `was and if adding this methods would interfere with that.

XuNeo, 2024-02-23:
`lv_obj_set/get_id` can be added similar to other properties.
`lv_obj_compare_id` however, I cannot see how to add a general implementation, since the ID is user-defined `void *` pointer.

MiSimon, 2024-02-23:
I think this can only work if `LV_USE_OBJ_ID_BUILTIN` is disables, then the user defines the ids and knows how to implement `lv_obj_compare_id`.

XuNeo, 2024-02-23:
Yes, `LV_USE_OBJ_ID_BUILTIN ` should be disabled. So you can define you own ID, or simply provide an empty function `lv_obj_assign_id ` so you can call `lv_obj_set_id` manually.

kisvegabor, 2024-02-24:
So in summary `lv_obj_set_id(obj, "string")` will be enabled only if `LV_USE_OBJ_ID_BUILTIN` is disabled to avoid conflicts. 
It makes sense to me.

kdschlosser, 2024-02-26:
What about this?

Provide a function that deals with doing the comparisons when LV_USE_OBJ_ID_BUILTIN set to 1. Create another macro called LV_OBJ_ID_SEARCH and point it to that function. The user would then be able to set that macro to a different function for the id comparisons if they wanted to. It would be a requirement for the user to set it if  LV_USE_OBJ_ID_BUILTIN is set to 0 and LV_USE_OBJ_ID is set to 1. The same kind of thing can also be done to handle collecting the ID that gets set and also freeing the id.

Add LVGL API functions for searching and also for getting the id. The type when searching would be a `void *` for passing in the id to search for and the returned value would be the object or `NULL`. the get id function would return a type of `void *`. It would be up to the user to take care of any casting that needs to be done.

Optionally you could simply lock the object ids to integers and the user if wanting to use some for of a name instead of an integer they would have to create a lookup table to do that. They would have to manage the table themselves.

The hard part with integers is the ability to recycle the id's of objects that have been deleted. You cannot change the id of an object that exists because of another object being deleted.... so say you have the data type set to uint32_t and while I find it highly unlikely the object ID count would exceed 4,294,967,295 it is not out of the realm of being possible if a users UI is creating and deleting a lot of objects all the time.

The best way is probably going to be adding those macros and removing the builtin handling all together and let the user supply the functions for collecting a new id, comparing ids and freeing ids.

if the user wants to be able to set the id manually using say a name they would supply a function that returns NULL for the function that LVGL is going to call to get an id. then the user would manually call `lv_obj_set_id(obj, (void *)some_id)`

MiSimon, 2024-02-26:
I would not enable `lv_obj_set_id` if `LV_USE_OBJ_ID_BUILTIN` is enabled, if the user wants to handle the ids then `LV_USE_OBJ_ID_BUILTIN` should be disabled and own functions should be provided.

`lv_obj_get_id` and the search methods can be enabled with `LV_USE_OBJ_ID_BUILTIN`, then `lv_obj_compare_id` has to compare the ids using `lv_obj_stringify_id`.

kisvegabor, 2024-03-03:
I also find cleaner to use ether the built in IDs or the user IDs.

kisvegabor, 2024-04-05:
Not stale

kisvegabor, 2024-06-20:
Implemented in https://github.com/lvgl/lvgl/pull/6278

closed 2024-06-20 by kisvegabor