lvgl/lvgl #5723: 'lv_image_header_t' has no member named 'magic' error from the C file created by the online image converter for lvgl v9

TheWaterDiviner, 2024-02-22:
### LVGL version

V9.0.0

### What happened?

The C file crated by the online converter adds a non existing member to the _lv_image_header_t_ structure.

### How to reproduce?

const lv_image_dsc_t drop40px = {
  .header.cf = LV_COLOR_FORMAT_ARGB8888,
  .header.magic = LV_IMAGE_HEADER_MAGIC,
  .header.w = 40,
  .header.h = 40,
  .data_size = 1600 * 4,
  .data = drop40px_map,
};

XuNeo, 2024-02-23:
LVGL v9 has magic defined, please check if lvgl version is v9?
```c
typedef struct {
    uint32_t magic: 8;          /*Magic number. Must be LV_IMAGE_HEADER_MAGIC*/
    uint32_t cf : 8;            /*Color format: See `lv_color_format_t`*/
    uint32_t flags: 16;         /*Image flags, see `lv_image_flags_t`*/

    uint32_t w: 16;
    uint32_t h: 16;
    uint32_t stride: 16;        /*Number of bytes in a row*/
    uint32_t reserved_2: 16;    /*Reserved to be used later*/
} lv_image_header_t;
```

TheWaterDiviner, 2024-02-23:
OK thanks, it looks I'm using a preliminary v9 version for Renesas. I will check with them.

TheWaterDiviner, 2024-03-11:
I was using a preliminary/development version from Renesas,
The latest release is working as expected.

closed 2024-03-11 by TheWaterDiviner