lvgl/lvgl #5724: Alpha image overlay linux framebuffer...

planevina, 2024-02-22:
### Introduce the problem

lvgl v8.3.11

I use lvgl with frame_buffer mode on arm linux device ( https://blog.lvgl.io/2018-01-03/linux_fb  )
It's work.
I use a  background video decoder(provide by arm linux)  to play mp4 . It decode video and draw it on **background layer below fb0**.

So my image's half alpha edge turn black ,(because my scr bg color is black) ,
lv_obj_t *screen_act = lv_scr_act();
    lv_obj_remove_style(screen_act, NULL, LV_PART_SCROLLBAR);
    lv_obj_clear_flag(screen_act, LV_OBJ_FLAG_SCROLLABLE);
    lv_obj_set_scrollbar_mode(screen_act, LV_SCROLLBAR_MODE_OFF);
    lv_obj_set_style_bg_color(screen_act, lv_color_black(), 0);
    lv_obj_set_style_bg_opa(screen_act, LV_OPA_TRANSP, 0);

like this 
[image]

I know why it's like this ..
Video frame is the bottom layer ,lvgl cannot mix alpha with it .. lvgl only control the fb0 layer ..

I want it mix with background pixel smoothly  like this ,Is there any way to do it?
[image]

Thank you.

planevina, 2024-02-23:
Can I set the lvgl screen to the ARGB8888 format, so that it retains the transparency channel information, allowing the underlying Linux graphics mixing engine to blend it?

planevina, 2024-02-23:
Through my tests, I found out that in the fbdev_flush function, the color value passed by lvgl does not contain transparency information, and it has already been blended. For example, when I display an image with 50% green transparency, the value in the color_p passed by lvgl is 0xFF008000, which means opaque green with a color value of 128. Is there a possibility to have it pass 0x8000FF00 instead?

planevina, 2024-02-23:
if( y > 223 && y < 256 )
            {
                uint32_t* p = (uint32_t*)color_p;
                printf("y:%u,color:%u\n", y, p[240]);

                for(int k = 223;k<256;k++)
                {
                    p[k] &= 0x00FFFFFF;
                    p[k] |= 0x80000000;
                }
            }

I have made an attempt in the fbdev_flush function to forcefully change the transparency of the image part from FF to 80. Linux is able to perform semi-transparent blending normally. Therefore, if lvgl could pass semi-transparent data, I would be able to implement the desired functionality.

[image]

planevina, 2024-02-23:
I conducted another test where I set the layer transparency to 50%. It was successfully passed to the flush function and overlaid properly. So, the only issue is that semi-transparent elements in the image cannot be passed through.

kisvegabor, 2024-02-24:
In v8 You need to enable `LV_COLOR_SCREEN_TRANSP` in `lv_conf.h`.

In v9 you can set `lv_display_set_color_format(NULL, LV_COLOR_FORMAT_ARGB8888);`.

planevina, 2024-02-24:
> In v8 You need to enable `LV_COLOR_SCREEN_TRANSP` in `lv_conf.h`.
> 
> In v9 you can set `lv_display_set_color_format(NULL, LV_COLOR_FORMAT_ARGB8888);`.

I already  enable it  LV_COLOR_SCREEN_TRANSP .

planevina, 2024-02-24:
I just discovered that not only semi-transparent images, but also drawing circles and diagonal lines will generate black pixels around them. Here is a video. This is a Linux single-board computer, using fb0 framebuffer mode. The bottom layer is playing a video, lvgl is on fb0, the screen background of lvgl is transparent, and I created a new custom object, its background is also transparent. I drew circles and diagonal lines inside this custom object.

https://github.com/lvgl/lvgl/assets/70463000/a10a4df1-fc82-4f20-a516-4f96b5051f79

planevina, 2024-02-24:
Okay, finally, I found a temporary solution. I modified the lv_color_mix function, just like this. Now it blends colors normally, which is great. Thanks to lvgl, it's a fantastic GUI library.

[image]

closed 2024-02-24 by planevina

PawnMa, 2024-12-03:
Thank you very much for providing your solution

planevina, 2024-12-03:
Finally, I found the correct way is  "  disp_drv.screen_transp = 1; "