lvgl/lvgl #5744: Clarification about lv_timer_handler documentation - What does “lower priority than lv_tick_inc()” mean?

LeCyberDucky, 2024-02-24:
### Introduce the problem

Hey, so in the [timer handler documentation](https://docs.lvgl.io/master/porting/timer_handler.html#timer) the following is mentioned:

> To handle the tasks of LVGL you need to call lv_timer_handler() periodically in one of the following:
> 
> while(1) of main() function
> timer interrupt periodically (lower priority than lv_tick_inc())
> an OS task periodically

I would like some clarification about what exactly the

> lower priority than lv_tick_inc()

part means.

I’m using an ESP32, and at the moment I handle my ticks like this:

    const esp_timer_create_args_t lvgl_tick_timer_args = {
        .callback = [](void *arg)
        {lv_tick_inc(LVGL_TICK_PERIOD_MS); },
        .name = "lvgl_tick"};
    esp_timer_handle_t lvgl_tick_timer = nullptr;
    ESP_ERROR_CHECK(esp_timer_create(&lvgl_tick_timer_args, &lvgl_tick_timer));
    ESP_ERROR_CHECK(esp_timer_start_periodic(lvgl_tick_timer, LVGL_TICK_PERIOD_MS * 1000));

Is there anything wrong with placing the call to lv_timer_handler() in there as well? Like this:

    const esp_timer_create_args_t lvgl_tick_timer_args = {
        .callback = [](void *arg)
        {lv_tick_inc(LVGL_TICK_PERIOD_MS);
        lv_timer_handler(); },
        .name = "lvgl_tick"};
    esp_timer_handle_t lvgl_tick_timer = nullptr;
    ESP_ERROR_CHECK(esp_timer_create(&lvgl_tick_timer_args, &lvgl_tick_timer));
    ESP_ERROR_CHECK(esp_timer_start_periodic(lvgl_tick_timer, LVGL_TICK_PERIOD_MS * 1000));

kisvegabor, 2024-03-03:
While `lv_timer_handler` is running LVGL needs to able to count the elapsed time to keep track of the progress of its timers.  

>    {lv_tick_inc(LVGL_TICK_PERIOD_MS);
    lv_timer_handler(); },

In this configuration if `lv_timer_handler` lasts for 20 ms, you will miss 19 tick which will mess up a the timing of LVGL. Therefore `lv_tick_inc` needs to be called periodically no matter what. 

FYI, in LVGL v9 you can set a callback for the tick as well. I've just update the docs a little to make it clearer. See #5781.

LeCyberDucky, 2024-03-03:
> In this configuration if `lv_timer_handler` lasts for 20 ms, you will miss 19 tick which will mess up a the timing of LVGL. Therefore `lv_tick_inc` needs to be called periodically no matter what.

That's a good point, I didn't consider that `lv_timer_handler` may take a while to run. 

Another problem, that I have since become aware of with this approach, is that calling `lv_timer_handler` separately from the main thread like this will cause problems due to a lack of thread-safety. To do this, you'd have to wrap all other calls to LVGL-related functions in locks.  

> FYI, in LVGL v9 you can set a callback for the tick as well. I've just update the docs a little to make it clearer. See #5781.

Would this be the preferred way to do this now?

hyansuper, 2024-03-03:
> In this configuration if lv_timer_handler lasts for 20 ms, you will miss 19 tick which will mess up a the timing of LVGL.

How about this :
```c
    TickType_t xLastWakeTime = xTaskGetTickCount();
    while(1) {
        vTaskDelayUntil(&xLastWakeTime, pdMS_TO_TICKS(LV_DEF_REFR_PERIOD));
        lv_tick_inc(LV_DEF_REFR_PERIOD);
        LVGL_LOCK(
            lv_task_handler();
        )
    }
```

hyansuper, 2024-03-06:
If lv_timer_handler() and lv_tick_inc() must be called in separate threads, should one be called more frequently than the other?

kisvegabor, 2024-03-11:
When you call `lv_timer_handler` itt will execute all kind of LVGL things, like redering, reading input devices, updating animations, etc. You can call it as often you want, but it's recommended to have at least a 5ms period to get a responsive result.

As LVGL doesn't know how often you call `lv_timer_handler` you need to tell LVGL what time is it, so that it will know the status of the animations, whether is needs to render now, etc.

closed 2024-04-03 by lvgl-bot