lvgl/lvgl #5746: The effect of setting lv_obj_set_style_transform_scale is different from setting scale_x/scale_y separately.

C-Ackerman, 2024-02-26:
### LVGL version

v9.0.0

### What happened?

```
void lv_example_test(void)
{
    lv_obj_t * cont = lv_obj_create(lv_screen_active());
    lv_obj_set_size(cont, 400, 500);
    lv_obj_center(cont);
    lv_obj_set_flex_flow(cont, LV_FLEX_FLOW_COLUMN);

    lv_obj_t * label = lv_label_create(cont);
    lv_obj_set_size(label, 100, 100);
    lv_label_set_text(label, "scale(0)\n");
    lv_obj_center(label);

    lv_obj_t * label1 = lv_label_create(cont);
    lv_obj_set_size(label1, 100, 100);
    lv_label_set_text(label1, "scale(0.5)\n");
    lv_obj_set_style_transform_scale(label1, (0.5 * LV_ZOOM_NONE), LV_PART_MAIN);
    lv_obj_center(label1);

    lv_obj_t * label2 = lv_label_create(cont);
    lv_obj_set_size(label2, 100, 100);
    lv_label_set_text(label2, "scalex(0.5)\n");
    lv_obj_set_style_transform_scale_x(label2, (0.5 * LV_ZOOM_NONE), LV_PART_MAIN);
    lv_obj_center(label2);

    lv_obj_t * label3 = lv_label_create(cont);
    lv_obj_set_size(label3, 100, 100);
    lv_label_set_text(label3, "scaley(0.5)\n");
    lv_obj_set_style_transform_scale_y(label3, (0.5 * LV_ZOOM_NONE), LV_PART_MAIN);
    lv_obj_center(label3);

}
```

[image]

The effect of setting lv_obj_set_style_transform_scale is different from setting scale_x/scale_y separately. As shown in the figure, setting scale_y separately is stretched, and scale_x is also a bit stretched.

kisvegabor, 2024-02-28:
It seems something in not consistent between the code and the image:
- in the code there is no border set
- the widget size is  set to  100x100, so a square, but even the first widget is a rectangle

Could you check it?

C-Ackerman, 2024-02-28:
Sorry, this picture is the effect of the example code. The stretching effect is the same as that of adding a border.

[image]

C-Ackerman, 2024-02-28:
[image]

remove size 100×100，Setting the size does not affect the effect

```
void lv_example_test(void)
{
    lv_obj_t * cont = lv_obj_create(lv_screen_active());
    lv_obj_set_size(cont, 400, 500);
    lv_obj_center(cont);
    lv_obj_set_flex_flow(cont, LV_FLEX_FLOW_COLUMN);

    lv_obj_t * label = lv_label_create(cont);
    // lv_obj_set_size(label, 100, 100);
    lv_label_set_text(label, "scale(0)\n");
    lv_obj_center(label);

    lv_obj_t * label1 = lv_label_create(cont);
    // lv_obj_set_size(label1, 100, 100);
    lv_label_set_text(label1, "scale(0)\n");
    lv_obj_set_style_transform_scale(label1, (0.5 * LV_ZOOM_NONE), LV_PART_MAIN);
    lv_obj_center(label1);

    lv_obj_t * label2 = lv_label_create(cont);
    // lv_obj_set_size(label2, 100, 100);
    lv_label_set_text(label2, "scale(0)\n");
    lv_obj_set_style_transform_scale_x(label2, (0.5 * LV_ZOOM_NONE), LV_PART_MAIN);
    lv_obj_center(label2);

    lv_obj_t * label3 = lv_label_create(cont);
    // lv_obj_set_size(label3, 100, 100);
    lv_label_set_text(label3, "scale(0)\n");
    lv_obj_set_style_transform_scale_y(label3, (0.5 * LV_ZOOM_NONE), LV_PART_MAIN);
    lv_obj_center(label3);

}
```

kisvegabor, 2024-03-03:
It seems correct to me. What the actual issue is?

C-Ackerman, 2024-03-04:
Sorry, I was wrong

closed 2024-03-04 by C-Ackerman