lvgl/lvgl #5747: compiles errors using MSVC caused by syntax that doesn't conform to c99 standard

kdschlosser, 2024-02-26:
### LVGL version

v9.0.0

### What happened?

compile errors
```

src\lvgl\src\draw\sw\lv_draw_sw_letter.c(82): error C2143: syntax error: missing ':' before '...'
src\lvgl\src\draw\sw\lv_draw_sw_letter.c(84): error C2065: 'mask_area': undeclared identifier
src\lvgl\src\draw\sw\lv_draw_sw_letter.c(84): error C2224: left of '.x2' must have struct/union type
src\lvgl\src\draw\sw\lv_draw_sw_letter.c(84): error C2065: 'mask_area': undeclared identifier
src\lvgl\src\draw\sw\lv_draw_sw_letter.c(84): error C2224: left of '.x1' must have struct/union type
src\lvgl\src\draw\sw\lv_draw_sw_letter.c(84): error C2065: 'mask_area': undeclared identifier
src\lvgl\src\draw\sw\lv_draw_sw_letter.c(91): error C2065: 'mask_area': undeclared identifier
src\lvgl\src\draw\sw\lv_draw_sw_letter.c(98): error C2043: illegal break
src\lvgl\src\draw\sw\lv_draw_sw_letter.c(99): error C2046: illegal case
src\lvgl\src\draw\sw\lv_draw_sw_letter.c(111): error C2043: illegal break
src\lvgl\src\draw\sw\lv_draw_sw_letter.c(112): error C2047: illegal default
src\lvgl\src\draw\sw\lv_draw_sw_letter.c(113): error C2043: illegal break
src\lvgl\src\draw\sw\lv_draw_sw_letter.c(118): error C2059: syntax error: 'if'
src\lvgl\src\draw\sw\lv_draw_sw_letter.c(121): error C2059: syntax error: '}'

```

here are some compiler warnings also

```
src\lvgl\src\layouts\lv_layout.c(71): warning C4018: '<=': signed/unsigned mismatch
src\lvgl\src\libs\gif\lv_gif.c(163): warning C4018: '<': signed/unsigned mismatch
src\lvgl\src\libs\lodepng\lodepng.c(762): warning C4334: '<<': result of 32-bit shift implicitly converted to 64 bits (was 64-bit shift intended?)
src\lvgl\src\libs\lodepng\lodepng.c(781): warning C4334: '<<': result of 32-bit shift implicitly converted to 64 bits (was 64-bit shift intended?)
src\lvgl\src\libs\lodepng\lodepng.c(5878): warning C4334: '<<': result of 32-bit shift implicitly converted to 64 bits (was 64-bit shift intended?)
src\lvgl\src\libs\qrcode\lv_qrcode.c(118): warning C4018: '>': signed/unsigned mismatch
src\lvgl\src\libs\tjpgd\lv_tjpgd.c(222): warning C4146: unary minus operator applied to unsigned type, result still unsigned
src\lvgl\src\misc\cache\_lv_cache_lru_rb.c(171): warning C4047: 'return': 'bool' differs in levels of indirection from 'void *'
src\lvgl\src\misc\cache\_lv_cache_lru_rb.c(195): warning C4047: 'return': 'bool' differs in levels of indirection from 'void *'
src\lvgl\src\misc\lv_rb.c(246): warning C4047: 'return': 'bool' differs in levels of indirection from 'void *'
src\lvgl\src\others\observer\lv_observer.c(275): warning C4018: '>=': signed/unsigned mismatch
src\lvgl\src\widgets\calendar\lv_calendar.c(136): warning C4018: '<': signed/unsigned mismatch
src\lvgl\src\widgets\imagebutton\lv_imagebutton.c(174): warning C4018: '>': signed/unsigned mismatch
src\lvgl\src\widgets\spinbox\lv_spinbox.c(447): warning C4018: '<': signed/unsigned mismatch
src\lvgl\src\draw\sw\lv_draw_sw_letter.c(84): warning C4133: 'function': incompatible types - from 'int *' to 'const lv_area_t *'
src\lvgl\src\draw\sw\lv_draw_sw_letter.c(91): warning C4133: '=': incompatible types - from 'int *' to 'const lv_area_t *'
```

the use of ellipsis in switch case is not c99 standard it is a gcc extension that allows you to use the ellipsis to specify a range. the use of curly brackets for case is also not apart of the c99 standard. 

The below code is not to c99 standards.

```c
static void LV_ATTRIBUTE_FAST_MEM draw_letter_cb(lv_draw_unit_t * draw_unit, lv_draw_glyph_dsc_t * glyph_draw_dsc,
                                                 lv_draw_fill_dsc_t * fill_draw_dsc, const lv_area_t * fill_area)
{
    if(glyph_draw_dsc) {
        switch(glyph_draw_dsc->format) {
            case LV_FONT_GLYPH_FORMAT_NONE: {
#if LV_USE_FONT_PLACEHOLDER
                    /* Draw a placeholder rectangle*/
                    lv_draw_border_dsc_t border_draw_dsc;
                    lv_draw_border_dsc_init(&border_draw_dsc);
                    border_draw_dsc.opa = glyph_draw_dsc->opa;
                    border_draw_dsc.color = glyph_draw_dsc->color;
                    border_draw_dsc.width = 1;
                    lv_draw_sw_border(draw_unit, &border_draw_dsc, glyph_draw_dsc->bg_coords);
#endif
                }
                break;
            case LV_FONT_GLYPH_FORMAT_A1 ... LV_FONT_GLYPH_FORMAT_A8: {
[... 93 lines cut ...]
                img_dsc.rotation = 0;
                img_dsc.scale_x = LV_SCALE_NONE;
                img_dsc.scale_y = LV_SCALE_NONE;
                img_dsc.opa = glyph_draw_dsc->opa;
                img_dsc.src = glyph_draw_dsc->glyph_data;
                lv_draw_sw_image(draw_unit, &img_dsc, glyph_draw_dsc->letter_coords);
#endif
                break;
            default:
                break;
        }

    }

    if(fill_draw_dsc && fill_area) {
        lv_draw_sw_fill(draw_unit, fill_draw_dsc, fill_area);
    }
}
```

I did use the code above and it compiles without any errors the the 2 warnings in lv_draw_sw_letter.c have been fixed. I am sure there is probably a better way of doing the same thing as what I did. from what I could tell is it looks like the lv_area_t structure stored in lv_draw_glyph_dsc_t.letter_coords is not supposed to be altered which is why I created a new lv_area_t variable and copied the contents over. copying the memory I would guess would be a better way of doing it instead of copying the values. I am not all that proficient in C code.

### How to reproduce?

compile using MSVC with LV_USE_DRAW_SW set to 1

XuNeo, 2024-02-26:
It has been fixed in master branch with CI enabled to check non-compliant code.

kdschlosser, 2024-02-28:
not quite. only the ellipsis has been corrected not the use of curly braces for the case statements and not the compiler warnings.

XuNeo, 2024-02-28:
I think both of them should work. The first case is only wrapping some statements into a block. Do you see warnings?

kdschlosser, 2024-02-28:
I have not checked to see if it is complaining about the braces. Are the braces serving a purpose? I don't really see a need for them to be there as they have zero effect on how the code runs.

I can tell you that there are compiler warnings for these 2 lines
```c
                    mask_area.x2 = mask_area.x1 + lv_draw_buf_width_to_stride(lv_area_get_width(&mask_area), LV_COLOR_FORMAT_A8) - 1;
```

```c
                    blend_dsc.mask_area = &mask_area;
```

they are noted in the original post

```
src\lvgl\src\draw\sw\lv_draw_sw_letter.c(84): warning C4133: 'function': incompatible types - from 'int *' to 'const lv_area_t *'
src\lvgl\src\draw\sw\lv_draw_sw_letter.c(91): warning C4133: '=': incompatible types - from 'int *' to 'const lv_area_t *'

```

with this change to the code there are no more warnings.

```
lv_area_t mask_area;
                lv_memzero(&mask_area, sizeof(mask_area));

                mask_area.x1 = glyph_draw_dsc->letter_coords->x1;
                mask_area.y1 = glyph_draw_dsc->letter_coords->y1;
                mask_area.x2 = glyph_draw_dsc->letter_coords->x2;
                mask_area.y2 = glyph_draw_dsc->letter_coords->y2;
                mask_area.x2 = mask_area.x1 + lv_draw_buf_width_to_stride(lv_area_get_width(glyph_draw_dsc->letter_coords), LV_COLOR_FORMAT_A8) - 1;

                lv_draw_sw_blend_dsc_t blend_dsc;
                lv_memzero(&blend_dsc, sizeof(blend_dsc));

                blend_dsc.color = glyph_draw_dsc->color;
                blend_dsc.opa = glyph_draw_dsc->opa;

                lv_draw_buf_t * draw_buf = glyph_draw_dsc->glyph_data;

                blend_dsc.mask_buf = draw_buf->data;
                blend_dsc.mask_area = &mask_area;
                blend_dsc.mask_stride = draw_buf->header.stride;
                blend_dsc.blend_area = glyph_draw_dsc->letter_coords;
                blend_dsc.mask_res = LV_DRAW_SW_MASK_RES_CHANGED;

                lv_draw_sw_blend(draw_unit, &blend_dsc);
```

I am sure there is a better way of handling the compiler warnings than what I have done above but it does work.

XuNeo, 2024-02-28:
Hi,
Could you push a PR to see where has been changed to solve the warning? I check with my local code and lv_draw_sw_letter.c:84 and lien 91 doesn't match with your warning.

kdschlosser, 2024-02-28:
I just removed the lines of code I added and it seems to compile without any warnings using the original code. The only things that are different is I removed the ellipsis and also the curly braces. Maybe the warnings were being caused by either of those things being there. Either way it is compiling without any additional errors or warnings so I will close this and leave it as it is.

If you ever need to have MSVC tested let me know. I wrote a build system for Python to compile LVGL and it does it in short order. 6 seconds from start to finish and that includes compiling the demos and examples as well. I have to locate the windows binaries for some of the extras that LVGL can use like libpng and the turbo JPG. nothing crazy tho.

https://youtu.be/Fn7NPscEypk