lvgl/lvgl #5755: LVGL V8.3 List the Chinese characters dsiplay missing

litian85734, 2024-02-27:
### LVGL version

V8.3

### What happened?

[image]
Dear Sir：
    I want show some Chinese characters in list， But I fond the head of the chracter "页" is not display， 
I use lable display is ok with same font.
    I am not sure is my usage issue of list or other reason.  Could you help to teach me how to fix it.

    Greate thanks for your help ^^.
     
Thanks

### How to reproduce?

void lv_example_list_1(void)
{
    /*Create a list*/
    lv_obj_t * list1 = lv_list_create(lv_scr_act());
    lv_obj_set_size(list1, 180, 220);
    lv_obj_center(list1);
   	lv_obj_set_style_text_font(list1, &lv_font_simsun_16, LV_PART_MAIN|LV_STATE_DEFAULT);
    lv_obj_align(list1, LV_ALIGN_CENTER, 0, 0);
    lv_obj_set_flex_flow(list1, LV_FLEX_FLOW_COLUMN_WRAP);
   
    /*Add buttons to the list*/
    lv_obj_t * btn;
    //Display NG when use list
    lv_list_add_text(list1, "页面！ABC");
    btn = lv_list_add_btn(list1, LV_SYMBOL_FILE, "页面！ABC");
    lv_obj_add_event_cb(btn, event_handler1, LV_EVENT_CLICKED, NULL);
    btn = lv_list_add_btn(list1, LV_SYMBOL_SAVE, "页面！ABC");
    lv_obj_add_event_cb(btn, event_handler1, LV_EVENT_CLICKED, NULL);
    btn = lv_list_add_btn(list1, LV_SYMBOL_CLOSE, "Delete");
    //Display OK wehn use label
    lv_obj_t * label2 = lv_label_create(lv_scr_act());
    lv_label_set_long_mode(label2, LV_LABEL_LONG_DOT);     /*Circular scroll*/
    lv_obj_set_pos(label2, 20, 0);
    lv_obj_set_size(label2, 200, 100);
    lv_label_set_text(label2, "页面！ABC ");
    lv_obj_align(label2, LV_ALIGN_LEFT_MID, 0, 20);
   lv_obj_set_style_text_font(label2, &lv_font_simsun_16, LV_PART_MAIN|LV_STATE_DEFAULT);
}

FASTSHIFT, 2024-02-27:
Can you provide font files?

litian85734, 2024-02-27:
[lv_font_simsun_16.zip](https://github.com/lvgl/lvgl/files/14414458/lv_font_simsun_16.zip)
Dear Sir:
  
   Thanks for your quickly help.
    Attached file is my font lib

Thanks

FASTSHIFT, 2024-02-27:
Please use the latest [font converter](https://lvgl.io/tools/fontconverter) to convert the font.
I tried to convert the font again and found that the `base_line` parameter had changed:
[image]

Fix after:
[image]

litian85734, 2024-02-28:
Dear Sir：

   Great!   Thanks you so much for your help ^^
   May I kow why Lable display is correctly,   Lable no need to refer base_line?

Thanks

FASTSHIFT, 2024-03-01:
The label is actually drawn outside the widget, probably related to the redraw area calculation process.

closed 2024-03-23 by lvgl-bot