lvgl/lvgl #5759: X11 driver hangs or crashes on assert on delete event

Maldus512, 2024-02-27:
### LVGL version

v9.0.0-197-gc25555107

### What happened?

The X11 display driver's `LV_EVENT_DELETE` handler passes a NULL buffer to `lv_display_set_buffers`, presumably to deinitialize it.

https://github.com/lvgl/lvgl/blob/c25555107ca42fd0051800823452d514497d9ae8/src/drivers/x11/lv_x11_display.c#L202

However `lv_display_set_buffers` immediately asserts that the received buffer is not NULL, leading to a crash or freezing the program:

https://github.com/lvgl/lvgl/blob/c25555107ca42fd0051800823452d514497d9ae8/src/display/lv_display.c#L402

What is the point of that call? Could it be removed altogether?

FASTSHIFT, 2024-03-01:
Have you tried removing it `lv_display_set_buffers(disp, NULL, NULL, 0, LV_DISPLAY_RENDER_MODE_PARTIAL);`? I think it should be unnecessary.

kisvegabor, 2024-03-05:
I confirm that it's not required to set buffer to `NULL`.

Fixed here: #5799

PR #5799 (2024-03-08): fix(x11): fix assert on delete
fixes #5759

### Description of the feature or fix

A clear and concise description of what the bug or new feature is.

### Notes
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.
- Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant.
- Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if applicable.
- If you added new options to `lv_conf_template.h` run [lv_conf_internal_gen.py](https://github.com/lvgl/lvgl/blob/master/scripts/lv_conf_internal_gen.py) and update [Kconfig](https:// [... cut]
closed 2024-03-08 by XuNeo