lvgl/lvgl #5760: LVGL V9: Object and Image creater do not free the memory

fatiihyildizz, 2024-02-27:
### Introduce the problem

Hello,

I designed a 2 page project using LVGL V9. Transitions between pages are done with gestures. When the Gesture event comes, I change the pages as follows (I did it using the GUI Guide as a reference) : 
[image]

The same event is also available on the other screen. But after scrolling through the pages for a while, I get the following error:
[image]

In the screen_init functions, I call the lv_obj_create function for the screen and add images to that object with lv_img_create. Why am I getting an overflow error? Shouldn't LVGL free some things when creating img and obj? Or is there something I'm missing?

kisvegabor, 2024-03-03:
The last parameter of `lv_screen_load_anim` is `false` now, which means the the screen that you are leaving won't be deleted automatically. 

So do you create the screen periodically somewhere or just create them once on start up?

fatiihyildizz, 2024-03-05:
@kisvegabor I reinstall it when I enter the relevant screen

kisvegabor, 2024-03-13:
- Do you get the same error is you call `lv_obj_invalidate(lv_screen_actiev())` after `lv_timer_handler` and wait a minute on each screen?
- The issue can be due to memory fragmentation too. Is it better if you increase `LV_MEM_SIZE`?

closed 2024-04-08 by lvgl-bot