lvgl/lvgl #5768: inlining lv_event_get_code(), _get_target, and get_user_data()

bryghtlabs-richard, 2024-02-28:
### Introduce the problem

I found that a few LVGL functions are both short and tend to be very commonly called, and when I inline the implementation in to lv_event.h, my binary is about 1KB smaller. This seemed like it might make sense to upstream.

void * lv_event_get_user_data(lv_event_t * e);
lv_obj_t * lv_event_get_target_obj(lv_event_t * e);
lv_event_code_t lv_event_get_code(lv_event_t * e);

### Proposal

Inline these three commonly called functions.

XuNeo, 2024-02-29:
That makes sense. Normally inline will get larger binary, but it seems for those three functions, calling a function is more complex than accessing the struct directly.

Could you get a PR for your change?

cc @kisvegabor This can't be done if struct is hidden, what do you think of this change?

kisvegabor, 2024-03-05:
@XuNeo 

> This can't be done if struct is hidden, what do you think of this change?
`struct _lv_event_t` is in `lv_event.h`. it required a little bit more parenthesis then I expected but this works for me:
```c
static inline lv_event_code_t lv_event_get_code(lv_event_t * e)
{
	return (lv_event_code_t)(((struct _lv_event_t*)e)->code & ~LV_EVENT_PREPROCESS);
}
``` 

@bryghtlabs-richard 
A PR  for v9 would be very welcome, thank you! :+1:

bryghtlabs-richard, 2024-03-20:
I don't expect to get back to this for several weeks. It's ok with me if this is auto-closed. We can pick it up later as time allows.

kisvegabor, 2024-03-25:
It's not a blocking update so take your time. :slightly_smiling_face:

closed 2024-04-29 by lvgl-bot