lvgl/lvgl #5771: ThorVG Lottie example

MR-VENOM, 2024-02-29:
### Introduce the problem

In change log it is mentioned that LVGL now support thorvg to canvas drawing i was thinking to use this to draw lottie images to the canvas but there is no example available showcasing this. I am trying run it on an esp32s3 with 8mb external psram

XuNeo, 2024-03-01:
For thorvg, it only cares the memory can be used to draw. In lvgl, you can get set the same memory to canvas so it can show on screen.

```cpp
static uint32_t buffer[WIDTH * HEIGHT];                                 //canvas target buffer

auto canvas = tvg::SwCanvas::gen();                                     //generate a canvas
canvas->target(buffer, WIDTH, WIDTH, HEIGHT, tvg::SwCanvas::ARGB8888);  //buffer, stride, w, h, Colorspace

```

```c

    /*Define a static buffer for the canvas*/
    LV_DRAW_BUF_DEFINE(draw_buf, CANVAS_WIDTH, CANVAS_HEIGHT, LV_COLOR_FORMAT_ARGB8888);

    /*Create a canvas and initialize its palette*/
    lv_obj_t * canvas = lv_canvas_create(lv_screen_active());
    lv_canvas_set_draw_buf(canvas, &draw_buf);
```

Now you can combine those two examples. Note the stride in lvgl is in byte unit, while in thorvg is in pixel unit.

espressif2022, 2024-03-07:
Hello @XuNeo, If I want to create a `lottie` obj, the current V9 `thorvg` directory does not contain the `loaders/lottie` directory of the original repository.
The current introduction of `rlottie` is still download the https://github.com/Samsung/rlottie.git sub-repository.

If I use thorvg to load lottie, do you have any suggestions? How will it to be designed in the future?

XuNeo, 2024-03-07:
`lvgl` downloaded `throvg` and modified a bit to compile with lvgl, which is not a good way to introduce 3rd party lib in my opinion. This is to make `lvgl` work out-of-box. 

`thorvg` serves a pure vector draw engine for lvgl, I'm unfortunately not familiar with how to enable lottie.

espressif2022, 2024-03-08:
OK, thanks for your reply.

kisvegabor, 2024-03-09:
I'm working or it. ThorVG Lottie is already working locally for me.

MR-VENOM, 2024-03-28:
@kisvegabor any update on this

kisvegabor, 2024-03-28:
Now it's working not only on my PC bit also on some embedded devices (Renesas and ST).
I'm preparing the demos for [Embedded World](https://www.embedded-world.de/en/visitor?gad_source=1&gclid=Cj0KCQjwqpSwBhClARIsADlZ_Tl62muu86-Eoff0S1XwgKD6UCzbQbz6Getj5tFxtcmjTmDns8W0W70aAlOpEALw_wcB) but after that I'll clean up my implementation and send a PR. 

If you need a temporary super hacky soluton right now, take a look at this [repo](https://github.com/kisvegabor/lvgl_upstream/tree/renesas/ebike). From here you need
- [lv_demo_lottie.c/h](https://github.com/kisvegabor/lvgl_upstream/blob/renesas/ebike/demos/ebike/src/lv_demo_lottie.c)
- The updated [ThorVG lib](https://github.com/kisvegabor/lvgl_upstream/tree/renesas/ebike/src/libs/thorvg) 
- [Usage example](https://github.com/kisvegabor/lvgl_upstream/blob/renesas/ebike/demos/ebike/src/lv_demo_ebike_settings.c#L79-L84) 
- Convert Lottie JSON files to hex array with [this script](https://github.com/kisvegabor/lvgl_upstream/blob/renesas/ebike/scripts/filetohex.py).

espressif2022, 2024-04-02:
> Now it's working not only on my PC bit also on some embedded devices (Renesas and ST). I'm preparing the demos for [Embedded [...]

Hello
I tried running this routine on our esp32_s3, and the data printed and parsed were all 0x00.
If it is converted to RGB565 and sent to the screen, it will indeed display a 100*100 black background. Is there any abnormality in the configuration?
[image]

kisvegabor, 2024-04-03:
I forgot to mention that `LV_USE_THORVG_INTERNAL` and `LV_USE_VECTOR_GRAPHIC` needs to be enabled.

kisvegabor, 2024-04-03:
I've opened #6021 to track the progress of the built in Lottie support.

espressif2022, 2024-04-07:
> Sorry, I forgot to mention that `LV_USE_THORVG_INTERNAL` and `LV_USE_VECTOR_GRAPHIC` needs to be enabled.
> 
> Also be sure that you are using a newer version of LVGL where [this line looks like this](https://github.com/lvgl/lvgl/blob/master/src/lv_conf_internal.h#L3300). There was [a typo](https://github.com/lvgl/lvgl/pull/5837) here earlier.

It worked, thanks! `LV_USE_VECTOR_GRAPHIC` is needed.

MichaelDu9226, 2024-04-10:
@espressif2022 I am testing LVGL9 and ThorVG Lottie animations based on your https://github.com/espressif2022/esp-bsp .
The hardware is on the s3_lcd_ev-board platform. IDF uses v5.1.
Currently, I am using https://github.com/kisvegabor/lvgl_upstream/tree/renesas/ebike , but encountered the same compilation error as @kisvegabor https://github.com/jcmvbkbc/gcc-xtensa/issues/14#issuecomment-1994490029
May I ask if you have any suggestions or successful cases to share.

kisvegabor, 2024-04-15:
> Loading and compilation are OK, but some lottiles crash during running, I haven't found the cause of the crash yet.

Probably you need to increase the stack size of the LVGL task. I suggest starting with 48kB.

kisvegabor, 2024-05-22:
Still fighting with the CI in https://github.com/lvgl/lvgl/pull/6081

Please comment in the PR if it's not working well.

closed 2024-05-22 by kisvegabor

dj140, 2024-07-23:
> > @espressif2022 I am testing LVGL9 and ThorVG Lottie animations based on your https://github.com/espressif2022/esp-bsp . The hardware is on the s3_lcd_ev-board platform. IDF uses v5.1. Currently, I am using https://github.com/kisvegabor/lvgl_upstream/tree/renesas/ebike , but encountered the same compilation error as @kisvegabor [jcmvbkbc/gcc-xtensa#14 (comment)](https://github.com/jcmvbkbc/gcc-xtensa/issues/14#issuecomment-1994490029) May I ask if you have any suggestions or successful cases to share.
> 
> I added this `-fno-if-conversion` in CMakeLists.txt Loading and compilation are OK, but some lottiles crash during running, I haven't found the cause of the crash yet.

i get the same issue when using -O2 optimization level, i try `add_compile_options(-fno-if-conversion)` but it didn't work, can you share your setup, I'd really appreciate it !

espressif2022, 2024-07-23:
> Probably you need to increase the stack size of the LVGL task. I suggest starting with 48kB.

Did you tried to incrase the stack size, I think It's the crash reason.

dj140, 2024-07-23:
> > Probably you need to increase the stack size of the LVGL task. I suggest starting with 48kB.
> 
> @dj140 ,Did you tried to incrase the stack size, I think It's the crash reason.

i get compile error `insn does not satisfy its constraints` using -O2 optimization level, it work fine in O0 optimization

espressif2022, 2024-07-23:
> > > Probably you need to increase the stack size of the LVGL task. I suggest starting with 48kB.
> > 
> > 
> > @dj140 ,Did you tried to incrase the stack size, I think It's the crash reason.
> 
> i get compile error `insn does not satisfy its constraints` using -O2 optimization level, it work fine in O0 optimization

I looked at the related issue and it should be a known problem.

kisvegabor, 2024-07-24:
> i get the same issue when using -O2 optimization level, i try add_compile_options(-fno-if-conversion) but it didn't work, can you share your setup, I'd really appreciate it !

Finally, I couldn't make it work on ESP-S3 due to this issue. :slightly_frowning_face: 
I suggested opening an issue in ESP-IDF's repo as it's more specific to ESP.