lvgl/lvgl #5773: some of screens shows normal,some shows no data

Knight-yang, 2024-02-29:
### LVGL version

v8.3.6 (IDF V4.4.2)

### What happened?

my project used CF_TRUE_COLOR_ALPHA type,and the screens shows normally,but  CF_TRUE_COLOR_ALPHA type used too much size of flash.
 So I changed CF_RAW_ALPHA type,then some of image shows well,some of shows no data,The strange thing is that the project did not report an error
I changed LV_MEM_SIZE from 32 to128, and try to set image cache size to number of images(8~15),the problem is still here.
I can't find other solution ,could anyone please help me to solve this problem?

### How to reproduce?

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x1650
load:0x403c9700,len:0xbb8
load:0x403cc700,len:0x2f88
entry 0x403c9954
I (24) boot: ESP-IDF v4.4.2 2nd stage bootloader
I (25) boot: compile time 22:25:45
I (25) boot: chip revision: 0
I (26) boot.esp32s3: Boot SPI Speed : 80MHz
I (31) boot.esp32s3: SPI Mode       : DIO
I (36) boot.esp32s3: SPI Flash Size : 16MB
I (41) boot: Enabling RNG early entropy source...
I (46) boot: Partition Table:
I (50) boot: ## Label            Usage          Type ST Offset   Length
I (57) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (64) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (72) boot:  2 factory          factory app      00 00 00010000 00800000
I (79) boot: End of partition table
I (83) esp_image: segment 0: paddr=00010020 vaddr=3c060020 size=101650h (1054288) map
I (281) esp_image: segment 1: paddr=00111678 vaddr=3fc92340 size=03210h ( 12816) load
I (284) esp_image: segment 2: paddr=00114890 vaddr=40374000 size=0b788h ( 46984) load
I (297) esp_image: segment 3: paddr=00120020 vaddr=42000020 size=57b28h (359208) map
I (362) esp_image: segment 4: paddr=00177b50 vaddr=4037f788 size=02bb0h ( 11184) load
I (365) esp_image: segment 5: paddr=0017a708 vaddr=50000000 size=00010h (    16) load
I (373) boot: Loaded app from partition at offset 0x10000
I (374) boot: Disabling RNG early entropy source...
I (391) cpu_start: Pro cpu up.
I (391) cpu_start: Starting app cpu, entry point is 0x40375284
0x40375284: call_start_cpu1 at /home/yt/Desktop/esp-idf/components/esp_system/port/cpu_start.c:160

I (0) cpu_start: App cpu up.
I (405) cpu_start: Pro cpu start user code
I (405) cpu_start: cpu freq: 160000000
I (405) cpu_start: Application information:
I (408) cpu_start: Project name:     hello
I (412) cpu_start: App version:      1
I (417) cpu_start: Compile time:     Feb 29 2024 22:34:35
I (423) cpu_start: ELF file SHA256:  72bb16d5b1658308...
I (429) cpu_start: ESP-IDF:          v4.4.2
I (434) heap_init: Initializing. RAM available for dynamic allocation:
I (441) heap_init: At 3FC9E7F0 len 00041810 (262 KiB): D/IRAM
I (447) heap_init: At 3FCE0000 len 0000EE34 (59 KiB): STACK/DRAM
I (454) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (460) heap_init: At 600FE000 len 00002000 (8 KiB): RTCRAM
I (467) spi_flash: detected chip: gd
I (471) spi_flash: flash io: dio
I (475) sleep: Configure to isolate all GPIO pins in sleep state
I (482) sleep: Enable automatic switching of GPIO sleep configuration
I (489) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (10) lvgl_helpers: Display buffer size: 24000
I (10) lcd_panel.st7796u: Initialize Intel 8080 bus
I (20) lcd_panel.st7796u: here 1
I (20) lcd_panel.st7796u: here 2
I (20) lcd_panel.st7796u: Install LCD driver of st7796u
I (30) gpio: GPIO[9]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (40) gpio: GPIO[4]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (50) gpio: GPIO[5]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (60) gpio: GPIO[39]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (70) gpio: GPIO[40]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (80) gpio: GPIO[38]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (180) disp_backlight: Setting LCD backlight: 100%
I (180) lvgl_helpers: i80 master for d
[image]
[image]

XuNeo, 2024-03-01:
Please check if PNG decoder is enabled in your lv_conf.h

Knight-yang, 2024-03-01:
> Please check if PNG decoder is enabled in your lv_conf.h

yes,PNG decoder is enabled

kisvegabor, 2024-03-05:
Please enable [logging](https://docs.lvgl.io/master/porting/log.html) with INFO level.

Knight-yang, 2024-03-08:
> Please enable [logging](https://docs.lvgl.io/master/porting/log.html) with INFO level.

Hi,thanks your reply,and i enabled INFO level,but it didn't shows anything

[image]

This is the log
I (472) spiram: Read only data copied and mapped to SPIRAM
I (561) cpu_start: Pro cpu start user code
I (561) cpu_start: cpu freq: 240000000
I (561) cpu_start: Application information:
I (564) cpu_start: Project name:     hello
I (568) cpu_start: App version:      1
I (573) cpu_start: Compile time:     Mar  8 2024 15:36:33
I (579) cpu_start: ELF file SHA256:  8a88ab8e23f0e3f1...
I (585) cpu_start: ESP-IDF:          v4.4.2-dirty
I (590) heap_init: Initializing. RAM available for dynamic allocation:
I (597) heap_init: At 3FC969B8 len 00049648 (293 KiB): D/IRAM
I (604) heap_init: At 3FCE0000 len 0000EE34 (59 KiB): STACK/DRAM
I (610) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (617) heap_init: At 600FE000 len 00002000 (8 KiB): RTCRAM
I (623) spiram: Adding pool of 6720K of external SPI memory to heap allocator
I (631) spi_flash: detected chip: gd
I (635) spi_flash: flash io: qio
I (639) sleep: Configure to isolate all GPIO pins in sleep state
I (646) sleep: Enable automatic switching of GPIO sleep configuration
I (653) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (673) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (673) lvgl_helpers: Display buffer size: 24000
I (683) lcd_panel.st7796u: Initialize Intel 8080 bus
I (683) lcd_panel.st7796u: here 1
I (693) lcd_panel.st7796u: here 2
I (693) lcd_panel.st7796u: Install LCD driver of st7796u
I (703) gpio: GPIO[9]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (713) gpio: GPIO[4]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (723) gpio: GPIO[5]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (733) gpio: GPIO[39]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (743) gpio: GPIO[40]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (743) gpio: GPIO[38]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (853) disp_backlight: Setting LCD backlight: 100%
I (853) lvgl_helpers: i80 master for display
I (43113) event: FastStartFun
I (45113) action: change_stop_arm_flag\n\r

I (46113) action: check_tank_water:0
I (46113) action: xTimerStop
I (46113) action: xTimerStart
I (47113) action: GUI_Switch_tank_wait\n\r

I (76113) action: abnomal_finish\n\r

I (89193) event: FastStartFun
I (90193) action: change_stop_arm_flag\n\r

I (91193) action: check_tank_water:0
I (91193) action: xTimerStop 
I (91193) action: xTimerStart
I (92193) action: GUI_Switch_tank_wait\n\r

I (121193) action: abnomal_finish\n\r

and screens shows like this,the font is vaguely visible, but the background is not shown,and shows no data

Knight-yang, 2024-03-08:
> > Please enable [logging](https://docs.lvgl.io/master/porting/log.html) with INFO level.
> 
> Hi,thanks your reply,and i enabled INFO level,but it didn't shows anything
> 
> ![07b081d86b2d8faf3282520e4f70d48](https://private-user-images.githubusercontent.com/68107467/311160405-a40df682-751a-4096-906e-cadf26423e52.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDk4ODU0NjAsIm5iZiI6MTcwOTg4NTE2MCwicGF0aCI6Ii82ODEwNzQ2Ny8zMTExNjA0MDUtYTQwZGY2ODItNzUxYS00MDk2LTkwNmUtY2FkZjI2NDIzZTUyLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAzMDglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMzA4VDA4MDYwMFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTQzYzcyZmQwNWQxYjgxMTMzMjg5NDFlOWY3MzhiZjcxMmM2ZDY3YjhiMTI1YmNhZjYzYzViODY1ZmYzNzdmODQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.PGAWT6y-Jul66y5q2E_DeXVHIVgaLhG_2RtuuDMYmus)
> 
> This is the log I (472) spiram: Read only data copied and mapped to SPIRAM I (561) cpu_start: Pro cpu start user code I (561) cpu_start: cpu freq: 240000000 I (561) cpu_start: Application information: I (564) cpu_start: Project name: hello I (568) cpu_start: App version: 1 I (573) cpu_start: Compile time: Mar 8 2024 15:36:33 I (579) cpu_start: ELF file SHA256: 8a88ab8e23f0e3f1... I (585) cpu_start: ESP-IDF: v4.4.2-dirty I (590) heap_init: Initializing. RAM available for dynamic allocation: I (597) heap_init: At 3FC969B8 len 00049648 (293 KiB): D/IRAM I (604) heap_init: At 3FCE0000 len 0000EE34 (59 KiB): STACK/DRAM I (610) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM I (617) heap_init: At 600FE000 len 00002000 (8 KiB): RTCRAM I (623) spiram: Adding pool of 6720K of external SPI memory to heap allocator I (631) spi_flash: detected chip: gd I (635) spi_flash: flash io: qio I (639) sleep: Configure to isolate all GPIO pins in sleep state I (646) sleep: Enable automatic switching of GPIO sleep configuration I (653) cpu_start: Starting scheduler on PRO CPU. I (0) cpu_start: Starting scheduler on APP CPU. I (673) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations I (673) lvgl_helpers: Display buffer size: 24000 I (683) lcd_panel.st7796u: Initialize Intel 8080 bus I (683) lcd_panel.st7796u: here 1 I (693) lcd_panel.st7796u: here 2 I (693) lcd_panel.st7796u: Install LCD driver of st7796u I (703) gpio: GPIO[9]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 I (713) gpio: GPIO[4]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 I (723) gpio: GPIO[5]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 I (733) gpio: GPIO[39]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 I (743) gpio: GPIO[40]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 I (743) gpio: GPIO[38]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 I (853) disp_backlight: Setting LCD backlight: 100% I (853) lvgl_helpers: i80 master for display I (43113) event: FastStartFun I (45113) action: change_stop_arm_flag\n\r
> 
> I (46113) action: check_tank_water:0 I (46113) action: xTimerStop I (46113) action: xTimerStart I (47113) action: GUI_Switch_tank_wait\n\r
> 
> I (76113) action: abnomal_finish\n\r
> 
> I (89193) event: FastStartFun I (90193) action: change_stop_arm_flag\n\r
> 
> I (91193) action: check_tank_water:0 I (91193) action: xTimerStop I (91193) action: xTimerStart I (92193) action: GUI_Switch_tank_wait\n\r
> 
> I (121193) action: abnomal_finish\n\r
> 
> and screens shows like this,the font is vaguely visible, but the background is not shown,and shows no data

[image]

kisvegabor, 2024-03-11:
Have you also set a [logger](https://docs.lvgl.io/master/porting/log.html) callback as described here?

Knight-yang, 2024-03-12:
> Have you also set a [logger](https://docs.lvgl.io/master/porting/log.html) callback as described here?

because nodata screens  shows randomly，so i have no ideal to set  logger callback in some where

closed 2024-04-08 by lvgl-bot