lvgl/lvgl #5774: Unhandled Exception - circ_calc from /draw/sw/lv_draw_sw_mask

sukesh-ak, 2024-02-29:
### LVGL version

v9.0.0

### What happened?

MCU: `ESP32 / 4MB Flash / 2MB PSRAM`
TFT: `ST7796 SPI`, Touch: `FT5x06 CTP`

I have a UI created like a dial for setting temperature (in this case).
When this UI shows up, MCU crashes and restarts.
Stack provided below. I tried enabling `trace` option in `lv_conf.h` but that ends up in a different issue.

```bash

Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400e8d57  PS      : 0x00060330  A0      : 0x800e9468  A1      : 0x3ffcef40  
0x400e8d57: circ_calc_aa4 at D:/Dev/ProjectName/managed_components/lvgl__lvgl/src/draw/sw/lv_draw_sw_mask.c:1122

A2      : 0x3ffb2e54  A3      : 0x00000000  A4      : 0x00000046  A5      : 0x00000238
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x3ffcef4c  
A10     : 0x00000046  A11     : 0x3ffcef48  A12     : 0x00000118  A13     : 0x400e513c
0x400e513c: dispatch at D:/Dev/ProjectName/managed_components/lvgl__lvgl/src/draw/sw/lv_draw_sw.c:282

A14     : 0x3ffb789c  A15     : 0x00000001  SAR     : 0x0000001b  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000000  LBEG    : 0x400891b0  LEND    : 0x400891cc  LCOUNT  : 0x00000000
0x400891b0: memcpy at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/memcpy.S:175
0x400891cc: memcpy at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/memcpy.S:197

Backtrace: 0x400e8d54:0x3ffcef40 0x400e9465:0x3ffcefa0 0x400e5fad:0x3ffcefc0 0x400e61df:0x3ffcf030 0x400e50ab:0x3ffcf140 0x400e511e:0x3ffcf160 0x400e516d:0x3ffcf180 0x400e3ae2:0x3ffcf1a0 0x400e3b10:0x3ffcf1d0 0x400e3b8e:0x3ffcf1f0 0x400e4a37:0x3ffcf210 0x400dc182:0x3ffcf290 0x400dc99f:0x3ffcf330 0x4011c4f9:0x3ffcf370 0x400dd733:0x3ffcf390 0x400dd7a6:0x3ffcf3b0 0x400e2600:0x3ffcf3f0 0x400e28f1:0x3ffcf510 0x400e26b6:0x3ffcf5c0 0x400e28f1:0x3ffcf6e0 0x400e26b6:0x3ffcf790 0x400e28f1:0x3ffcf8b0 0x400e2af4:0x3ffcf960 0x400e2c42:0x3ffcf980 0x400e2dc3:0x3ffcf9a0 0x400e2ec6:0x3ffcf9e0 0x400e306f:0x3ffcfa00 0x400eee49:0x3ffcfa20 0x400eef03:0x3ffcfa40 0x400db7a0:0x3ffcfa60 0x4008e2cd:0x3ffcfa80
0x400e8d54: circ_calc_aa4 at D:/Dev/ProjectName/managed_components/lvgl__lvgl/src/draw/sw/lv_draw_sw_mask.c:1122
0x400e9465: lv_draw_sw_mask_radius_init at D:/Dev/ProjectName/managed_components/lvgl__lvgl/src/draw/sw/lv_draw_sw_mask.c:351
0x400e5fad: shadow_draw_corner_buf at D:/Dev/ProjectName/managed_components/lvgl__lvgl/src/draw/sw/lv_draw_sw_box_shadow.c:585
0x400e61df: lv_draw_sw_box_shadow at D:/Dev/ProjectName/managed_components/lvgl__lvgl/src/draw/sw/lv_draw_sw_box_shadow.c:120
0x400e50ab: execute_drawing at D:/Dev/ProjectName/managed_components/lvgl__lvgl/src/draw/sw/lv_draw_sw.c:363
0x400e511e: execute_drawing_unit at D:/Dev/ProjectName/managed_components/lvgl__lvgl/src/draw/sw/lv_draw_sw.c:245
0x400e516d: dispatch at D:/Dev/ProjectName/managed_components/lvgl__lvgl/src/draw/sw/lv_draw_sw.c:314
0x400e3ae2: lv_draw_dispatch_layer at D:/Dev/ProjectName/managed_components/lvgl__lvgl/src/draw/lv_draw.c:257
0x400e3b10: lv_draw_dispatch at D:/Dev/ProjectName/managed_components/lvgl__lvgl/src/draw/lv_draw.c:161
0x400e3b8e: lv_draw_finalize_task_creation at D:/Dev/ProjectName/managed_components/lvgl__lvgl/src/draw/lv_draw.c:138
0x400e4a37: lv_draw_rect at D:/Dev/ProjectName/managed_components/lvgl__lvgl/src/draw/lv_draw_rect.c:146
0x400dc182: lv_obj_draw at D:/Dev/ProjectName/managed_components/lvgl__lvgl/src/core/lv_obj.c:458
0x400dc99f: lv_obj_event at D:/Dev/ProjectName/managed_components/lvgl__lvgl/src/core/lv_obj.c:711
0x4011c4f9: lv_obj_event_base at D:/Dev/ProjectName/managed_components/lvgl__lvgl/src/core/lv_obj_event.c:86
0x400dd733: event_send_core at D:/Dev/ProjectName/managed_components/lvgl__lvgl/src/core/lv_obj_event.c:339
0x400dd7a6: lv_obj_send_event at D:/Dev/ProjectName/managed_components/lvgl__lvgl/src/core/lv_obj_event.c:64
0x400e2600: lv_obj_redraw at D:/Dev/ProjectName/managed_components/lvgl__lvgl/src/core/lv_refr.c:110
0x400e28f1: refr_obj at D:/Dev/ProjectName/managed_components/lvgl__lvgl/src/core/lv_refr.c:887
0x400e26b6: lv_obj_redraw at D:/Dev/ProjectName/managed_components/lvgl__lvgl/src/core/lv_refr.c:161 (discriminator 3)
0x400e28f1: refr_obj at D:/Dev/ProjectName/managed_components/lvgl__lvgl/src/core/lv_refr.c:887
0x400e26b6: lv_obj_redraw at D:/Dev/ProjectName/managed_components/lvgl__lvgl/src/core/lv_refr.c:161 (discriminator 3)
0x400e28f1: refr_obj at D:/Dev/ProjectName/managed_components/lvgl__lvgl/src/core/lv_refr.c:887
0x400e2af4: refr_obj_and_children at D:/Dev/ProjectName/managed_components/lvgl__lvgl/src/core/lv_refr.c:784
0x400e2c42: refr_area_part at D:/Dev/ProjectName/managed_components/lvgl__lvgl/src/core/lv_refr.c:717
0x400e2dc3: refr_area at D:/Dev/ProjectName/managed_components/lvgl__lvgl/src/core/lv_refr.c:650 (discriminator 2)
0x400e2ec6: refr_invalid_areas at D:/Dev/ProjectName/managed_components/lvgl__lvgl/src/core/lv_refr.c:566
0x400e306f: _lv_display_refr_timer at D:/Dev/ProjectName/managed_components/lvgl__lvgl/src/core/lv_refr.c:374
0x400eee49: lv_timer_exec at D:/Dev/ProjectName/managed_components/lvgl__lvgl/src/misc/lv_timer.c:300 (discriminator 2)
0x400eef03: lv_timer_handler at D:/Dev/ProjectName/managed_components/lvgl__lvgl/src/misc/lv_timer.c:105
0x400db7a0: lvgl_port_task at D:/Dev/ProjectName/components/esp_lvgl_port/src/lvgl9/esp_lvgl_port.c:154
0x4008e2cd: vPortTaskWrapper at C:/espressif/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:162

```

### How to reproduce?

Base project is the same as here but the current project is for commercial customer. 
https://github.com/sukesh-ak/IDF-ESP_LCD-LVGL

Looking for suggestions on next steps to nail the issue.

FASTSHIFT, 2024-03-01:
You can try to reproduce the problem on the [simulator](https://github.com/lvgl/lv_port_pc_eclipse).

sukesh-ak, 2024-03-01:
> You can try to reproduce the problem on the [simulator](https://github.com/lvgl/lv_port_pc_eclipse).

That's the problem. It works in the simulator (was designed there first) but everything else is different there including available resources.

FASTSHIFT, 2024-03-02:
Have you tried adjusting the stack size? Try to troubleshoot system problems first. Or you can provide minimal code snippets that reproduce the problem.

kisvegabor, 2024-03-05:
To add a few points to @FASTSHIFT's comment
- Increasing LV_MEM_SIZE
- Besides the example, please also attach your lv_conf.h

sukesh-ak, 2024-03-06:
> Have you tried adjusting the stack size? Try to troubleshoot system problems first. Or you can provide minimal code snippets that reproduce the problem.

@FASTSHIFT 
As an ex-diagnostic engineer who was writing debugger for analyzing memory dump, increasing stack size was the first thing I tried. Was running out of buffer memory with that change.

Then I had to move the lvgl buffer to PSRAM and doubled the stack size but didn't make any difference to this specific issue.

@kisvegabor 
I doubled `LV_MEM_SIZE` from `64*1024U` to `128*1024U`.
This makes the difference and its not crashing anymore.

I have used a simpler UI instead of dial to work around it for now due to the urgency for completion. I will update in a week and will provide a sample if it's still an issue.

Thanks for your response @FASTSHIFT & @kisvegabor

sukesh-ak, 2024-03-13:
Will close this for now.
Will test again if I get time but crash had stopped with the increase in `LV_MEM_SIZE`.