lvgl/lvgl #5785: Trailing \n ignored in `span`

s-hadinger, 2024-03-03:
### LVGL version

v9.0.0

### What happened?

When setting text of `lv_span`, a carriage return is accepted to get to the next line, except when it's the last character of the span.

However it works well if the `\n` is the first character of a span.

Is it by design?

### How to reproduce?

Example of Micropython in LVGL9 simulator:

```python

# Initialize

import display_driver
import lvgl as lv

# Create a button with a label

scr = lv.obj()
label = lv.label(scr)

text = lv.spangroup(scr)
text.set_pos(10,10)
text.set_size(100,100)
text.set_style_bg_color(lv.color_hex(0x000088), 0)

sp1 = text.new_span()
sp1.set_text("Hello\n")
sp2 = text.new_span()
sp2.set_text(" World")

sp1 = text.new_span()
sp1.set_text("Hello")
sp2 = text.new_span()
sp2.set_text("\n World")

lv.screen_load(scr)
```

Output is:
[image]

kisvegabor, 2024-03-05:
It's not by design, but it's really a bug. 

Now I'm trying to catch up with the comments and issues so I can look into it only a little bit later. If you see where the problem could be, please send a PR.

C47D, 2024-03-11:
So, is it expected to have both `Hello World` in the same line? Or should they be spitted as the `Hello World` below?

PR #5957 (2024-03-27): fix(span): handle trailing newline
### Description of the feature or fix

Fixes #5785

Spans do not acknowledge a newline if it's the last char.

Example
```c
    lv_obj_t * obj = lv_obj_create(lv_screen_active());
    lv_obj_set_pos(obj, 10, 10);
    lv_obj_set_size(obj, LV_SIZE_CONTENT, LV_SIZE_CONTENT);
    lv_obj_set_style_bg_color(obj, lv_color_hex(0xcccccc), 0);

    lv_obj_t * spangroup = lv_spangroup_create(obj);
    lv_obj_set_size(spangroup, 300, 200);

    lv_span_set_text(lv_spangroup_new_span(spangroup), "Hello\n");
    lv_span_set_text(lv_spangroup_new_span(spangroup), " World"); [... cut]
closed 2024-03-27 by kisvegabor