lvgl/lvgl #5786: where is the meter?

xinsuinizhuan, 2024-03-04:
### Introduce the problem

In 8.3 i could find it. But in 9.0, i could not find

xinsuinizhuan, 2024-03-04:
I need meter operator, as 8.3, in src\extra\widgets\meter

xinsuinizhuan, 2024-03-04:
I find the scale，but it so hard to use. Not simple as meter int 8.3. Could and the meter to 9.0?

xinsuinizhuan, 2024-03-04:
could add meter back?

xinsuinizhuan, 2024-03-20:
why remove the meter, i think the meter is so easy to make, but the scale is hard to make, meter is static, but the scale is dynamic?

kisvegabor, 2024-03-20:
Scale is more modular an universal. But please show your use case about the lv_meter.

xinsuinizhuan, 2024-03-21:
this is a example，meter:

//Write codes screen
	ui->screen = lv_obj_create(NULL);
	lv_obj_set_size(ui->screen, 480, 272);
	lv_obj_set_scrollbar_mode(ui->screen, LV_SCROLLBAR_MODE_OFF);

	//Write style for screen, Part: LV_PART_MAIN, State: LV_STATE_DEFAULT.
	lv_obj_set_style_bg_opa(ui->screen, 236, LV_PART_MAIN|LV_STATE_DEFAULT);
	lv_obj_set_style_bg_color(ui->screen, lv_color_hex(0x00ff16), LV_PART_MAIN|LV_STATE_DEFAULT);
	lv_obj_set_style_bg_grad_dir(ui->screen, LV_GRAD_DIR_NONE, LV_PART_MAIN|LV_STATE_DEFAULT);

	//Write codes screen_meter_1
	ui->screen_meter_1 = lv_meter_create(ui->screen);
	// add scale screen_meter_1_scale_0
	lv_meter_scale_t *screen_meter_1_scale_0 = lv_meter_add_scale(ui->screen_meter_1);
	lv_meter_set_scale_ticks(ui->screen_meter_1, screen_meter_1_scale_0, 41, 2, 10, lv_color_hex(0x00ff1d));
	lv_meter_set_scale_major_ticks(ui->screen_meter_1, screen_meter_1_scale_0, 8, 7, 18, lv_color_hex(0x30ff00), 10);
	lv_meter_set_scale_range(ui->screen_meter_1, screen_meter_1_scale_0, 1, 100, 320, 110);

	// add arc for screen_meter_1_scale_0
	ui->screen_meter_1_scale_0_arc_0 = lv_meter_add_arc(ui->screen_meter_1, screen_meter_1_scale_0, 10, lv_color_hex(0x00ffff), 10);
	lv_meter_set_indicator_start_value(ui->screen_meter_1, ui->screen_meter_1_scale_0_arc_0, 6);
	lv_meter_set_indicator_end_value(ui->screen_meter_1, ui->screen_meter_1_scale_0_arc_0, 35);

	// add scale line for screen_meter_1_scale_0
	lv_meter_indicator_t *screen_meter_1_scale_0_scaleline_0;
	screen_meter_1_scale_0_scaleline_0 = lv_meter_add_scale_lines(ui->screen_meter_1, screen_meter_1_scale_0, lv_color_hex(0xffff00), lv_color_hex(0xff00ff), true, 0);
	lv_meter_set_indicator_start_value(ui->screen_meter_1, screen_meter_1_scale_0_scaleline_0, 17);
	lv_meter_set_indicator_end_value(ui->screen_meter_1, screen_meter_1_scale_0_scaleline_0, 29);

	// add needle line for screen_meter_1_scale_0.
	ui->screen_meter_1_scale_0_ndline_0 = lv_meter_add_needle_line(ui->screen_meter_1, screen_meter_1_scale_0, 5, lv_color_hex(0xff6500), -10);
	lv_meter_set_indicator_value(ui->screen_meter_1, ui->screen_meter_1_scale_0_ndline_0, 7);

	// add needle line for screen_meter_1_scale_0.
	ui->screen_meter_1_scale_0_ndline_1 = lv_meter_add_needle_line(ui->screen_meter_1, screen_meter_1_scale_0, 5, lv_color_hex(0x000000), -10);
	lv_meter_set_indicator_value(ui->screen_meter_1, ui->screen_meter_1_scale_0_ndline_1, 10);

	// add needle line for screen_meter_1_scale_0.
	ui->screen_meter_1_scale_0_ndline_2 = lv_meter_add_needle_line(ui->screen_meter_1, screen_meter_1_scale_0, 5, lv_color_hex(0x000000), -10);
	lv_meter_set_indicator_value(ui->screen_meter_1, ui->screen_meter_1_scale_0_ndline_2, 10);
	lv_obj_set_pos(ui->screen_meter_1, 42, 56);
	lv_obj_set_size(ui->screen_meter_1, 166, 166);
	lv_obj_add_flag(ui->screen_meter_1, LV_OBJ_FLAG_CLICKABLE);
	lv_obj_add_flag(ui->screen_meter_1, LV_OBJ_FLAG_SCROLLABLE);

	//Write style for screen_meter_1, Part: LV_PART_MAIN, State: LV_STATE_DEFAULT.
	lv_obj_set_style_bg_opa(ui->screen_meter_1, 255, LV_PART_MAIN|LV_STATE_DEFAULT);
	lv_obj_set_style_bg_color(ui->screen_meter_1, lv_color_hex(0xffd400), LV_PART_MAIN|LV_STATE_DEFAULT);
	lv_obj_set_style_bg_grad_dir(ui->screen_meter_1, LV_GRAD_DIR_NONE, LV_PART_MAIN|LV_STATE_DEFAULT);
	lv_obj_set_style_radius(ui->screen_meter_1, 100, LV_PART_MAIN|LV_STATE_DEFAULT);
	lv_obj_set_style_border_width(ui->screen_meter_1, 5, LV_PART_MAIN|LV_STATE_DEFAULT);
	lv_obj_set_style_border_opa(ui->screen_meter_1, 255, LV_PART_MAIN|LV_STATE_DEFAULT);
	lv_obj_set_style_border_color(ui->screen_meter_1, lv_color_hex(0x2195f6), LV_PART_MAIN|LV_STATE_DEFAULT);
	lv_obj_set_style_border_side(ui->screen_meter_1, LV_BORDER_SIDE_FULL | LV_BORDER_SIDE_TOP, LV_PART_MAIN|LV_STATE_DEFAULT);
	lv_obj_set_style_shadow_width(ui->screen_meter_1, 4, LV_PART_MAIN|LV_STATE_DEFAULT);
	lv_obj_set_style_shadow_color(ui->screen_meter_1, lv_color_hex(0xff0088), LV_PART_MAIN|LV_STATE_DEFAULT);
	lv_obj_set_style_shadow_opa(ui->screen_meter_1, 255, LV_PART_MAIN|LV_STATE_DEFAULT);
	lv_obj_set_style_shadow_spread(ui->screen_meter_1, 4, LV_PART_MAIN|LV_STATE_DEFAULT);
	lv_obj_set_style_shadow_ofs_x(ui->screen_meter_1, 4, LV_PART_MAIN|LV_STATE_DEFAULT);
	lv_obj_set_style_shadow_ofs_y(ui->screen_meter_1, 4, LV_PART_MAIN|LV_STATE_DEFAULT);

	//Write style for screen_meter_1, Part: LV_PART_TICKS, State: LV_STATE_DEFAULT.
	lv_obj_set_style_text_color(ui->screen_meter_1, lv_color_hex(0xff0000), LV_PART_TICKS|LV_STATE_DEFAULT);
	lv_obj_set_style_text_font(ui->screen_meter_1, &lv_font_montserratMedium_12, LV_PART_TICKS|LV_STATE_DEFAULT);
	lv_obj_set_style_text_opa(ui->screen_meter_1, 255, LV_PART_TICKS|LV_STATE_DEFAULT);

	//Write style for screen_meter_1, Part: LV_PART_INDICATOR, State: LV_STATE_DEFAULT.
	lv_obj_set_style_bg_opa(ui->screen_meter_1, 255, LV_PART_INDICATOR|LV_STATE_DEFAULT);
	lv_obj_set_style_bg_color(ui->screen_meter_1, lv_color_hex(0x000000), LV_PART_INDICATOR|LV_STATE_DEFAULT);
	lv_obj_set_style_bg_grad_dir(ui->screen_meter_1, LV_GRAD_DIR_NONE, LV_PART_INDICATOR|LV_STATE_DEFAULT);

xinsuinizhuan, 2024-03-21:
> this is a example，meter, writed by 8.13:
> 
> //Write codes screen ui->screen = lv_obj_create(NULL); lv_obj_set_size(ui->screen, 480, 272); lv_obj_set_scrollbar_mode(ui->screen, LV_SCROLLBAR_MODE_OFF);
> 
> ```
> //Write style for screen, Part: LV_PART_MAIN, State: LV_STATE_DEFAULT.
> lv_obj_set_style_bg_opa(ui->screen, 236, LV_PART_MAIN|LV_STATE_DEFAULT);
> lv_obj_set_style_bg_color(ui->screen, lv_color_hex(0x00ff16), LV_PART_MAIN|LV_STATE_DEFAULT);
> lv_obj_set_style_bg_grad_dir(ui->screen, LV_GRAD_DIR_NONE, LV_PART_MAIN|LV_STATE_DEFAULT);
> 
> //Write codes screen_meter_1
> ui->screen_meter_1 = lv_meter_create(ui->screen);
> // add scale screen_meter_1_scale_0
> lv_meter_scale_t *screen_meter_1_scale_0 = lv_meter_add_scale(ui->screen_meter_1);
> lv_meter_set_scale_ticks(ui->screen_meter_1, screen_meter_1_scale_0, 41, 2, 10, lv_color_hex(0x00ff1d));
> lv_meter_set_scale_major_ticks(ui->screen_meter_1, screen_meter_1_scale_0, 8, 7, 18, lv_color_hex(0x30ff00), 10);
> lv_meter_set_scale_range(ui->screen_meter_1, screen_meter_1_scale_0, 1, 100, 320, 110);
> 
> // add arc for screen_meter_1_scale_0
> ui->screen_meter_1_scale_0_arc_0 = lv_meter_add_arc(ui->screen_meter_1, screen_meter_1_scale_0, 10, lv_color_hex(0x00ffff), 10);
> lv_meter_set_indicator_start_value(ui->screen_meter_1, ui->screen_meter_1_scale_0_arc_0, 6);
> lv_meter_set_indicator_end_value(ui->screen_meter_1, ui->screen_meter_1_scale_0_arc_0, 35);
> 
> // add scale line for screen_meter_1_scale_0
> lv_meter_indicator_t *screen_meter_1_scale_0_scaleline_0;
> screen_meter_1_scale_0_scaleline_0 = lv_meter_add_scale_lines(ui->screen_meter_1, screen_meter_1_scale_0, lv_color_hex(0xffff00), lv_color_hex(0xff00ff), true, 0);
> lv_meter_set_indicator_start_value(ui->screen_meter_1, screen_meter_1_scale_0_scaleline_0, 17);
> lv_meter_set_indicator_end_value(ui->screen_meter_1, screen_meter_1_scale_0_scaleline_0, 29);
> 
> // add needle line for screen_meter_1_scale_0.
> ui->screen_meter_1_scale_0_ndline_0 = lv_meter_add_needle_line(ui->screen_meter_1, screen_meter_1_scale_0, 5, lv_color_hex(0xff6500), -10);
> lv_meter_set_indicator_value(ui->screen_meter_1, ui->screen_meter_1_scale_0_ndline_0, 7);
> 
> // add needle line for screen_meter_1_scale_0.
> ui->screen_meter_1_scale_0_ndline_1 = lv_meter_add_needle_line(ui->screen_meter_1, screen_meter_1_scale_0, 5, lv_color_hex(0x000000), -10);
> lv_meter_set_indicator_value(ui->screen_meter_1, ui->screen_meter_1_scale_0_ndline_1, 10);
> 
> // add needle line for screen_meter_1_scale_0.
> ui->screen_meter_1_scale_0_ndline_2 = lv_meter_add_needle_line(ui->screen_meter_1, screen_meter_1_scale_0, 5, lv_color_hex(0x000000), -10);
> lv_meter_set_indicator_value(ui->screen_meter_1, ui->screen_meter_1_scale_0_ndline_2, 10);
> lv_obj_set_pos(ui->screen_meter_1, 42, 56);
> lv_obj_set_size(ui->screen_meter_1, 166, 166);
> lv_obj_add_flag(ui->screen_meter_1, LV_OBJ_FLAG_CLICKABLE);
> lv_obj_add_flag(ui->screen_meter_1, LV_OBJ_FLAG_SCROLLABLE);
> 
> //Write style for screen_meter_1, Part: LV_PART_MAIN, State: LV_STATE_DEFAULT.
> lv_obj_set_style_bg_opa(ui->screen_meter_1, 255, LV_PART_MAIN|LV_STATE_DEFAULT);
> lv_obj_set_style_bg_color(ui->screen_meter_1, lv_color_hex(0xffd400), LV_PART_MAIN|LV_STATE_DEFAULT);
> lv_obj_set_style_bg_grad_dir(ui->screen_meter_1, LV_GRAD_DIR_NONE, LV_PART_MAIN|LV_STATE_DEFAULT);
> lv_obj_set_style_radius(ui->screen_meter_1, 100, LV_PART_MAIN|LV_STATE_DEFAULT);
> lv_obj_set_style_border_width(ui->screen_meter_1, 5, LV_PART_MAIN|LV_STATE_DEFAULT);
> lv_obj_set_style_border_opa(ui->screen_meter_1, 255, LV_PART_MAIN|LV_STATE_DEFAULT);
> lv_obj_set_style_border_color(ui->screen_meter_1, lv_color_hex(0x2195f6), LV_PART_MAIN|LV_STATE_DEFAULT);
> lv_obj_set_style_border_side(ui->screen_meter_1, LV_BORDER_SIDE_FULL | LV_BORDER_SIDE_TOP, LV_PART_MAIN|LV_STATE_DEFAULT);
> lv_obj_set_style_shadow_width(ui->screen_meter_1, 4, LV_PART_MAIN|LV_STATE_DEFAULT);
> lv_obj_set_style_shadow_color(ui->screen_meter_1, lv_color_hex(0xff0088), LV_PART_MAIN|LV_STATE_DEFAULT);
> lv_obj_set_style_shadow_opa(ui->screen_meter_1, 255, LV_PART_MAIN|LV_STATE_DEFAULT);
> lv_obj_set_style_shadow_spread(ui->screen_meter_1, 4, LV_PART_MAIN|LV_STATE_DEFAULT);
> lv_obj_set_style_shadow_ofs_x(ui->screen_meter_1, 4, LV_PART_MAIN|LV_STATE_DEFAULT);
> lv_obj_set_style_shadow_ofs_y(ui->screen_meter_1, 4, LV_PART_MAIN|LV_STATE_DEFAULT);
> 
> //Write style for screen_meter_1, Part: LV_PART_TICKS, State: LV_STATE_DEFAULT.
> lv_obj_set_style_text_color(ui->screen_meter_1, lv_color_hex(0xff0000), LV_PART_TICKS|LV_STATE_DEFAULT);
> lv_obj_set_style_text_font(ui->screen_meter_1, &lv_font_montserratMedium_12, LV_PART_TICKS|LV_STATE_DEFAULT);
> lv_obj_set_style_text_opa(ui->screen_meter_1, 255, LV_PART_TICKS|LV_STATE_DEFAULT);
> 
> //Write style for screen_meter_1, Part: LV_PART_INDICATOR, State: LV_STATE_DEFAULT.
> lv_obj_set_style_bg_opa(ui->screen_meter_1, 255, LV_PART_INDICATOR|LV_STATE_DEFAULT);
> lv_obj_set_style_bg_color(ui->screen_meter_1, lv_color_hex(0x000000), LV_PART_INDICATOR|LV_STATE_DEFAULT);
> lv_obj_set_style_bg_grad_dir(ui->screen_meter_1, LV_GRAD_DIR_NONE, LV_PART_INDICATOR|LV_STATE_DEFAULT);
> ```
> 
> ![图片](https://private-user-images.githubusercontent.com/40679769/314787720-60403e99-9808-4288-b7e2-b27f2bca624c.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTA5ODM5ODcsIm5iZiI6MTcxMDk4MzY4NywicGF0aCI6Ii80MDY3OTc2OS8 [... cut ...] dvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAzMjElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMzIxVDAxMTQ0N1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTg4YmM2ZDk0YmRmNjlmNWU5NTA4NWZlZGEyMzNjMWIyYzM4ZGI5OGEyMDcwZDdiOTMyMGExN2Q0YWM2OTRjYTUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.4KmMH1rmI-ih29T3mgyN8NiV1-RDz-V5Cbscn4EmmC0)

how to realize the meter by 9.0? In 9.0, the scale example is simple

xinsuinizhuan, 2024-03-21:
and how about this issue?
https://github.com/lvgl/lv_port_pc_visual_studio/issues/69

xinsuinizhuan, 2024-03-21:
this is a scale meter, example, but could not work?

void scle_meter()
{
	lv_obj_t* scale3;

	lv_style_t scale3_section1_main_style;
	lv_style_t scale3_section1_indicator_style;
	lv_style_t scale3_section1_tick_style;
	lv_style_t scale3_section2_main_style;
	lv_style_t scale3_section2_indicator_style;
	lv_style_t scale3_section2_tick_style;
	lv_style_t scale3_section3_main_style;
	lv_style_t scale3_section3_indicator_style;
	lv_style_t scale3_section3_tick_style;

	scale3 = lv_scale_create(lv_screen_active());
	lv_scale_set_mode(scale3, LV_SCALE_MODE_ROUND_INNER);
	lv_scale_set_post_draw(scale3, true);
	lv_obj_set_width(scale3, LV_PCT(100));

	//scale3 = create_scale_box(parent, "Network Speed", "Low speed", "Normal Speed", "High Speed");

	lv_scale_set_range(scale3, 0, 60);
	lv_scale_set_total_tick_count(scale3, 21);
	lv_scale_set_major_tick_every(scale3, 4);
	lv_obj_set_style_length(scale3, 10, LV_PART_ITEMS);
	lv_obj_set_style_length(scale3, 30, LV_PART_INDICATOR);
	lv_scale_set_angle_range(scale3, 225);
	lv_scale_set_rotation(scale3, 135);

	lv_style_init(&scale3_section1_main_style);
	lv_style_set_arc_width(&scale3_section1_main_style, 8);
	lv_style_set_arc_color(&scale3_section1_main_style, lv_palette_main(LV_PALETTE_RED));

	lv_style_init(&scale3_section1_indicator_style);
	lv_style_set_line_width(&scale3_section1_indicator_style, 4);
	lv_style_set_line_color(&scale3_section1_indicator_style, lv_palette_darken(LV_PALETTE_RED, 2));

	lv_style_init(&scale3_section1_tick_style);
	lv_style_set_line_width(&scale3_section1_tick_style, 12);
	lv_style_set_line_color(&scale3_section1_tick_style, lv_palette_darken(LV_PALETTE_RED, 2));

	lv_style_init(&scale3_section2_main_style);
	lv_style_set_arc_width(&scale3_section2_main_style, 8);
	lv_style_set_arc_color(&scale3_section2_main_style, lv_palette_main(LV_PALETTE_BLUE));

	lv_style_init(&scale3_section2_indicator_style);
	lv_style_set_line_width(&scale3_section2_indicator_style, 4);
	lv_style_set_line_color(&scale3_section2_indicator_style, lv_palette_darken(LV_PALETTE_BLUE, 2));

	lv_style_init(&scale3_section2_tick_style);
	lv_style_set_line_width(&scale3_section2_tick_style, 4);
	lv_style_set_line_color(&scale3_section2_tick_style, lv_palette_darken(LV_PALETTE_BLUE, 2));

	lv_style_init(&scale3_section3_main_style);
	lv_style_set_arc_width(&scale3_section3_main_style, 8);
	lv_style_set_arc_color(&scale3_section3_main_style, lv_palette_main(LV_PALETTE_GREEN));

	lv_style_init(&scale3_section3_indicator_style);
	lv_style_set_line_width(&scale3_section3_indicator_style, 4);
	lv_style_set_line_color(&scale3_section3_indicator_style, lv_palette_darken(LV_PALETTE_GREEN, 2));

	lv_style_init(&scale3_section3_tick_style);
	lv_style_set_line_width(&scale3_section3_tick_style, 4);
	lv_style_set_line_color(&scale3_section3_tick_style, lv_palette_darken(LV_PALETTE_GREEN, 2));

	//lv_obj_add_event_cb(scale3, scale3_delete_event_cb, LV_EVENT_DELETE, NULL);

	lv_scale_section_t* section;
	section = lv_scale_add_section(scale3);
	lv_scale_section_set_range(section, 10, 20);
	lv_scale_section_set_style(section, LV_PART_MAIN, &scale3_section1_main_style);
	lv_scale_section_set_style(section, LV_PART_INDICATOR, &scale3_section1_indicator_style);
	lv_scale_section_set_style(section, LV_PART_ITEMS, &scale3_section1_tick_style);
	section = lv_scale_add_section(scale3);
	lv_scale_section_set_range(section, 20, 40);
	lv_scale_section_set_style(section, LV_PART_MAIN, &scale3_section2_main_style);
	lv_scale_section_set_style(section, LV_PART_INDICATOR, &scale3_section2_indicator_style);
	lv_scale_section_set_style(section, LV_PART_ITEMS, &scale3_section2_tick_style);

	section = lv_scale_add_section(scale3);
	lv_scale_section_set_range(section, 40, 60);
	lv_scale_section_set_style(section, LV_PART_MAIN, &scale3_section3_main_style);
	lv_scale_section_set_style(section, LV_PART_INDICATOR, &scale3_section3_indicator_style);
	lv_scale_section_set_style(section, LV_PART_ITEMS, &scale3_section3_tick_style);

	LV_IMG_DECLARE(img_demo_widgets_needle);
	lv_obj_t* needle = lv_image_create(scale3);
	lv_image_set_src(needle, &img_demo_widgets_needle);
	lv_image_set_pivot(needle, 3, 4);
	lv_obj_align(needle, LV_ALIGN_CENTER, 47, -2);

	lv_obj_t* mbps_label = lv_label_create(scale3);
	lv_label_set_text(mbps_label, "-");
	//lv_obj_add_style(mbps_label, &style_title, 0);

	lv_obj_t* mbps_unit_label = lv_label_create(scale3);
	lv_label_set_text(mbps_unit_label, "Mbps");

	lv_anim_t a;
	lv_anim_init(&a);
	lv_anim_set_values(&a, 10, 60);
	lv_anim_set_repeat_count(&a, LV_ANIM_REPEAT_INFINITE);
	//lv_anim_set_exec_cb(&a, scale3_anim_cb);
	lv_anim_set_var(&a, scale3);
	lv_anim_set_duration(&a, 4100);
	lv_anim_set_playback_duration(&a, 800);
	lv_anim_start(&a);

	lv_obj_align(mbps_label, LV_ALIGN_TOP_MID, 10, lv_pct(55));
	lv_obj_align_to(mbps_unit_label, mbps_label, LV_ALIGN_OUT_RIGHT_BOTTOM, 10, 0);
}

kisvegabor, 2024-06-20:
I close this issue because there was no activity here for a while.

If you have remarks about this topic feel free to comment here and reopen the issue.

closed 2024-06-20 by kisvegabor

lorddevereux, 2024-06-27:
I think there are situations which the meter could achieve but scale cannot.

The classic Meter example of having multiple arcs inside each other [https://docs.lvgl.io/8.4/widgets/extra/meter.html](as in the 8.4 docs). I can't obviously see a way to achieve this effect with scale.

I am happy to be corrected if I'm missing something, but it would be useful if the docs could explain how to achieve it to assist with the migration.

kisvegabor, 2024-06-27:
To achieve something like this you can just create `lv_arc`s on an `lv_scale`, and set the arcs' angles and range accordingly. 
[image]

WebDust21, 2024-07-09:
And here I spent a day porting LVGL 9.1 into an ESP_Panel project, looking to base the project on the "latest and greatest LVGL"...but the project goal was specifically to utilize the "LV_METER" functionality identically to the above picture.  I was not aware that the LV_METER functionality had been deprecated in LVGL 9 until after getting it ported...
(I have 3 MPPT charge controllers, and wanted to be able to visualize their individual outputs with a multi-meter gauge.)

> To achieve something like this you can just create `lv_arc`s on an `lv_scale`, and set the arcs' angles and range accordingly. 
![image](https://private-user-images.githubusercontent.com/7599318/343738058-db90a703-efcf-4ad1-a7b0-570ba7cd5931.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjA0NzIxNTMsIm5iZiI6MTcyMDQ3MTg1MywicGF0aCI6Ii83NTk5MzE4LzM0MzczODA1OC1kYjkwYTcwMy1lZmNmLTRhZDEtYTdiMC01NzBiYTdjZDU5MzEucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDcwOCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA3MDhUMjA1MDUzWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9YjA0MGRlZGQ4MjYzYjQ1NmZlMWM0NWU1ZDJiMWE4ZTc2ZmJlNjVjNzcwOTdkNmQyMTIxYTcwNTUwYzI0MDJhNyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.Pzrv5O7e3fmMcpa0QLpZaolBk0zuedK2_PejoU4ARH8)

So I'm trying to achieve this.  Starting with a simple "lv_scale" based on the LVGL 9 example:
[image]
Then putting an "lv_arc" [scaled LV_PCT(100) in both axis] as a child of "lv_scale":
[image]

OK, looks like we need to find a way to size the "lv_scale" drawn meter down, so the "lv_arc" is outside the ticks (as per the reference screenshot from LVGL 8).

However, there seems to be a fundamentally missing functionality with "lv_scale" for this to be possible.
In the internal "lv_scale" function "scale_draw_main", I noticed that only the following parameters are utilized:
https://github.com/lvgl/lvgl/blob/d12f88fd95f937a67862c07d6b840a787c8a9ec4/src/widgets/scale/lv_scale.c#L752-L757

If I set padding with:
```
lv_obj_set_style_pad_all(scaleObject, 50, LV_PART_MAIN);
```
it has the desired drawing effect on the "lv_scale"--however, the "lv_arc" [scaled with LV_PCT(100)] _also_ scales down (despite padding not being set on "lv_scale"):
[image]

So I tried adjusting the only other setting utilized by "lv_scale":
```
lv_obj_set_style_border_width(scaleObject, 50, LV_PART_MAIN);
```
And inexplicably, the "lv_arc" gets scaled down, even while the outside line is drawn at full "lv_scale" bounds:
[image]

Testing further, if I scale the "lv_arc" with higher than 100% [i.e. LV_PCT(125)], it does go outside of the scaled-down "lv_scale"--which is an improvement, and definitely getting somewhat usable:
[image]
...but the ticks don't go through the "lv_arc" as per the reference benchmark screenshot.

Yes, the ticks' length can be adjusted, as found here:
https://github.com/lvgl/lvgl/blob/d12f88fd95f937a67862c07d6b840a787c8a9ec4/src/widgets/scale/lv_scale.c#L949-L955
...BUT they will lengthen INWARDS, not outwards.

If we set the "lv_scale" to _outwards_ (LV_SCALE_MODE_ROUND_OUTER), the ticks go the correct direction--**but** we can't separate the ticks' direction from the text labels' direction, as both are based on the "lv_scale" round outer/inner setting (i.e. "scale->mode").  
Ticks ->
https://github.com/lvgl/lvgl/blob/d12f88fd95f937a67862c07d6b840a787c8a9ec4/src/widgets/scale/lv_scale.c#L1066-L1079
and text ->
https://github.com/lvgl/lvgl/blob/d12f88fd95f937a67862c07d6b840a787c8a9ec4/src/widgets/scale/lv_scale.c#L655-L662
resulting in this:
[image]

My 2 cents: If "lv_scale" + "lv_arc" is supposed to be the replacement for "lv_meter", then at the very least, we need some way to specify the "ticks" to be OUTER, and the text to be "INNER".

WebDust21, 2024-07-09:
Aargh...if using a padding on the "lv_scale" to provide room for the arcs, and setting the "lv_arc"s size to >100%, the "lv_scale" becomes scrollable!

fairly easy to fix with
```
lv_obj_remove_flag(lv_scale_object, LV_OBJ_FLAG_SCROLLABLE);
```

kisvegabor, 2024-07-10:
I'm sorry that gave you hard times... The goal with replacing meter with scale was to improve flexibility but it seems still there is some work to do.

[This discussion](https://github.com/lvgl/lvgl/issues/6449) can be related. By specifying `LV_SCALE_MODE_ROUND_OUTER` and applying a negative `RADIAL_OFFSET` (as proposed in the linked issue) you could move the labels inwards. 

What do you think?

---

For the whole configuration maybe you should create the scale on the arc. This way the arc can be easily larger and the ticks will overlap the arc (if this is the intension).

WebDust21, 2024-07-10:
> [This discussion](https://github.com/lvgl/lvgl/issues/6449) can be related. By specifying `LV_SCALE_MODE_ROUND_OUTER` and applying a negative `RADIAL_OFFSET` (as proposed in the linked issue) you could move the labels inwards.
> 
> What do you think?

If it'll get the job done, it sounds like a solution to me!  The linked suggestions are considerably more powerful/flexible than simply separating labels and ticks from both referencing "scale->mode", and look like a solid way forwards.

WebDust21, 2024-07-10:
> For the whole configuration maybe you should create the scale on the arc.

Well, that becomes a bit problematic when there's more than one arc:
[image]

though not to say that this is trouble-free: I'm having a serious issue with the "lv_arc"s malfunctioning and getting into an internal dead loop if they don't like the input values (first seemed that out of range values caused this--despite the internal code gates on "lv_arc_set_value"--but now when trying to use all 3 arcs simultaneously, it's failing with in-range values.)  Separate issue, most likely due to project configuration; I'll research it further in issues and LVGL forum to see if there's a solution already found...
[image]

WebDust21, 2024-07-10:
above arc-draw issue -> https://forum.lvgl.io/t/lvgl-v9-memory-usage/14520
Apparently drawing arcs is highly memory intensive...had to increase the buffer to 128KB for it to stop locking up upon "lv_arc_set_value".

kisvegabor, 2024-07-11:
I've commented in https://forum.lvgl.io/t/lvgl-v9-memory-usage/14520/2 about the larger memory usage.