lvgl/lvgl #5790: LVGL V9.0 Tabbar right padding label with `LV_OBJ_FLAG_IGNORE_LAYOUT` affects layout and scroll

mr-sven, 2024-03-04:
### LVGL version

v9.0.0

### What happened?

I Have a tab view and I want to add a label or button to the right side of the tabs.
As soon as I align the label right of the last tab, the layout add its width and position to the layout so it gets scrollable even if the flag `LV_OBJ_FLAG_IGNORE_LAYOUT` is set.
The label on the left side does not affect this behavior.
[image]
[image]

As soon as the x value is 0, the container is not scrolling.
[image]

### How to reproduce?

```c
lv_obj_t* tabview = lv_tabview_create(lv_screen_active());
lv_tabview_set_tab_bar_size(tabview, 45);
lv_obj_t* tab1 = lv_tabview_add_tab(tabview, "Tab 1");
lv_obj_t* tab2 = lv_tabview_add_tab(tabview, "Tab 2");

lv_obj_t* tab_bar = lv_tabview_get_tab_bar(tabview);
lv_obj_set_style_pad_left(tab_bar, LV_HOR_RES / 4, 0);
lv_obj_set_style_pad_right(tab_bar, 50, 0);

lv_obj_t* label = lv_label_create(tab_bar);
lv_label_set_text(label, "Titlebar");
lv_obj_add_flag(label, LV_OBJ_FLAG_IGNORE_LAYOUT);
lv_obj_align(label, LV_ALIGN_LEFT_MID, -LV_HOR_RES / 4 + 20, 0);

label = lv_label_create(tab_bar);
lv_label_set_text(label, LV_SYMBOL_WIFI);
lv_obj_set_style_text_opa(label, LV_OPA_20, 0);
lv_obj_add_flag(label, LV_OBJ_FLAG_IGNORE_LAYOUT);
lv_obj_align(label, LV_ALIGN_RIGHT_MID, 40, 0); // with horizontal scroll
// lv_obj_align(label, LV_ALIGN_RIGHT_MID, 0, 0); // without horizontal scroll

label = lv_label_create(tab1);
lv_label_set_text(label, "Hello, I'm LVGL!");
lv_obj_align(label, LV_ALIGN_CENTER, 0, 0);
```

liamHowatt, 2024-03-14:
I see what you mean, @mr-sven,

This first snippet is where it decides whether to scroll. In this case `lv_obj_get_scroll_left` and `lv_obj_get_scroll_right` are of interest.
https://github.com/lvgl/lvgl/blob/d53b9145a23c1227f5958bf4b1948c755904e681/src/indev/lv_indev_scroll.c#L335-L357

<hr>

The below snippet has the definitions of `lv_obj_get_scroll_left` and `lv_obj_get_scroll_right`.

As you observed, it does not scroll left. That's because `lv_obj_get_scroll_left` returns at line 170 since the `tab_bar` (`obj` here) is not a `LV_BASE_DIR_RTL`, it is a `LV_BASE_DIR_LTR` (Left To RIght (?)). It returns 0 because `scroll.x` is zero since it hasn't scrolled yet.

Moving onto `lv_obj_get_scroll_right`, it does _not_ enter the `if` block on line 211 this time. In the `for` loop it iterates the four children and find the rightmost one which is the wifi label. It counts the wifi label because it is neither `LV_OBJ_FLAG_HIDDEN | LV_OBJ_FLAG_FLOATING`. It gets the `space_right`, which is the padding you added. The calculation on line 232 turns `child_res` into the amount that it overlaps with the padding. That's the value that gets returned and it tries to scroll that amount further into view.
https://github.com/lvgl/lvgl/blob/d53b9145a23c1227f5958bf4b1948c755904e681/src/core/lv_obj_scroll.c#L162-L240

<hr>

I personally don't think there's a bug here. @kisvegabor, what do you think? I think the RTL/LTR logic is objectively correct but kind of coincidental. It's a coincidence that it stops it from scrolling left. And then decision for it to scroll right makes sense because it's basically assuming anything in the padding area is "not fully in view" and needs to be scrolled into view.

## Current Fix For User Code
Apply the `LV_OBJ_FLAG_FLOATING` flag to the WiFi label.

mr-sven, 2024-03-14:
Hi, ok that works, I was confused about `LV_OBJ_FLAG_IGNORE_LAYOUT` I thought it is the correct setting for that.
For me it is fixed.

closed 2024-03-20 by mr-sven