lvgl/lvgl #5791: We are hiring!

kisvegabor, 2024-03-05:
### Introduce the problem

-

### Proposal

Hey Guys,

I’m excited to share that we are extending our team with these two new positions. Of course, in both cases it’s a huge plus, if you are already and LVGL contributor. :slightly_smiling_face:

**Project Maintainer**
As a Project Maintainer, you will play a crucial role in steering the direction of the LVGL library. Your responsibilities will include managing the project’s roadmap, overseeing contributions, and ensuring the library remains cutting-edge and user-friendly. If you have a strong background in software project management, a passion for open-source projects, and excellent communication skills, we want to hear from you! Learn more [here](https://blog.lvgl.io/2024-03-04/hiring-maintainer).

**OpenGL Expert**
We are also seeking an OpenGL Expert to enrich our team’s expertise in advanced graphics rendering techniques. This role involves optimizing LVGL’s use of OpenGL for enhanced performance and capabilities, developing new features, and contributing to the library’s evolution. If you have deep knowledge of OpenGL, experience in embedded systems, and a desire to push the boundaries of what’s possible in embedded graphics, let’s connect. Learn more [here](https://blog.lvgl.io/2024-03-04/hiring-opengl).

Joining LVGL means becoming part of a dynamic, supportive community dedicated to making high-quality embedded graphics accessible to everyone. We offer a collaborative environment where your work will have a direct impact on millions of devices worldwide.

If you are ready to take on exciting challenges and contribute to the future of embedded graphics, check out the detailed job descriptions and apply through the links provided. We’re excited to welcome new talents to our team and together, continue shaping the future of embedded graphics!

I’m looking forward to hear from you! :blush:

Cuixudong, 2024-03-18:
I really want to participate in project maintenance, but I can't even use github proficiently, it seems like a joke
[image]

kisvegabor, 2024-03-18:
If you are interested in contributing to LVGL, we can help you t get started. After creating a few PR it won't be that scary :slightly_smiling_face:

kisvegabor, 2024-03-18:
Thank you for all the applicants! I'm so happy that we have received many great CVs and met amazing people!  Although we could hire only one person now, I'm glad to share that @liamHowatt  just joined as a project maintainer. See more info [here](https://github.com/lvgl/lvgl/issues/5044#issuecomment-2004659674).

For the OpenGL position we have found an interested person internally.

closed 2024-03-18 by kisvegabor

Cuixudong, 2024-03-19:
It's nice to see someone join you