lvgl/lvgl #5792: flex: LV_FLEX_ALIGN_SPACE_BETWEEN unexpected behavior with single item

romdt, 2024-03-05:
### LVGL version

v9.0.0

### What happened?

```c
lv_obj_set_flex_flow(row, LV_FLEX_FLOW_ROW);
lv_obj_set_flex_align(row, LV_FLEX_ALIGN_SPACE_BETWEEN, LV_FLEX_ALIGN_CENTER, LV_FLEX_ALIGN_CENTER);
```
I have expected that a single item would be aligned to the left
Am I correct?

[image]

Thanks

### How to reproduce?

```c
lv_obj_t* item(lv_obj_t* parent, lv_color_t color, lv_coord_t width)
{
    auto obj = lv_obj_create(parent);
    lv_obj_set_size(obj, width, width);
    lv_obj_set_style_radius(obj, LV_RADIUS_CIRCLE, 0);
    lv_obj_set_style_bg_color(obj, color, 0);
    return obj;
}

void flex_bug()
{
    lv_obj_t* row = lv_obj_create(lv_screen_active());
    lv_obj_set_size(row, 200, 100);
    lv_obj_set_style_pad_all(row, 10, 0);

    lv_obj_set_flex_flow(row, LV_FLEX_FLOW_ROW);
    lv_obj_set_flex_align(row, LV_FLEX_ALIGN_SPACE_BETWEEN, LV_FLEX_ALIGN_CENTER, LV_FLEX_ALIGN_CENTER);

    item(row, lv_color_black(), 30);
    item(row, lv_color_black(), 60);
}
```

liamHowatt, 2024-03-14:
> `LV_FLEX_ALIGN_SPACE_BETWEEN`: items are evenly distributed in the track: first item is on the start line, last item on the end line. Not applies to track_cross_place.

Good point, @romdt. A single item should be on the left according to the documentation. Now it's just a question of whether the docs should be corrected to say "unless there's only 1 item" or if the behavior should be changed to match the docs.

A change something like this
```diff
diff --git a/src/layouts/flex/lv_flex.c b/src/layouts/flex/lv_flex.c
index 168f9de90..1dd9198b6 100644
--- a/src/layouts/flex/lv_flex.c
+++ b/src/layouts/flex/lv_flex.c
@@ -472,7 +472,6 @@ static void place_content(lv_flex_align_t place, int32_t max_size, int32_t conte
 {
     if(item_cnt <= 1) {
         switch(place) {
-            case LV_FLEX_ALIGN_SPACE_BETWEEN:
             case LV_FLEX_ALIGN_SPACE_AROUND:
             case LV_FLEX_ALIGN_SPACE_EVENLY:
                 place = LV_FLEX_ALIGN_CENTER;
@@ -492,7 +491,7 @@ static void place_content(lv_flex_align_t place, int32_t max_size, int32_t conte
             *start_pos += max_size - content_size;
             break;
         case LV_FLEX_ALIGN_SPACE_BETWEEN:
-            *gap = (int32_t)(max_size - content_size) / (int32_t)(item_cnt - 1);
+            if (item_cnt > 1) *gap = (int32_t)(max_size - content_size) / (int32_t)(item_cnt - 1);
             break;
         case LV_FLEX_ALIGN_SPACE_AROUND:
             *gap += (int32_t)(max_size - content_size) / (int32_t)(item_cnt);
```

kisvegabor, 2024-03-19:
I've checked that in HTML it also align the single item  to `START`. (space evenly and space around aligns to the center)
See here: https://codepen.io/kisvegabor/pen/xxegYbx

@liamHowatt please send a PR with the fix.

kisvegabor, 2024-04-08:
Fixed by https://github.com/lvgl/lvgl/pull/5915

@liamHowatt you can add 
```
fixes #5915
```
in the description of the PRs to make GitHub close the related issues automatically when the PR is merged. See [this](https://github.com/lvgl/lvgl/pull/5804) as an example.

closed 2024-04-08 by kisvegabor