lvgl/lvgl #5803: lv_imagebutton_set_src failed to display image without left & right elements

PierreRambaud, 2024-03-05:
### LVGL version

v9.0.0-212-g481dd7b91

### What happened?

When trying to use `lv_imagebutton_set_src` without left & right elements, it fails to render the image.
As far as I know, both parameters are not mandatory, I'm able to use NULL inside.

https://github.com/lvgl/lvgl/assets/1462701/81e9f93b-bca0-4fac-a270-256cc6c0c814

### How to reproduce?

```c
  lv_obj_t* button = lv_imagebutton_create(parent);
  lv_obj_add_style(button, &style_default_text, 0);
  lv_obj_set_width(button, 150);

  lv_imagebutton_set_src(button, LV_IMAGEBUTTON_STATE_RELEASED, NULL, img_src, NULL);
  lv_imagebutton_set_src(button, LV_IMAGEBUTTON_STATE_PRESSED, NULL, img_pressed_src, NULL);

  lv_obj_t* label = lv_label_create(button);
  lv_label_set_text(label, title);
  lv_obj_align(label, LV_ALIGN_CENTER, 10, 4);
```

PierreRambaud, 2024-03-12:
I still have the problem with latest changes, can you show me which commit should have fix  this please?

kisvegabor, 2024-03-12:
I hoped this: https://github.com/lvgl/lvgl/pull/5719

Please send the required images too to reproduce the issue.

cc @liamHowatt, could you help us the debug it?

liamHowatt, 2024-03-12:
@kisvegabor in my debugging I couldn't find anything wrong with left and right NULL or not, but I did have issues with drawing mid. I opened a PR for a fix that may be the problem here.

#5849

@PierreRambaud please try my fix to see if it solves your issue. The commit is 2845d5c99b9385c8f5f2c38d673f7c88b4fdf89f

PierreRambaud, 2024-03-13:
> @kisvegabor in my debugging I couldn't find anything wrong with left and right NULL or not, but I did have issues with drawing mid. I opened a PR for a fix that may be the problem here.
>
> #5849 [...]

Everything seems to be ok with your latest fix! I can use NULL for left and right parameter :pray:

kisvegabor, 2024-03-28:
Should be resolved by https://github.com/lvgl/lvgl/pull/5849.

closed 2024-03-28 by kisvegabor