lvgl/lvgl #5805: how to set span group frame border?

xinsuinizhuan, 2024-03-06:
### LVGL version

9.0.1

### What happened?

I create a spangroup, but all set is set he header information, no set to frame border, and show no frame boder,as fllower:
[image]

same as the lable widget

xinsuinizhuan, 2024-03-22:
I want a group like this, Group have the border, and have the it's title, and group all it's self widgets:
[image]
But the real is only have the title, have no border:
[image]

kisvegabor, 2024-03-25:
Can you just place a label on top of a border? You also need to set the `bg_color` of the label to the same color as the background of the border to make it covert the border.

closed 2024-04-29 by lvgl-bot