lvgl/lvgl #5809: Possible double free error in lv_obj_remove_style()

Ju1He1, 2024-03-06:
### LVGL version

v9.0.0

### What happened?

Hey guys,
on our target build we are running into an assertion which tells me that I'm trying to free an unallocated block.
On our target wee are wrapping all malloc/free calls to the FreeRTOS malloc/free implementation because FreeRTOS heap_4 is a very performant heap implementation.
It also has some benefits like checking which free() that the block which is going to be freed is actually allocated. This is why FreeRTOS is asserting within lv_obj_remove_style().

This looks to me like a double free issue or freeing some uninitialized memory.  Maybe some pointer was not set to NULL after calling free() ?

Here are some screenshots

[image]

[image]

[image]

[image]

[image]

### How to reproduce?

To reproduce you would need to wrap your malloc()/free() calls to the freertos functions, because their malloc/free implementation is the only one I know of which keeps track of all allocated and freed memory blocks. Neither the lvgl malloc() nor the stdlib malloc() does do that.

We have trimmed down our application to this

```
#include <FreeRTOS/FreeRTOS.h>
#include <FreeRTOS/task.h>

static TaskHandle_t mg_Task = nullptr;

static const std::string mg_strTaskName = "Task";

/**
 * @brief Execute FreeRTOS Task
 * @param pvParameters
*/
static void RunInTask(void* pvParameters);

///Sends a shutdown request to all tasks
static void sendShutdownRequestToAllTasks();

///Block until all user tasks are finnished
static void waitForAllTasksToFinnish();

main()
{
// launch freertos
 configASSERT(pdTRUE == xTaskCreate(
    RunInTask,       /* Function that implements the task. */
    mg_strTaskName.c_str(),          /* Text name for the task. */
    configMINIMAL_STACK_SIZE,      /* Stack size in words, not bytes. */
    (void*)1,    /* Parameter passed into the task. */
    5,/* Priority at which the task is created. */
    &mg_Task));      /* Used to pass out the created task's handle. */
   vTaskStartScheduler();
}

static void RunInTask(void* pvParameters)
{
  lv_init();
  auto* pDisplay = lv_display_create(240, 320);
 for(;;)
  { 
    //task loop.... you should not get here
  }
}
```

do you have an idea where the error occurs?

Ju1He1, 2024-03-06:
here are the picture sin higher resolution
[image]
[image]
[image]

liamHowatt, 2024-03-10:
Hey @Ju1He1 ,

I think I know what the problem may be.

May I ask how you wrapped `malloc` and `free`? I think there's a chance the `malloc` and `free` calls are using the heap_4 implementations as you expect, however the `realloc` call is using a different allocator, such as LVGL's builtin or clib. Therefore, `realloc` is producing pointers from an unexpected heap and then they're being passed to heap_4's free. The pointers from `realloc` are alien to heap_4 so it asserts, as it should.

It seems that FreeRTOS's allocator interface doesn't have `realloc`. portable.h defines `pvPortMalloc`, `pvPortCalloc`, and `vPortFree`, but not `pvPortRealloc`. Since `realloc` can't directly be implemented in terms of `malloc`, and LVGL makes extensive use of `realloc` there may need to be a larger discussion about how to make LVGL use an allocator implementation that doesn't provide `realloc`, such as FreeRTOS, if there's a strong case for it.

For now, please confirm yourself whether an unexpected `realloc` is being called.

Ju1He1, 2024-03-11:
hey @liamHowatt  thx for your feedback
You did point out some interesting hints.
We are indeed only wrapping malloc and free (not realloc)
```
set(wrap_list "malloc" "free")

  # append wrap list to linker flags todo use target_link_options instead
  foreach(wrap ${wrap_list})
    set(CMAKE_EXE_LINKER_FLAGS "${CMAKE_EXE_LINKER_FLAGS} -Wl,--wrap=${wrap}")
  endforeach()
```

```

void* __wrap_malloc(size_t size)
{
  bool bInHandlerMode = inHandlerMode();
  //freertos Malloc is not ISR safe!
  configASSERT(false == bInHandlerMode);
  void* pRes = pvPortMalloc(size);
   return pRes;
}

/* wrapper for stdlib call free */
void __wrap_free(void* p)
{
  bool bInHandlerMode = inHandlerMode();
  //freertos free is not ISR safe!
  configASSERT(false == bInHandlerMode);

  return vPortFree(p);
}
```

After replacing the malloc/free within lv_mem_core_clib.c with the proper freertos malloc/free I can reproduce the issue in the simulation as well.
I did replace the realloc as well, now I'm running in a nullptr. Not sure If this is an issue on my side. I will have to further evaluate. Thx for your ffedback I'll close this issue for now until I did some research.

closed 2024-03-11 by Ju1He1