lvgl/lvgl #5810: compile error when #define LV_USE_OS   LV_OS_FREERTOS

BahuMan, 2024-03-06:
### LVGL version

9.0.1

### What happened?

In Visual Studio + PlatformIO, when I change lv_conf.h to set **LV_USE_OS**  to **LV_OS_FREERTOS**, I get the following compile error:

```
In file included from C:/Users/Bart/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32s3/include/freertos/include/freertos/portable.h:51,
                 from C:/Users/Bart/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32s3/include/freertos/include/freertos/FreeRTOS.h:63,
                 from lib/lvgl/src/osal/lv_freertos.h:26,
                 from lib/lvgl/src/osal/lv_os.h:30,
                 from lib/lvgl/src/osal/lv_freertos.c:15:
lib/lvgl/src/osal/lv_freertos.c: In function 'prvCheckMutexInit':
C:/Users/Bart/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32s3/include/freertos/port/xtensa/include/freertos/portmacro.h:476:53: error: too few arguments to function 'vPortEnterCritical'
 #define portENTER_CRITICAL(mux)                     vPortEnterCritical(mux)
                                                     ^~~~~~~~~~~~~~~~~~
C:/Users/Bart/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32s3/include/freertos/include/freertos/task.h:222:35: note: in expansion of macro 'portENTER_CRITICAL'
 #define taskENTER_CRITICAL( x )   portENTER_CRITICAL( x )
                                   ^~~~~~~~~~~~~~~~~~
lib/lvgl/src/osal/lv_freertos.c:339:9: note: in expansion of macro 'taskENTER_CRITICAL'
         taskENTER_CRITICAL();
         ^~~~~~~~~~~~~~~~~~
C:/Users/Bart/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32s3/include/freertos/port/xtensa/include/freertos/portmacro.h:576:51: note: declared here 
 static inline void __attribute__((always_inline)) vPortEnterCritical(portMUX_TYPE *mux)

``` 

The problem seems to be in  lib/lvgl/src/osal/lv_freertos.c:339:9 because taskENTER_CRITICAL() is called without mutex argument.

I'm using the espressif32 platform.

```
Processing esp32-s3-devkitc-1 (platform: espressif32; board: esp32-s3-devkitc-1; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-s3-devkitc-1.html
PLATFORM: Espressif 32 (6.5.0) > Espressif ESP32-S3-DevKitC-1-N8 (8 MB QD, No PSRAM)
HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
DEBUG: Current (esp-builtin) On-board (esp-builtin) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.20014.231204 (2.0.14)
 - tool-esptoolpy @ 1.40501.0 (4.5.1)
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5   
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 35 compatible libraries
Scanning dependencies...
Dependency Graph
|-- TFT_eSPI @ 2.5.43
|-- lvgl @ 9.0.1-dev

```

[lv_conf.zip](https://github.com/lvgl/lvgl/files/14512478/lv_conf.zip)

### How to reproduce?

My full lv_conf.zp is attached for reference.

My main.cpp is only a skeleton file so far:
```

#include <Arduino.h>
#define COLOR_DEPTH 16

#include <TFT_eSPI.h>
TFT_eSPI    tft = TFT_eSPI();
//#include <../lv_conf.h>
#include "lvgl.h"

lv_display_t *display;

void setup() {
  lv_init();
  display = lv_display_create(296, 240);
}

void loop() {
}

```

HanesSciarrone, 2024-03-08:
Hi @BahuMan, I hope you are well.

After investigating a little more in detail I found that the problem is that the FreeRTOS used by Espressif takes into account the  multiple CPU usage and for that they change the definitions of `taskENTER_CRITICAL` and `taskEXIT_CRITICAL`.

Here is a brief explanation found in the Espressif portmacro.h file.

```
/*
Modifications to portENTER_CRITICAL:

The original portENTER_CRITICAL only disabled the ISRs. This is enough for single-CPU operation: by 
disabling the interrupts, there is no task switch so no other tasks can meddle in the data, and because
interrupts are disabled, ISRs can't corrupt data structures either.

For multiprocessing, things get a bit more hairy. First of all, disabling the interrupts doesn't stop
the tasks or ISRs on the other processors meddling with our CPU. For tasks, this is solved by adding
a spinlock to the portENTER_CRITICAL macro. A task running on the other CPU accessing the same data will
spinlock in the portENTER_CRITICAL code until the first CPU is done.

For ISRs, we now also need muxes: while portENTER_CRITICAL disabling interrupts will stop ISRs on the same
CPU from meddling with the data, it does not stop interrupts on the other cores from interfering with the
data. For this, we also use a spinlock in the routines called by the ISR, but these spinlocks
do not disable the interrupts (because they already are).

This all assumes that interrupts are either entirely disabled or enabled. Interrupr priority levels
will break this scheme.
*/
```

Currently LVGL is ported to the FreeRTOS standard [(FreeRTOS API)](https://www.freertos.org/taskENTER_CRITICAL_taskEXIT_CRITICAL.html) so the API provided by LVGL won't be useful in your case. I suggest to set `LV_OS_NONE`  and use the API provided by Espressif.

Regards
Hanes Nahuel Sciarrone

mr-sven, 2024-03-08:
Hi,
I commented this in issue #5515. I would prefer to implement the mutex behavior via a multicore compiler macro. Maybe more µC using more than one core and run into the same problem.

And I'm not a fan of double implementation, so if LVGL implements FreeRTOS task handling, why not use it.
Regards

BahuMan, 2024-03-08:
So would this be a feature request rather than a bug report?

kisvegabor, 2024-03-08:
> I would prefer to implement the mutex behavior via a multicore compiler macro.

If I understand correctly this is a custom modification by Espressif and an other multi-core vendor might come up with a different solution.

IMO we can have `LV_OS_FREERTOS` for the normal FreeRTOS and `LV_OS_FREERTOS_ESP` for the ESP version. 

What do you think?

HanesSciarrone, 2024-03-08:
Hi @kisvegabor 

Sounds like an excellent idea to me. I'll work on it. If you can @BahuMan I would like to ask you if you could test my changes when finished  because I don't have an development environment for ESP. I can share with you @BahuMan the branch where I'll make  the changes.

Regards
Hanes

kisvegabor, 2024-03-09:
@HanesSciarrone do you have esp dev board at home and only the software environment is missing?

BahuMan, 2024-03-09:
I'm happy with a new LV_OS_FREERTOS_ESP and I'm happy to test the changes! I have several ESP32 & ESP8266 lying around.
Since I'm pretty new with lvgl, could you give me some pointers on what kind of scenario I need to test where I actually need the mutex?

Perhaps point me to an existing example where the mutex is used?

HanesSciarrone, 2024-03-10:
Sorry for the delay @kisvegabor and @BahuMan . 

Yes @kisvegabor, I've and ESP32, I don't remember which ont exactly but I've one. To test the implementation I need to prepare the development environment because I've never developed with the ESP32 and I must try to add the FreeRTOS of Espressif. Anyway I would like to prepare the development environment because the ESP32 is used a lot with LVGL and it could be a starting point for other issues.

Regarding testing @BahuMan. The first thing I would like to achieve is a successful compilation with the Espressif of FreeRTOS in your environment. After that we can test it with an FreeRTOS and LVGL implementation.

I'll try to have the changes ready for this Monday in a new branch and I'll share them with you.

mr-sven, 2024-03-11:
In ESP-IDF are samples available: https://github.com/espressif/esp-idf/tree/master/examples/peripherals/lcd

mr-sven, 2024-03-11:
I've currently a working setup with a CYD (cheap yellow display), I can test also the implementation.

HanesSciarrone, 2024-03-12:
Hi @BahuMan and @mr-sven hope you are well, I've finished the changes to fix the compile error. The repository is: https://github.com/HanesSciarrone/lvgl.git.

Please could you test it in your project?. Also, please note that the changes are in `feat/freertos_esp` branch. There is a new parameter in lv_conf_template.h called LV_OS_FREERTOS_ESP, this is the parameter you need to set in LV_USE_OS.

HanesSciarrone, 2024-03-12:
@kisvegabor In the meantime I'll configure my development environment for ESP32 for future issues in this platform.

BahuMan, 2024-03-12:
### First test successful!

I see your changes in lv_freertos.c and the project compiles without errors if I #define LV_USE_OS   LV_OS_FREERTOS_ESP
The small project I have so far was deployed succesfully to an ESP32-S3 with TFT screen and shows the layout I expected.

Ideally, I should find a scenario to cause that code to activate and verify that the mutex you've chosen is valid at runtime, but I don't know lvgl (or esp flavor freertos) well enough.

**_Should I close the issue?_**

mr-sven, 2024-03-12:
I'm using raw IDF environment without wrappers like platformio or arduino

BahuMan, 2024-03-12:
[src.zip](https://github.com/lvgl/lvgl/files/14574981/src.zip)
[lv_conf.zip](https://github.com/lvgl/lvgl/files/14574993/lv_conf.zip)

I'm guessing you don't need the platformio.ini

mr-sven, 2024-03-12:
I see you are calling the `lv_timer_handler` by hand in loop, afaik. this should do the freertos implementation by itself.

mr-sven, 2024-03-12:
Here is my test branch https://github.com/mr-sven/esp32-8048S050C/tree/lvgl-test the main branch contains implementation without the inbuild freertos task.

mr-sven, 2024-03-12:
I thought about it. For my point of view the compile error is fixed, so that can be merged and the issue that the FreeRTOS Task doesn't work can be tracked in issue #5515

kisvegabor, 2024-03-13:
@HanesSciarrone, yes, please send a PR. It will be easier to review the changes there.

HanesSciarrone, 2024-03-13:
@kisvegabor in the PR one check was cancelled and I don't know if I have permission to rerun it. Could you run again to have the all check green?

kisvegabor, 2024-03-19:
The PR is merged, I'm closing this issue. Thank you for your contribution!

closed 2024-03-19 by kisvegabor