lvgl/lvgl #5815: lv_obj_refresh_style gets slower for every new object

Viatorus, 2024-03-08:
### LVGL version

v8.3.10

### What happened?

We have noticed, that the creation of objects gets slower over time for every new object created. We have a view where many objects  (~2500) are created at once and even on a Linux workstation it takes 400 ms (UI hangs) to just create these items. The drawing itself fast as usually.

I ran perf over it and saw, that `lv_obj_refresh_style` called in [`lv_obj_class_init_obj`](https://github.com/lvgl/lvgl/blob/8d453b57052a2c6991e40a0d3a4c66cc7f778b9e/src/core/lv_obj_class.c#L105) get's slower over time.

[image]

I even tried it by just creating 2500 screens `lv_obj_create(nullptr)`. The result is the same.

Any idea how this can be optimized?

XuNeo, 2024-03-08:
Hi,
Interesting perf result! 
Have you tried using LVGL v9 to run same perf test? LVGL v9 has added style cache(LV_OBJ_STYLE_CACHE=1) to optimize style finding process. 

Could you try it and post the result? Thank you.

Viatorus, 2024-03-08:
Result are very similar (9 is a bit slower).

| Objects | 8.3 | 9.x (master) |
|---------|-----|-----|
| 1000    | 6 ms   | 7 ms  |
| 10000   | 111 ms | 130 ms |
| 20000   | 219 ms | 226 ms |
|         |  [image]   |   [image]  |

<details>

<summary>Code</summary>

```cpp
#include "lvgl/lvgl.h"

#include <chrono>
#include <iostream>

#define DISP_HOR_RES 800
#define DISP_VER_RES 600

int main() {
    lv_init();

#if LVGL_VERSION_MAJOR == 8
    static lv_disp_draw_buf_t draw_buf;
    static lv_color_t buf1[DISP_HOR_RES * DISP_VER_RES / 10];
    lv_disp_draw_buf_init(&draw_buf, buf1, NULL, DISP_HOR_RES * DISP_VER_RES / 10);

    static lv_disp_drv_t disp_drv;
    lv_disp_drv_init(&disp_drv);
    disp_drv.draw_buf = &draw_buf;
    lv_disp_drv_register(&disp_drv);
#else
    lv_display_t *display = lv_display_create(DISP_HOR_RES, DISP_VER_RES);
    static lv_color_t buf1[DISP_HOR_RES * DISP_VER_RES / 10];
    lv_display_set_buffers(display, buf1, NULL, sizeof(buf1), LV_DISPLAY_RENDER_MODE_PARTIAL);
#endif

    for (size_t i = 0; i < 20; i++) {
        auto start = std::chrono::steady_clock::now();

        for (size_t j = 0; j < 1000; j++) {
            lv_obj_create(nullptr);
        }

        auto end = std::chrono::steady_clock::now();

        std::cout << i << ": " << std::chrono::duration_cast<std::chrono::milliseconds>(end - start).count() << '\n';
    }

    return 0;
}
```

</details>

Viatorus, 2024-03-08:
I found the function `lv_display_enable_invalidation` setting and used it to set display invalidation to false. This reduces the time for 20000 objects to:
* 158 ms (8.3)
* 159 ms (9.x)

Still, there is a linear increasement.

For me it is still not clear, why every new object has to touch all other objects on creation...

Viatorus, 2024-03-08:
Now with 20'000 objects instead of screens:

| Screens | 8.3 | 9.x (master) |
|---------|-----|-----|
| 1000    | 6 ms   | 7 ms  |
| 10000   | 96 ms | 50 ms |
| 20000   | 195 ms | 85 ms |
|         |  
[image]
   |  
[image]
 |

<details>

<summary>Code</summary>

```cpp
#include "lvgl/lvgl.h"

#include <chrono>
#include <iostream>

#define DISP_HOR_RES 800
#define DISP_VER_RES 600

int main() {
    lv_init();

#if LVGL_VERSION_MAJOR == 8
    static lv_disp_draw_buf_t draw_buf;
    static lv_color_t buf1[DISP_HOR_RES * DISP_VER_RES / 10];
    lv_disp_draw_buf_init(&draw_buf, buf1, NULL, DISP_HOR_RES * DISP_VER_RES / 10);

    static lv_disp_drv_t disp_drv;
    lv_disp_drv_init(&disp_drv);
    disp_drv.draw_buf = &draw_buf;
    lv_disp_drv_register(&disp_drv);
#else
    lv_display_t *display = lv_display_create(DISP_HOR_RES, DISP_VER_RES);
    static lv_color_t buf1[DISP_HOR_RES * DISP_VER_RES / 10];
    lv_display_set_buffers(display, buf1, NULL, sizeof(buf1), LV_DISPLAY_RENDER_MODE_PARTIAL);
#endif

    auto screen = lv_obj_create(NULL);

    for (size_t i = 0; i < 20; i++) {
        auto start = std::chrono::steady_clock::now();

        for (size_t j = 0; j < 1000; j++) {
            lv_obj_create(screen);
        }

        auto end = std::chrono::steady_clock::now();

        std::cout << i << ": " << std::chrono::duration_cast<std::chrono::milliseconds>(end - start).count() << '\n';
    }

    return 0;
}
```

</details>

V9 is more than twice as fast (maybe because of LV_OBJ_STYLE_CACHE?) But why this linear increase? (looking into ... `lv_obj_get_index`)

Viatorus, 2024-03-08:
So I think I found the "issues".

First, if an object does have many objects (like in my second test) the linear search in [`lv_obj_get_index`](https://github.com/lvgl/lvgl/blob/8d453b57052a2c6991e40a0d3a4c66cc7f778b9e/src/core/lv_obj_tree.c#L425) is the bottleneck because the newly created object is always the last in the list. The function is used during creation because of apply the default theme [here](https://github.com/lvgl/lvgl/blob/8d453b57052a2c6991e40a0d3a4c66cc7f778b9e/src/themes/default/lv_theme_default.c#L758).

But in my application I don't use any default theme. BUT I had activated `LV_USE_ASSERT_OBJ`, which in turn, iterates over all objects of a screen, every time lv_obj_invalidate is called, which later calls obj_valid_child and so on...

I know, that the define of this assert says it is slow, but I never had any performance issues before.

XuNeo, 2024-03-09:
Thank you for the throughput analysis, the new result looks really promising!

To conclude,
1. The linear search is the bottleneck when there are really many objects.
- For creating many screens (`lv_obj_create(NULL)`, the [lv_obj_get_disp](https://github.com/lvgl/lvgl/blob/8d453b57052a2c6991e40a0d3a4c66cc7f778b9e/src/core/lv_obj_tree.c#L311) linear search will slow down.
- For creating normal obj (`lv_obj_create(screen)`), the  [lv_obj_get_index](https://github.com/lvgl/lvgl/blob/8d453b57052a2c6991e40a0d3a4c66cc7f778b9e/src/core/lv_obj_tree.c#L425) linear search will slow down. It's used for applying theme.
2. `LV_USE_ASSERT_OBJ ` will iterates overall objects of a screen, thus also slow.
3. LVGL v9 style cache seems do the trick to make v9 faster than v8.

> I think it would be great to make LV_ASSERT_OBJ more configurable

Agree. By default `LV_ASSERT_OBJ` checks NULL pointer, and also add `LV_ASSERT_OBJ_CLASS`, `LV_ASSERT_OBJ_VALID` to check `lv_obj_has_class` and `lv_obj_is_valid`.

> V9 is more than twice as fast (maybe because of LV_OBJ_STYLE_CACHE?) 

Could you also test the 20000 obj creation time without LV_OBJ_STYLE_CACHE on v9? To make sure it's the style cache does the trick.

Viatorus, 2024-03-09:
> Could you also test the 20000 obj creation time without LV_OBJ_STYLE_CACHE on v9? To make sure it's the style cache does the trick.

In my previous test of v9 I didn't enabled `LV_OBJ_STYLE_CACHE` (wasn't aware that this is an opt-in). With `LV_OBJ_STYLE_CACHE == 1`, the execution seems to be bit faster (1 - 2 ms for 20000 objects) but nothing notable in my simple benchmark.

kisvegabor, 2024-03-12:
> For creating normal obj (lv_obj_create(screen)), the [lv_obj_get_index](https://github.com/lvgl/lvgl/blob/8d453b57052a2c6991e40a0d3a4c66cc7f778b9e/src/core/lv_obj_tree.c#L425) linear search will slow down. It's used for applying theme.

This one can be resolved by reordering [this](https://github.com/lvgl/lvgl/blob/8d453b57052a2c6991e40a0d3a4c66cc7f778b9e/src/themes/default/lv_theme_default.c#L758):
```c
     if(lv_obj_get_index(obj) == 0 && lv_obj_check_type(lv_obj_get_parent(obj), &lv_win_class))
```
to 
```c
     if(lv_obj_check_type(lv_obj_get_parent(obj), &lv_win_class) && lv_obj_get_index(obj) == 0)
```

> For creating many screens (lv_obj_create(NULL), the [lv_obj_get_disp](https://github.com/lvgl/lvgl/blob/8d453b57052a2c6991e40a0d3a4c66cc7f778b9e/src/core/lv_obj_tree.c#L311) linear search will slow down.

For this I'm thinking about having a dedicated `lv_screen` widget, which is just an `lv_obj` under the hood. But once it's a new widget type we can have a pointer in it to its display as well.

kisvegabor, 2024-03-27:
I've opened #5971 to resolve [this](https://github.com/lvgl/lvgl/issues/5815#issuecomment-1991651039).

PR #5971 (2024-04-10): perf(theme): optimize the order of function calls
fixes #5815

### Description of the feature or fix

A clear and concise description of what the bug or new feature is.

### Notes
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.
- Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant.
- Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if applicable.
- If you added new options to `lv_conf_template.h` run [lv_conf_internal_gen.py](https://github.com/lvgl/lvgl/blob/master/scripts/lv_conf_internal_gen.py) and update [... cut]
closed 2024-04-10 by FASTSHIFT