lvgl/lvgl #5816: Is it a requirement to call `lv_timer_handler` in its own thread in a RTOS project?

danielfu11, 2024-03-08:
### LVGL version

v8.3.11

### What happened?

Hi,

I was wondering if it is a requirement to call `lv_timer_handler` in its own thread in a RTOS project?

I see in the example in [docs](https://docs.lvgl.io/master/porting/os.html) that all the `lv_*` calls are in a separate thread. Is that necessary? Can I have all the `lv_*` calls plus `lv_timer_handler` in the same thread?

Also: are there any downsides to calling `lv_timer_handler` as frequently as possible?

XuNeo, 2024-03-08:
LVGL's all API are not thread safe, thus it's REQUIRED to call lv_* from same thread. The doc explains situations if you have to call lvgl API from another thread, that you have to make certain protection to make sure exclusive calls to lv_*.

> Also: are there any downsides to calling lv_timer_handler as frequently as possible?

No downsides but from practical point of view, if you have an RTOS running, then it's recommended to get the return value from `lv_timer_handler ` which is the time before next event lvgl should take action. Sleep for specified time that get from `lv_timer_handler` will yield the CPU to other threads, so CPU can run background tasks etc.

danielfu11, 2024-03-08:
That makes sense - thank you @XuNeo !

closed 2024-03-08 by danielfu11