lvgl/lvgl #5817: Running into memory allocator assert (prev_free field can not be null)

danielfu11, 2024-03-08:
### LVGL version

v8.3.11

### What happened?

Hi,

I am running into an tlsf_assert in remote_free_block where “prev_free field can not be null”. What does this mean and how can this condition be triggered? If the block is the head of the list, is it not expected that the prev_free points to NULL?

How can I go about debugging this issue?

I am running v8.3.11. Any pointers or help would be much appreciated

Thanks!

FASTSHIFT, 2024-03-10:
Can you reproduce the problem in the [simulator](https://github.com/lvgl/lv_port_pc_eclipse)? Or give a snippet of code that can reproduce the BUG?
This is usually a heap corruption caused by memory access out of bounds or use after memory release.

closed 2024-04-02 by lvgl-bot

fbiego, 2025-02-19:
This happens when you don't call `lv_init();`
Leaving this here incase someone has the same issue

danielfu11, 2025-02-19:
@fbiego That was not my issue, unfortunately.

The issue was an undocumented silicon bug in the imxrt106X MCU causing SDRAM accesses to fail. See forum thread here: https://community.nxp.com/t5/i-MX-RT-Crossover-MCUs/i-MXRT1060-SEMC-SDRAM-Data-Corruption/m-p/1172919