lvgl/lvgl #5820: Lvgl start cost too much time

ChowhoundWu, 2024-03-09:
### LVGL version

v8.20

### What happened?

I have created many complex interfaces using lvgl v8.2, which resulted in too many lv_obj objects and styles, causing long startup times. On a desktop computer, it takes about 3 seconds to start, while on a Linux development board, it takes around 30 seconds. Is there any good solution?

kisvegabor, 2024-03-09:
Hi,

Have you disabled the LV_USE_ASSERTs in lv_conf.h?

ChowhoundWu, 2024-03-09:
In my lv_conf.h it's ：

#define LV_USE_ASSERT_NULL          1   /*Check if the parameter is NULL. (Very fast, recommended)*/
#define LV_USE_ASSERT_MALLOC        1   /*Checks is the memory is successfully allocated or no. (Very fast, recommended)*/
#define LV_USE_ASSERT_STYLE         0   /*Check if the styles are properly initialized. (Very fast, recommended)*/
#define LV_USE_ASSERT_MEM_INTEGRITY 0   /*Check the integrity of `lv_mem` after critical operations. (Slow)*/
#define LV_USE_ASSERT_OBJ           0   /*Check the object's type and existence (e.g. not deleted). (Slow)*/

I assigned a lot of lv_obj, each lv_obj had its own style, and I had a lot of lv_obj (about 30?). As a page container, then 10 page containers were created and tested on a desktop, which took about 2.5 seconds, but it would take a lot of time to generate an obj on a Linux board even if the compilation is optimized to O2.

ChowhoundWu, 2024-03-09:
Also the pages are a lot, and these are all different from each other.

kisvegabor, 2024-03-09:
On Linux can you run a profiler to see what takes that much time? 

Can you share a small code snippet which is slow on your end?

I don't know if it's related, but it was already reported that textarea creation is slower than it ought to be.

ChowhoundWu, 2024-03-09:
FMotorControl = new TCardInverter(this);
    SetPos(FMotorControl->GetLvObj(), 0, 52);
    SetSize(FMotorControl->GetLvObj(), 675, 225);

ChowhoundWu, 2024-03-11:
I drew a simple UML diagram to illustrate：

ChowhoundWu, 2024-03-11:
I drew a simple UML diagram to illustrate：
[image]

ChowhoundWu, 2024-03-11:
It's a simple uml, I use C++, new TCard Cont 20 times cost about 1.6s on PC. And on a Linux development board  it cost about 16s.

kisvegabor, 2024-03-12:
Please send a code snippet that I can use the reproduce the slow creation.

closed 2024-04-07 by lvgl-bot