lvgl/lvgl #5821: How to set button long press time through functions?

IAMMX, 2024-03-09:
### Introduce the problem

For example, I have two buttons, one button that triggers an event by long pressing for 1 second, and the other button that triggers an event by long pressing for 2 seconds. What should I do?

liamHowatt, 2024-03-10:
Hey @IAMMX ,

This should achieve what want. Please let me know if this is what you were looking for.

```c
typedef void (*custom_press_cb_t)(lv_obj_t * obj);

typedef struct {
    lv_obj_t * obj;
    custom_press_cb_t press_cb;
    lv_timer_t * timer;
} custom_press_t;

static void my_custom_press_cb(lv_obj_t * btn)
{
    const char * btn_data_str = lv_obj_get_user_data(btn);
    LV_LOG_USER("btn %s custom long pressed", btn_data_str);
}

static void custom_press_timer_cb(lv_timer_t * timer)
{
    custom_press_t * cp = lv_timer_get_user_data(timer);
    cp->press_cb(cp->obj);
}

static void custom_press_event_cb(lv_event_t * e)
{
    custom_press_t * cp = lv_event_get_user_data(e);
    lv_event_code_t event_code = lv_event_get_code(e);
    switch(event_code) {
        case LV_EVENT_PRESSED:
            LV_LOG_USER("custom_press LV_EVENT_PRESSED");
            lv_timer_reset(cp->timer);
            lv_timer_set_repeat_count(cp->timer, 1);
            lv_timer_resume(cp->timer);
            break;
        case LV_EVENT_RELEASED:
            LV_LOG_USER("custom_press LV_EVENT_RELEASED");
            lv_timer_pause(cp->timer);
            break;
        case LV_EVENT_DELETE:
            LV_LOG_USER("custom_press LV_EVENT_DELETE");
            lv_timer_delete(cp->timer);
            lv_free(cp);
            break;
        default:
            break;
    }
}

static void custom_press(lv_obj_t * obj, uint32_t press_length, custom_press_cb_t press_cb)
{
    custom_press_t * cp = lv_malloc(sizeof(custom_press_t));
    cp->obj = obj;
    cp->press_cb = press_cb;
    cp->timer = lv_timer_create(custom_press_timer_cb, press_length, cp);
    lv_timer_pause(cp->timer);
    lv_timer_set_auto_delete(cp->timer, false);

    lv_obj_add_event_cb(obj, custom_press_event_cb, LV_EVENT_PRESSED, cp);
    lv_obj_add_event_cb(obj, custom_press_event_cb, LV_EVENT_RELEASED, cp);
    lv_obj_add_event_cb(obj, custom_press_event_cb, LV_EVENT_DELETE, cp);
}

static void demo_custom_press(void)
{
    lv_obj_t * cont = lv_obj_create(lv_screen_active());
    lv_obj_align(cont, LV_ALIGN_CENTER, 0, 0);
    lv_obj_set_size(cont, LV_SIZE_CONTENT, LV_SIZE_CONTENT);
    lv_obj_set_flex_flow(cont, LV_FLEX_FLOW_ROW);

    lv_obj_t * btn1 = lv_button_create(cont);
    lv_obj_set_size(btn1, 100, 50);
    lv_obj_set_user_data(btn1, "1");
    lv_obj_t * label1 = lv_label_create(btn1);
    lv_label_set_text(label1, "1");
    lv_obj_center(label1);
    custom_press(btn1, 1000, my_custom_press_cb);

    lv_obj_t * btn2 = lv_button_create(cont);
    lv_obj_set_size(btn2, 100, 50);
    lv_obj_set_user_data(btn2, "2");
    lv_obj_t * label2 = lv_label_create(btn2);
    lv_label_set_text(label2, "2");
    lv_obj_center(label2);
    custom_press(btn2, 2000, my_custom_press_cb);
}
```

IAMMX, 2024-03-11:
@liamHowatt, your approach is feasible, but it's too cumbersome for the users.
 
Currently, the official library has the function `lv_obj_add_event_cb(button, Button_Event, LV_EVENT_LONG_PRESSED, NULL)`, but it seems that the time can only be set through the macro definition `#define LV_INDEV_DEF_LONG_PRESS_TIME 400`.

I wonder if the official library can add the function `lv_obj_set_long_pressed_time(button, 1000)` to set the long-press time?
cc @kisvegabor

liamHowatt, 2024-03-11:
@IAMMX I agree that it is cumbersome. Regarding `lv_obj_set_long_pressed_time(button, 1000)`, it should be noted that the long pressed time is a property of the input device, not the object. The field in the input device would be trivial to add a setter to, but it would affect the press time for _all_ objects that are listening for long press events. Some logic could be reworked to get the `lv_obj_set_long_pressed_time(button, 1000)` effect.

kisvegabor, 2024-03-11:
@liamHowatt 
Thanks for the example! 

What do you think about this?
- Save the current tick in `LV_EVENT_PRESSED` in the `user_data`: 
```c
lv_obj_set_user_data(obj, (uint32_t) lv_tick_get());
```
- And in `LV_EVENT_PRESSING`: 
```c
t = (uint32_t) lv_obj_get_user_data(obj);
if(lv_tick_elaps(t) > 1000) do_something();`
```

If you don't want to use the `user_data` you can do this:
```c
static uint32_t t;
lv_obj_add_event_cb(obj, event_cb, LV_EVENT_ALL, &t); //Use the event's user_data instead of the widget's user_data
```

`t` can be malloc-ed too and freed in the delete event.

---

Having `lv_obj_set_long_pressed_time` would be simpler for sure, but we need to balance between API size and convenience. As this features wasn't request yet, I wouldn't add it now.

liamHowatt, 2024-03-12:
@kisvegabor that's brilliant! I totally overlooked `LV_EVENT_PRESSING` as being useful for this. And tick replaces the timer management which could be seen as a "cumbersome" part.

I always weigh new ideas but these user solutions are flexible and truly small, all considered.

kisvegabor, 2024-03-13:
Happy that you like it :slightly_smiling_face: 
I hope @IAMMX will find it useful too.

IAMMX, 2024-03-14:
Thanks @kisvegabor @liamHowatt

closed 2024-03-14 by IAMMX