lvgl/lvgl #5822: Can't set transform properties on button in lvgl 8.3.11

tyronehou, 2024-03-09:
### LVGL version

8.3.11

### What happened?

Running on ESP32 dev kit v1 connected to ILI9341 3.2" LCD touchscreen
Esp-idf v5.1.2
Compiling via vs code esp-idf extension

Button is not rendering when setting transform zoom or angle properties. When I comment out both these lines I can render the button  just fine. Using a slightly modified version of the esp-idf example code.

This is what I expected:
[image]

This is what shows up:
(No transform)
[image]

(with transform angle/zoom)
[image]

### How to reproduce?

LVGL code:

```
/*
 * SPDX-FileCopyrightText: 2021-2022 Espressif Systems (Shanghai) CO LTD
 *
 * SPDX-License-Identifier: CC0-1.0
 */

// This demo UI is adapted from LVGL official example: https://docs.lvgl.io/master/widgets/extra/meter.html#simple-meter

#include "lvgl.h"

static lv_obj_t * btn;
static lv_disp_rot_t rotation = LV_DISP_ROT_NONE;

static void btn_cb(lv_event_t * e)
{
    lv_disp_t *disp = lv_event_get_user_data(e);
    rotation++;
    if (rotation > LV_DISP_ROT_270) {
        rotation = LV_DISP_ROT_NONE;
    }
    lv_disp_set_rotation(disp, rotation);
}

void example_lvgl_demo_ui(lv_disp_t *disp)
{
    lv_obj_t *scr = lv_disp_get_scr_act(disp);

    btn = lv_btn_create(scr);
    lv_obj_t * lbl = lv_label_create(btn);
    lv_label_set_text_static(lbl, LV_SYMBOL_REFRESH" ROTATE");
    lv_obj_align(btn, LV_ALIGN_BOTTOM_LEFT, 30, -30);
    lv_obj_set_size(btn, 100, 40);
    lv_obj_set_style_transform_angle(btn, 150, 0);        /*15 deg*/
    lv_obj_set_style_transform_zoom(btn, 256 + 64, 0);   /*1.25x*/
    lv_obj_set_style_transform_pivot_x(btn, 50, 0);
    lv_obj_set_style_transform_pivot_y(btn, 20, 0);
    lv_obj_set_style_opa(btn, LV_OPA_50, 0);
    lv_obj_align(btn, LV_ALIGN_CENTER, 0, 70);
    /*Button event*/
    lv_obj_add_event_cb(btn, btn_cb, LV_EVENT_CLICKED, disp);
}
```

liamHowatt, 2024-03-14:
Hey @tyronehou,

I was able to reproduce this and I see a warning being logged.
```
[Warn]  (3.970, +35)     lv_draw_sw_layer_create: Rendering this widget needs LV_COLOR_SCREEN_TRANSP 1  (in lv_draw_sw_layer.c line #47)
[Warn]  (3.970, +0)      refr_obj: Couldn't create a new layer context  (in lv_refr.c line #998)
```

I set `LV_COLOR_SCREEN_TRANSP` to `1` in my lv_conf.h and it worked for me.

Hopefully that works for you too 🙂️

Also, you can enable logging in your lv_conf.h if you want, so you can see these warning logs in your serial console.

tyronehou, 2024-03-16:
@liamHowatt Thanks for the tips!

Seems when I set that to 1 get a memset error, wondering if I need to increase the size of my draw buffers? (Not sure exactly what other settings might need to be configured for this, will keep playing around with it)

I see it'll render up until the position of the test widget that's being transformed, which in this case happens to be near the top of the screen.
[image]

```
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x40132c12  PS      : 0x00060330  A0      : 0x8010d6e8  A1      : 0x3ffc0d30
0x40132c12: lv_memset_00 at C:/Users/tyron/Development/ESPer_pokedex/pokedex/managed_components/lvgl__lvgl/src/misc/lv_mem.c:490

A2      : 0x00000000  A3      : 0x0000085b  A4      : 0x00000054  A5      : 0x00000001
A6      : 0x00000002  A7      : 0x00000050  A8      : 0x00000000  A9      : 0x3ffc0d10
A10     : 0x00000000  A11     : 0x0000085c  A12     : 0x0000002c  A13     : 0x00000100  
A14     : 0xffffffce  A15     : 0xffffffce  SAR     : 0x0000001a  EXCCAUSE: 0x0000001d
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff
0x4000c2e0: memcpy in ROM

0x4000c2f6: memcpy in ROM

Backtrace: 0x40132c0f:0x3ffc0d30 0x4010d6e5:0x3ffc0d50 0x400f1a70:0x3ffc0d80 0x400f0641:0x3ffc0db0 0x400f058f:0x3ffc0e00 0x400f05d9:0x3ffc0e40 0x400f0868:0x3ffc0e90 0x400f0a9e:0x3ffc0eb0 0x400f0c1b:0x3ffc0f20 0x400f0d28:0x3ffc0f50 0x400f0ef6:0x3ffc0f70 0x400fecbd:0x3ffc0fa0 0x400fed7b:0x3ffc0fc0 0x400d6a68:0x3ffc0fe0 0x40134587:0x3ffc1100 0x40089291:0x3ffc1130
0x40132c0f: lv_memset_00 at C:/Users/tyron/Development/ESPer_pokedex/pokedex/managed_components/lvgl__lvgl/src/misc/lv_mem.c:481

0x4010d6e5: lv_draw_sw_layer_create at C:/Users/tyron/Development/ESPer_pokedex/pokedex/managed_components/lvgl__lvgl/src/draw/sw/lv_draw_sw_layer.c:77

0x400f1a70: lv_draw_layer_create at C:/Users/tyron/Development/ESPer_pokedex/pokedex/managed_components/lvgl__lvgl/src/draw/lv_draw_layer.c:58

0x400f0641: refr_obj at C:/Users/tyron/Development/ESPer_pokedex/pokedex/managed_components/lvgl__lvgl/src/core/lv_refr.c:996

0x400f058f: lv_obj_redraw at C:/Users/tyron/Development/ESPer_pokedex/pokedex/managed_components/lvgl__lvgl/src/core/lv_refr.c:182 (discriminator 3)

0x400f05d9: refr_obj at C:/Users/tyron/Development/ESPer_pokedex/pokedex/managed_components/lvgl__lvgl/src/core/lv_refr.c:974

0x400f0868: refr_obj_and_children at C:/Users/tyron/Development/ESPer_pokedex/pokedex/managed_components/lvgl__lvgl/src/core/lv_refr.c:856

0x400f0a9e: refr_area_part at C:/Users/tyron/Development/ESPer_pokedex/pokedex/managed_components/lvgl__lvgl/src/core/lv_refr.c:791

0x400f0c1b: refr_area at C:/Users/tyron/Development/ESPer_pokedex/pokedex/managed_components/lvgl__lvgl/src/core/lv_refr.c:680 (discriminator 2)

0x400f0d28: refr_invalid_areas at C:/Users/tyron/Development/ESPer_pokedex/pokedex/managed_components/lvgl__lvgl/src/core/lv_refr.c:618

0x400f0ef6: _lv_disp_refr_timer at C:/Users/tyron/Development/ESPer_pokedex/pokedex/managed_components/lvgl__lvgl/src/core/lv_refr.c:325

0x400fecbd: lv_timer_exec at C:/Users/tyron/Development/ESPer_pokedex/pokedex/managed_components/lvgl__lvgl/src/misc/lv_timer.c:313 (discriminator 2)

0x400fed7b: lv_timer_handler at C:/Users/tyron/Development/ESPer_pokedex/pokedex/managed_components/lvgl__lvgl/src/misc/lv_timer.c:109

0x400d6a68: app_main at C:/Users/tyron/Development/ESPer_pokedex/pokedex/main/main.c:317 (discriminator 1)

0x40134587: main_task at C:/Users/tyron/esp/esp-idf/components/freertos/app_startup.c:208 (discriminator 13)

0x40089291: vPortTaskWrapper at C:/Users/tyron/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:162
```

kisvegabor, 2024-03-16:
Note that a 100x40 button requires 16kB extra RAM. Can you increase LV_MEM_SIZE?

tyronehou, 2024-03-17:
This fixed it after changing LV_MEM_SIZE to 128kb, I found 64kb was enough for small widgets, but needed more memory once I increased the widget size.

Thank you both! Closing the issue now.

closed 2024-03-17 by tyronehou