lvgl/lvgl #5823: FATFS don't detect end of entries on fs_dir_read

serper, 2024-03-10:
### LVGL version

v9.0.0

### What happened?

Function fs_dir_read don't take care of end of entries doing that never end reading the next in a loop

### How to reproduce?

If you for example use lv_file_explorer on read files it get on a loop until some other error is fired and the list show all files and hundred os empty entries.

serper, 2024-03-10:
To correct it, only is needed to put one line in file _lvgl/src/libs/fsdrv/lv_fs_fatfs.c_ on the funcion:

```
static lv_fs_res_t fs_dir_read(lv_fs_drv_t * drv, void * dir_p, char * fn, uint32_t fn_len)
{
    LV_UNUSED(drv);
    FRESULT res;
    FILINFO fno;
    fn[0] = '\0';

    do {
        res = f_readdir(dir_p, &fno);
        if(res != FR_OK) return LV_FS_RES_UNKNOWN;

        if (fno.fname[0] == 0) break; /* End of the directory */

        if(fno.fattrib & AM_DIR) {
            lv_snprintf(fn, fn_len, "/%s", fno.fname);
        }
        else lv_strncpy(fn, fno.fname, fn_len);

    } while(lv_strcmp(fn, "/.") == 0 || lv_strcmp(fn, "/..") == 0);

    return LV_FS_RES_OK;
}
```
This line is that is needed:  
```
if (fno.fname[0] == 0) break; /* End of the directory */
```

FASTSHIFT, 2024-03-10:
Very good! Can you submit a PR?

serper, 2024-03-10:
Yes
https://github.com/lvgl/lvgl/pull/5826

closed 2024-03-12 by FASTSHIFT