lvgl/lvgl #5824: rotation animation leave

hyansuper, 2024-03-10:
### LVGL version

v9.0.0

### What happened?

Tested in [online python simulator](https://sim.lvgl.io/v9.0/micropython/ports/javascript/index.html)
object rotation animation leaves unwanted trace on it's path

https://github.com/lvgl/lvgl/assets/44317361/c478bd9a-b589-4ce0-9d50-b88d8c8e1139

### How to reproduce?

```python
import lvgl as lv

import display_driver

def anim_rotation_cb(obj, v):
    obj.set_style_transform_rotation(v, 0)

obj = lv.obj(lv.screen_active())
obj.set_style_bg_color(lv.palette_main(lv.PALETTE.RED), 0)
obj.set_size(20, 20)
obj.set_style_border_width(0, 0)
obj.set_style_shadow_width(0, 0)
obj.align(lv.ALIGN.CENTER, 0, -50)
obj.set_style_transform_pivot_x(10, 0)
obj.set_style_transform_pivot_y(60, 0)

a1 = lv.anim_t()
a1.init()
a1.set_var(obj)
a1.set_values(0, 3600)
a1.set_duration(10000)
a1.set_custom_exec_cb(lambda a1,val: anim_rotation_cb(obj,val))
lv.anim_t.start(a1)

```

XuNeo, 2024-03-11:
It cannot be reproduced by latest master version lvgl. Could you also try latest code?

```c

static void anim_rotation_cb(void * a, int32_t v) {
    lv_obj_t * obj = (lv_obj_t *)a;
    lv_obj_set_style_transform_rotation(obj, v, 0);
}

void lv_ex_anim_1(void) {
    lv_obj_t * obj = lv_obj_create(lv_scr_act());
    lv_obj_set_style_bg_color(obj, lv_palette_main(LV_PALETTE_RED), 0);
    lv_obj_set_size(obj, 20, 20);
    lv_obj_set_style_border_width(obj, 0, 0);
    lv_obj_set_style_shadow_width(obj, 0, 0);
    lv_obj_align(obj, LV_ALIGN_CENTER, 0, -50);
    lv_obj_set_style_transform_pivot_x(obj, 10, 0);
    lv_obj_set_style_transform_pivot_y(obj, 60, 0);

    lv_anim_t a1;
    lv_anim_init(&a1);
    lv_anim_set_var(&a1, obj);
    lv_anim_set_values(&a1, 0, 3600);
    lv_anim_set_time(&a1, 10000);
    lv_anim_set_exec_cb(&a1, anim_rotation_cb);
    lv_anim_start(&a1);
}
```

liamHowatt, 2024-03-11:
Fixed by #5548

PGNetHun, 2024-03-11:
Hi @hyansuper ,
Which LVGL-MicroPython binding do you use?
1. https://github.com/lvgl/lv_micropython
2. https://github.com/kdschlosser/lvgl_micropython

If the second ([kdschlosser/lvgl_micropython](https://github.com/kdschlosser/lvgl_micropython)), then you might receive the bugfix faster, because I don't want to upgrade [lvgl/lv_micropython](https://github.com/lvgl/lv_micropython) after every LVGL fix/commit, only when there is a new release (https://github.com/lvgl/lvgl/releases), so we can keep [lvgl/lv_micropython](https://github.com/lvgl/lv_micropython) relatively stable (but less up-to-date).

kisvegabor, 2024-03-12:
FYI, I've just opened this issue about v9.1: #5843

kisvegabor, 2024-03-27:
Closing should be fixed by https://github.com/lvgl/lvgl/pull/5548

closed 2024-03-27 by kisvegabor