lvgl/lvgl #5827: If the offset parameter in function lv_canvas_transform is negative, it cannot be displayed properly

sywxf, 2024-03-10:
### LVGL version

8.3.10

### What happened?

[image]
[image]

### How to reproduce?

lv_canvas_transform(canv, &imgs[4], 0, LV_IMG_ZOOM_NONE, -10, -10, 0, 0, true);

liamHowatt, 2024-03-12:
## Reproduction

I reproduced the issue on v8.3.10 using the example `lv_example_canvas_1` and adjusting this line:

https://github.com/lvgl/lvgl/blob/06e229cc7679c01119ca5d905a1f9dad96f475f0/examples/widgets/canvas/lv_example_canvas_1.c#L50

0, 0 -- unmodified, okay
`lv_canvas_transform(canvas, &img, 120, LV_IMG_ZOOM_NONE, 0, 0, CANVAS_WIDTH / 2, CANVAS_HEIGHT / 2, true);`
[image]

0, -50 -- y-offset of -50, displayed incorrectly
`lv_canvas_transform(canvas, &img, 120, LV_IMG_ZOOM_NONE, 0, -50, CANVAS_WIDTH / 2, CANVAS_HEIGHT / 2, true);`
[image]

## Fix
Branch: https://github.com/liamHowatt/lvgl/tree/fix/canvtranf-offset-y

https://github.com/liamHowatt/lvgl/commit/a9dbcb59cfd85639976361065228e790e6f337a0?diff=unified&w=1

I removed an `if` block that stopped it from drawing the rows above the original position of the image. I can't determine why this was there in the first place or whether there are any side effects for removing it.

Now,
0, -50 -- y-offset of -50, okay
`lv_canvas_transform(canvas, &img, 120, LV_IMG_ZOOM_NONE, 0, -50, CANVAS_WIDTH / 2, CANVAS_HEIGHT / 2, true);`
[image]

This function `lv_canvas_transform` no longer exists in v9 and I'm not sure how fixes for older major versions are handled. Open to input from others on this.

kisvegabor, 2024-03-12:
@liamHowatt 
Thanks for the investigation! 

> This function lv_canvas_transform no longer exists in v9 and I'm not sure how fixes for older major versions are handled. Open to input from others on this.

I consider it a s bugfix. Could you send a Pull request to the `release/v8.3` branch?

kisvegabor, 2024-03-27:
Foxed by https://github.com/lvgl/lvgl/pull/5846

closed 2024-03-27 by kisvegabor