lvgl/lvgl #5830: Parallel RGB LCD panel with ESP32s3 and lvgl 9

Ka1mes, 2024-03-11:
### Introduce the problem

I am currently trying to port an older project, which uses lvgl 8.3 on an ESP32 board. I am developing the project using the esp-idf. In LVGL 8.3 I was able to easily setup my display by configuring the ports using the ESP driver and then I initialized the display to lvgl using "lv_disp_drv_register". Since in LVGL 9 "lv_disp_drv_t" no longer exists I am unsure how it can work and if it even is supported right now.

closed 2024-03-11 by Ka1mes

anujdeshpande, 2024-04-18:
@Ka1mes how did you go about with this? trying to find a porting guide from 8 to 9 - but I don't think there exists such a thing

Ka1mes, 2024-04-19:
Since the lv_disp_drv_t struct no longer exists to pass it the esp_lcd_panel_handle_t struct you have to add it as user data to the lv_disp_t struct containing your display. I also changed the display flush callback to handle RGB 565 like shown in the documentation