lvgl/lvgl #5843: Prepare v9.1

kisvegabor, 2024-03-12:
### Introduce the problem

-

### Proposal

In the past 2 month I think we could hammer out the initial issues of LVGL v9 and we are ready to release v9.1. In parallel, I think it makes sense to release v8.4 too in order to have a fresh start after v8.3.11. 

I'm targeting next Tuesday (March 19) with the release.

If there are any critical PRs/issues that we need to take care of before the release, please let me know.

IAMMX, 2024-03-12:
Can you add a formatting function to textarea? For example, `lv_textarea_set_text_fmt` .
Because I need to set a default value for textarea depending on the situation. But there is no such formatting function, I need to `lv_snprintf` on one buff at a time, and then write it using `lv_textarea_set_text` . This is especially troublesome.

liamHowatt, 2024-03-12:
Re v8.4 PRs: #5846

kisvegabor, 2024-03-16:
@IAMMX 
> Can you add a formatting function to textarea? For example, lv_textarea_set_text_fmt .

I've looked into into but due to [this part](https://github.com/lvgl/lvgl/blob/master/src/widgets/textarea/lv_textarea.c#L291-L292) in some cases we need to create a buffer internally anyway. So `lv_textarea_set_text_fmt` can be just a helper and cannot avoid the extra temporal allocation.  Do you think it's still worth it?

IAMMX, 2024-03-17:
@kisvegabor 
> I've looked into into but due to [this part](https://github.com/lvgl/lvgl/blob/master/src/widgets/textarea/lv_textarea.c#L291-L292) in some cases we need to create a buffer internally anyway. So [...]

I think it's worth it. Because the extra cache is already integrated into the function, users do not need to allocate it themselves. As long as it is simple to use, users generally do not care how your function is implemented inside.

100ask, 2024-03-18:
> If there are any critical PRs/issues that we need to take care of before the release, please let me know.

@kisvegabor  Don't forget update README: #5841

Cuixudong, 2024-03-18:
want to continue to support drag and drop of controls

Cuixudong, 2024-03-18:
@kisvegabor ，This is my rewrite of the Chinese input method code, it works very well in the V7 version, but there is no good method to implement on V9, I hope you can try it if you have time，https://github.com/Cuixudong/ReLVGL_Widgets

XuNeo, 2024-03-18:
Please let me confirm latest version `lvgl` works on `nuttx` before releasing it. https://github.com/apache/nuttx-apps/pull/2312

kisvegabor, 2024-03-18:
@100ask 
> @kisvegabor Don't forget update README: https://github.com/lvgl/lvgl/pull/5841

Merged!

@Cuixudong 
> This is my rewrite of the Chinese input method code, it works very well in the V7 version, but there is no good method to implement on V9, I hope you can try it if you have time，https://github.com/Cuixudong/ReLVGL_Widgets

The link drops 404 error :slightly_frowning_face: 

@XuNeo 
> Please let me confirm latest version lvgl works on nuttx before releasing it.

Sure, thanks!

Cuixudong, 2024-03-19:
@kisvegabor Sorry I forgot to set the repository to public, now it has been changed to public

kisvegabor, 2024-03-19:
@GorgonMeducer 
Could you prepare the release fir both v8.4 and v9.1?

Cuixudong, 2024-03-19:
i find no **_colorwheel_** controls in V9.0

GorgonMeducer, 2024-03-19:
@kisvegabor I will do this today.

kisvegabor, 2024-03-20:
Thank you Gabriel! Both PRs are merged, I tagged the versions, and made the releases.

@tore-espressif 
Publishing v8.4 to the Espressif registry is [failed](https://github.com/lvgl/lvgl/actions/runs/8355367583/job/22870316791). Could you take a look at it? (v9.1 was [working well](https://github.com/lvgl/lvgl/actions/runs/8355367554))

@ivankravets 
Publishing to PIO is also [failed](https://github.com/lvgl/lvgl/actions/runs/8355367561/job/22870316257), but I could [publish it manually](https://registry.platformio.org/libraries/lvgl/lvgl/versions).

XuNeo, 2024-03-20:
> Please let me confirm latest version `lvgl` works on `nuttx` before releasing it. [apache/nuttx-apps#2312](https://github.com/apache/nuttx-apps/pull/2312)

Hi Gabor, I confirmed latest master `lvgl` works on `nuttx`. :)

ivankravets, 2024-03-20:
How did it work before from CI? You need to pass `--no-interactive`. See https://docs.platformio.org/en/latest/core/userguide/pkg/cmd_publish.html#cmdoption-pio-pkg-publish-no-interactive

kisvegabor, 2024-03-20:
> Hi Gabor, I confirmed latest master lvgl works on nuttx. :)

Ouch, I forgot to wait for that. Thanks! :slightly_smiling_face: 

> How did it work before from CI? You need to pass --no-interactive

Hm, I don't know either :slightly_smiling_face:  
I've just added it here:  #5924

tore-espressif, 2024-03-21:
> @tore-espressif
Publishing v8.4 to the Espressif registry is [failed](https://github.com/lvgl/lvgl/actions/runs/8355367583/job/22870316791). Could you take a look at it? (v9.1 was [working well](https://github.com/lvgl/lvgl/actions/runs/8355367554))

@kisvegabor The upload failed on `8.4` tag, but then passed on `8.4.0` tag (we are too strict :D ). So LVGL v8.4.0 is uploaded correctly :)

kisvegabor, 2024-03-25:
Ah, cool, thank you! :slightly_smiling_face: 

As all seems good with the releases now, I'm closing this issue.

closed 2024-03-25 by kisvegabor