lvgl/lvgl #5847: Trace output inside lv_init() disabled when using a custom log function

antonio-pasini, 2024-03-12:
### LVGL version

V9.0.0

### What happened?

When a trace callback has been registered with lv_log_register_print_cb() before calling lv_init(), the initial "begin" message will be printed, but not the final "finished" trace line. 

During lv_init() the custom_log_print_cb field is zeroed, overriding any previoysly assigned callback.

Not really an issue and probably not even intended to work, but since the LV_LOG_INFO() macros are used inside lv_init(), would be more coherent if those worked also with a custom printf function. 

Probably inside lv_global_init() could be saved a previously assigned callback ?

    lv_log_print_g_cb_t prev = global->custom_log_print_cb;
    lv_memzero(global, sizeof(lv_global_t));
    global->custom_log_print_cb = prev;

### How to reproduce?

lv_conf.h: #define LV_LOG_PRINTF 0

lv_log_register_print_cb(&my_trace_cb);
lv_init();   // <-- will trace only "begin", not "finished"

kisvegabor, 2024-03-19:
Saving/Restoring the `log_cb` could work. My only concern is what will happen if `lv_globals` contains garbage and we keep that garbage for the callback. 

Usually we are using a `lv_global_t lv_global;` (a global C variable, no pun intended) which is automatically initialized to zero. So we are safe to save/restore the callback. If the user uses a custom global data, it can be his/her responsibility to handle initialize it to zero. 

cc @XuNeo @bjsylvia

kisvegabor, 2024-04-12:
cc @XuNeo @bjsylvia could you take a look?

FASTSHIFT, 2024-04-14:
We can consider adding a function similar to [LV_SYSMON_GET_IDLE](https://github.com/lvgl/lvgl/blob/master/lv_conf_template.h#L776) (such add `LV_LOG_CUSTOM_INIT`) that actively calls the user-defined log initialization function after global init.

kisvegabor, 2024-04-20:
That's a good idea as it's a platform specific option that can be set in compile time. I've opened #6095

PR #6095 (2024-04-25): feat(log): add LV_LOG_PRINT_CB to set a default log print cb
fixes #5847

### Description of the feature or fix

A clear and concise description of what the bug or new feature is.

### Notes
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.
- Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant.
- Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if applicable.
- If you added new options to `lv_conf_template.h` run [lv_conf_internal_gen.py](https://github.com/lvgl/lvgl/blob/master/scripts/lv_conf_internal_gen.py) [... cut]
closed 2024-04-20 by kisvegabor