lvgl/lvgl #5853: lvgl\draw\sw\blend\neon\lv_blend_neon.S.o] Error 1

lvthong312, 2024-03-13:
### LVGL version

9.0.0

### What happened?

Hi, I have downloaded latest master v9.0.1-dev, but it's still not working. The root issue here [5850](https://github.com/lvgl/lvgl/issues/5850). Please help me check it, thanks 

platform = espressif32
board = lilygo-t-displays3

lvgl/lvgl@9.0.0
bodmer/TFT_eSPI@^2.5.43

### How to reproduce?

the lv_config.h file of latestmaster on below:

/**

https://github.com/file lv_conf.h
Configuration file for v9.0.1-dev
*/
/*

Copy this file as lv_conf.h
simply next to the lvgl folder
or any other places and
define LV_CONF_INCLUDE_SIMPLE
add the path as include path
*/
/* clang-format off */
#if 1 /Set it to "1" to enable content/

#ifndef LV_CONF_H
#define LV_CONF_H

/*If you need to include anything here, do it inside the __ASSEMBLY__ guard */
#if 0 && defined(ASSEMBLY)
#include "my_include.h"
#endif

/*====================
COLOR SETTINGS
====================/

/Color depth: 8 (A8), 16 (RGB565), 24 (RGB888), 32 (XRGB8888)/
#define LV_COLOR_DEPTH 16

/*=========================
STDLIB WRAPPER SETTINGS
=========================/

/* Possible values

LV_STDLIB_BUILTIN: LVGL's built in implementation
LV_STDLIB_CLIB: Standard C functions, like malloc, strlen, etc
LV_STDLIB_MICROPYTHON: MicroPython implementation
LV_STDLIB_RTTHREAD: RT-Thread implementation
LV_STDLIB_CUSTOM: Implement the functions externally
*/
#define LV_USE_STDLIB_MALLOC LV_STDLIB_BUILTIN
#define LV_USE_STDLIB_STRING LV_STDLIB_BUILTIN
#define LV_USE_STDLIB_SPRINTF LV_STDLIB_BUILTIN
#if LV_USE_STDLIB_MALLOC == LV_STDLIB_BUILTIN
/Size of the memory available for lv_malloc() in bytes (>= 2kB)/
#define LV_MEM_SIZE (64 * 1024U) /[bytes]/

/*Size of the memory expand for `lv_malloc()` in bytes*/
#define LV_MEM_POOL_EXPAND_SIZE 0
[... 909 lines cut ...]
#define LV_DEMO_MUSIC_LANDSCAPE 0
#define LV_DEMO_MUSIC_ROUND 0
#define LV_DEMO_MUSIC_LARGE 0
#define LV_DEMO_MUSIC_AUTO_PLAY 0
#endif

/Flex layout demo/
#define LV_USE_DEMO_FLEX_LAYOUT 0

/Smart-phone like multi-language demo/
#define LV_USE_DEMO_MULTILANG 0

/Widget transformation demo/
#define LV_USE_DEMO_TRANSFORM 0

/Demonstrate scroll settings/
#define LV_USE_DEMO_SCROLL 0

/Vector graphic demo/
#define LV_USE_DEMO_VECTOR_GRAPHIC 0
/--END OF LV_CONF_H--/

#endif /LV_CONF_H/

#endif /End of "Content enable"/

XuNeo, 2024-03-13:
Not sure if it's relevant, it's also reported here: https://github.com/lvgl/lvgl/issues/5620

lvthong312, 2024-03-13:
I tried update lv_conf.h like that, specifically below but it's still not working.
Update your lv_conf.h from the new lv_conf_template.h (from the lvgl dir)
Set LV_USE_TFT_ESPI 1 in your lv_conf.h
Use the includes as shown above!

I'm using PlatformIO Core, version 6.1.13. when i back lvgl8.3.11 it's work. on lvgl v9.0.0 it's generate .S file and compile bug

lvthong312, 2024-03-14:
Thank you. I have adready fixed by initial new project and it's worked. Now i will close the issue

closed 2024-03-14 by lvthong312