lvgl/lvgl #5854: manually change the state of an item in the table to focused

liangfei2021, 2024-03-13:
### LVGL version

lvgl 8.2

### What happened?

donot find the handle

liangfei2021, 2024-03-14:
@kisvegabor

kisvegabor, 2024-03-16:
There is no API for it but you can do this:
```c
    lv_table_t * table = (lv_table_t *) obj;
    table->col_act = 1;
    table->row_act = 3;
    lv_obj_invalidate(obj);
```

liangfei2021, 2024-03-18:
thank you，I try

liangfei2021, 2024-03-18:
There are multiple items in the table, how to make them scroll up (my screen can't be touched, only the relevant functions can be called)

liamHowatt, 2024-03-19:
You could use `lv_obj_scroll_by`, `lv_obj_scroll_to` on the table.

If my understanding is correct, the cells aren't `obj`s so they can't have the `LV_OBJ_FLAG_SCROLL_ON_FOCUS` flag set or have `lv_obj_scroll_to_view` done to them.

If `get_row_height` was public then it could be used in conjunction with `lv_obj_scroll_to`. `lv_table_set_col_width` is public.

closed 2024-04-14 by lvgl-bot