lvgl/lvgl #5855: LV_USE_DRAW_ARM2D_SYNC does not work

KlemenDEV, 2024-03-13:
### LVGL version

v9.0.0

### What happened?

When enabling LV_USE_DRAW_ARM2D_SYNC, nothing really happens. I have enabled this on STM32H5 which does not even have DMA2D (ChromART), but everything still compiles.

I have checked the files and it seems to enable LV_USE_DRAW_ARM2D_SYNC does not include any additional headers by default.

Is STM32 ChromART (e.g. H7 or U5 or F4 core) acceleration still supported in v9.0.0 and if yes, how does one use it?

KlemenDEV, 2024-03-13:
I have found https://docs.lvgl.io/master/CHANGELOG.html:

> STM32's DMA2D (Chrom-ART) support is removed for now. It will be added again when an official partnership starts with ST too.

Are there any estimates on this?

kisvegabor, 2024-03-16:
Right now Arm2D is useful only if the MCH has [Helium](https://www.arm.com/technologies/helium) (SIMD) instruction set. (Only in Cortex M85 and M55)  So you need to enable `LV_USE_DRAW_SW_ASM     LV_DRAW_SW_ASM_HELIUM` too. 

We really need better docs on this part.

cc @GorgonMeducer 

> Are there any estimates on this?

There is a progress, but no final agreement yet. To not disclose any possible sensitive business information, I can say that I expect some larger steps forward in a month.

GorgonMeducer, 2024-03-18:
@kisvegabor We do have document that explain how Arm-2D works with LVGL9.

https://docs.lvgl.io/master/overview/renderers/arm2d.html

kisvegabor, 2024-03-21:
> If there is anything missing, please let me know.

I really forgot this docs page, however it doesn't mention `LV_DRAW_SW_ASM` which is required to use Arm2D.

GorgonMeducer, 2024-03-22:
@kisvegabor Opps... You are right. I will update the doc.

closed 2024-04-07 by KlemenDEV