lvgl/lvgl #5858: background-size implementation issues

C-Ackerman, 2024-03-13:
### LVGL version

v9.0.0

### What happened?

How to implement background-size: contain; or background-size: 5%;
https://developer.mozilla.org/en-US/docs/Web/CSS/background-size

i try lv_image_set_scale(bgimg, LV_SCALE_NONE * 0.5); and lv_image_set_align(bgimg, LV_IMAGE_ALIGN_TILE); setting scale at the same time does not take effect

[image]

[image]

```
void lv_example_style_6(void)
{

    lv_obj_t * native = lv_obj_create(lv_screen_active());
    lv_obj_set_size(native, 450, 450);
    static lv_style_t style;
    lv_style_init(&style);
    // lv_style_set_bg_color(&style, lv_palette_lighten(LV_PALETTE_GREY, 3));
    lv_style_set_border_width(&style, 2);
    lv_style_set_border_color(&style, lv_palette_lighten(LV_PALETTE_YELLOW, 3));
    lv_obj_add_style(native, &style, 0);

    lv_obj_t * bgimg = lv_image_create(native);
    lv_image_set_src(bgimg, "/data/app/logo.png");
    lv_image_set_scale(bgimg, LV_SCALE_NONE * 0.5);
    lv_image_set_align(bgimg, LV_IMAGE_ALIGN_TILE);

    lv_obj_t * panel1 = lv_obj_create(bgimg);
    lv_obj_set_size(panel1, 440, 440);
    lv_obj_center(panel1);
    static lv_style_t style1;
    lv_style_init(&style1);
    lv_style_set_opa(&style1, 50);
    lv_style_set_border_width(&style1, 2);
    lv_style_set_border_color(&style1, lv_palette_lighten(LV_PALETTE_PINK, 3));
    lv_obj_add_style(panel1, &style1, 0);

    lv_obj_t * label = lv_label_create(bgimg);
    lv_label_set_text(label, "Widgets demo");
    lv_obj_set_size(label, 100, 200);
    lv_obj_align(label, LV_ALIGN_CENTER, 50, 100);

}
```

C-Ackerman, 2024-03-15:
There is another problem. If you use a picture object as the background image, then the child nodes of the picture need to set transparency (lv_style_set_opa) to see the picture.

kisvegabor, 2024-03-20:
`background-size: contain` is the similar to `LV_IMAGE_ALIGN_STRETCH`. Is it good enough for your needs?

`background-size: 5%` + `repeat` is really not supported. I think drawing that many runtime scaled images wouldn't be that effective. 

> If you use a picture object as the background image, then the child nodes of the picture need to set transparency (lv_style_set_opa) to see the picture.

I don't understand the problem. Could you send an image bout the issue?

C-Ackerman, 2024-03-20:
In the issue https://github.com/lvgl/lvgl/issues/4733, it is not possible to set the background-size and other capabilities of the image through lv_obj_set_style_bg_image_src, so now the following method is used to use bgimg as the background image of panel1. Panel1 You need to set lv_style_set_opa to display the subsequent pictures. What other methods can currently be used to control background-size on background-image?

lv_obj_t * bgimg = lv_image_create(native);
...
lv_obj_t * panel1 = lv_obj_create(bgimg);

closed 2024-04-15 by C-Ackerman