lvgl/lvgl #5861: Image caching issue (out of memory)

Superberti, 2024-03-13:
### LVGL version

v9.0.0

### What happened?

Hi,

I'm not sure, if this is a bug or if the problem sits on front of the monitor ;-)

I'm using an ESP32-S3 board with 8 MB PSRAM on a 800x480 pixel display. I'm loading a quite large .png image from a serial line and it will be drawn on a tab sheet. This works quite good.
To decode the .png images faster, I'm using a large cache (#define LV_CACHE_DEF_SIZE       (2*1024*1024)).

This speeds greatly up the image drawing!
I'm setting a new arrived .png image with

```
lv_image_cache_drop(&QxDia);
lv_image_set_src(QxDiaImage, &QxDia);
```

where "QxDia" is a lv_image_dsc_t type. This works the first time, the second time, but the third time the PNG-decoding fails with 

```
[Warn]  (69.026, +69026)         lv_draw_buf_create: No memory: 694x396, cf: 16, stride: 2776, 1099296Byte,  lv_draw_buf.c:186
[Warn]  (69.026, +0)     decoder_open: Error decoding PNG
 lv_lodepng.c:186
[Error] (69.032, +6)     _lv_draw_image_normal_helper: Failed to open image lv_draw_image.c:155
[Warn]  (69.210, +178)   lv_draw_buf_create: No memory: 694x396, cf: 16, stride: 2776, 1099296Byte,  lv_draw_buf.c:186
[Warn]  (69.210, +0)     decoder_open: Error decoding PNG
```

So maybe lv_image_cache_drop does not work as expected or I'm suffering from memory fragmentation. When the images are decrompressed they are around ~1 MiB large, but two of them should fit in the cache.

If I try to use "lv_image_cache_drop(NULL)" I will get the warning:

```
[Warn]  (126.754, +89620)        drop_all_cb: entry (0x3fcc4474) is still referenced (38) _lv_cache_lru_rb.c:380
[Warn]  (126.760, +6)    drop_all_cb: 1 entries are still referenced _lv_cache_lru_rb.c:385
```

So, any comments to this? I don't know who still references the image.

Thanks and bye,
Oliver

kisvegabor, 2024-03-19:
Hi Oliver,

Thank you for reporting the issue. We need to investigate it for sure.

Do you use other images in the UI as well?

cc @W-Mai

W-Mai, 2024-03-19:
Hi @Superberti,

I seem to have encountered this problem before. You can try to update the lvgl version to the latest version of master to try to reproduce the problem.

Superberti, 2024-03-19:
@kisvegabor: The mentioned PNG was the only image in the cache. I'm using some other images in the flash memory, from the image converter. But they are all static.

I'd like to update LVGL but I'm using it as a managed ESP-IDF component and there's no update available yet from espressif.
Maybe I have to switch to a self managed component model, but I don't know the implications right now...

kisvegabor, 2024-03-20:
We have released v9.1 today, but there was [an issue](https://github.com/lvgl/lvgl/issues/5843#issuecomment-2008970884) while pushing it to ESP's registry. It should be resolved in a few days.

kisvegabor, 2024-03-20:
Sorry, v8.4 was faild to push to the registry. [v9.1 is already there](https://components.espressif.com/components/lvgl/lvgl).

Superberti, 2024-03-20:
Great news, thanks for that.

BTW: I made a great looking GUI demonstration for an instrument in just two weeks with LVGL. I wonder how useful it is even on such a low-cost and low-power platform (ESP32-S3 on a Waveshare 4.3 IPS display, ~40$).

So my thanks to all the LVGL maintainers here!

kisvegabor, 2024-03-25:
Happy that it worked well and you like LVGL :slightly_smiling_face:

closed 2024-04-10 by Superberti