lvgl/lvgl #5866: set different obj grad style but result same style...

Staok, 2024-03-14:
### LVGL version

v8.3.11

### What happened?

the code:
[image]

[image]

the result is:
[image]

why??????????????????????????
they are the same grad result, is the default style inst of all new obj is the same one?

C47D, 2024-03-15:
Can you try initializing the colors as indicated in the documentation? See https://docs.lvgl.io/8.3/overview/color.html?highlight=lv_color_t#creating-colors

Staok, 2024-03-18:
@C47D use the "lv_color_make()" or "lv_color_hex" to init color get the same result, thses functions essentially is fill the lv_color_t var and return it, the same I fill it directly...
I am wondering is it only the lvgl source code I am using that has this issue?
Do you all not have this situation?

Staok, 2024-03-18:
@kisvegabor looking forward for reply, I waited for days, is this happened because the wrong usage I did or the LVGL bug, can it be fixed, waiting for help

liamHowatt, 2024-03-18:
@Staok 

I wasn't able to reproduce the issue.

Could you please share the part of your code that initializes the `ui_fileBrowserInternalInstListPanel`? Please send it as a code block instead of a screenshot.

When initializing the color structs manually with a 32 bit value like that, ensure `COLOR_DEPTH` is `32` in your lv_conf.h. (though using `16` still didn't reproduce the issue for me)

[image]

```c
static void gradient_issue(void) {
    lv_obj_t * cont = lv_obj_create(lv_disp_get_scr_act(disp)); // (disp from file scope)
    lv_obj_align(cont, LV_ALIGN_CENTER, 0, 0);
    lv_obj_set_size(cont, LV_SIZE_CONTENT, LV_SIZE_CONTENT);
    lv_obj_set_flex_flow(cont, LV_FLEX_FLOW_ROW);

    lv_color_t color1 = {.full=0xffe3e12e};
    lv_color_t color4 = {.full=0xffffffff};
    lv_color_t color5 = {.full=0xff000000};
    lv_color_t color6 = {.full=0xff3b00ff};

    static lv_grad_dsc_t grad[3];

    grad[0].dither = LV_DITHER_NONE;
    grad[0].dir = LV_GRAD_DIR_HOR;
    grad[0].stops_count = 2;
    grad[0].stops[0].color = color5;
    grad[0].stops[0].frac = 0;
    grad[0].stops[1].color = color6;
    grad[0].stops[1].frac = 255;

    grad[1].dither = LV_DITHER_NONE;
    grad[1].dir = LV_GRAD_DIR_HOR;
    grad[1].stops_count = 2;
    grad[1].stops[0].color = color4;
    grad[1].stops[0].frac = 0;
    grad[1].stops[1].color = color1;
    grad[1].stops[1].frac = 255;

    lv_obj_t *panel0 = lv_obj_create(cont);
    lv_obj_set_style_bg_grad(panel0, &grad[0], LV_PART_MAIN | LV_STATE_DEFAULT);

    lv_obj_t *panel1 = lv_obj_create(cont);
    lv_obj_set_style_bg_grad(panel1, &grad[1], LV_PART_MAIN | LV_STATE_DEFAULT);

    LV_UNUSED(panel0);
    LV_UNUSED(panel1);
}
```

liamHowatt, 2024-03-18:
Was able to reproduce your issue by setting `LV_GRAD_CACHE_DEF_SIZE` to larger values. I will look into the cause. In the meantime set it to `0` as kisvegabor suggested.

liamHowatt, 2024-03-18:
The problem is that cached gradients resolve to the same key if they are the same width, wrongly causing a cache hit. The gradient cache key is a combination of the dimensions of the gradient and the memory address of the gradient descriptor. The memory address in this case is a stack value so it's the same as the other one because the two call stacks alias each other.

The stack value whose memory address is being used to compute the key
https://github.com/lvgl/lvgl/blob/74d0a816a440eea53e030c4f1af842a94f7ce3d3/src/core/lv_obj.c#L526

The place where the key is being computed
https://github.com/lvgl/lvgl/blob/74d0a816a440eea53e030c4f1af842a94f7ce3d3/src/draw/sw/lv_draw_sw_gradient.c#L265

Backtrace
```
#0  lv_gradient_get (g=0x7fffffffc958, w=128, h=128) at src/draw/sw/lv_draw_sw_gradient.c:265
#1  0x000055555568aa8a in draw_bg (draw_ctx=0x555555707410 <work_mem_int+3120>, dsc=0x7fffffffc950, coords=0x7fffffffc7d0) at src/draw/sw/lv_draw_sw_rect.c:171
#2  0x0000555555689e04 in lv_draw_sw_rect (draw_ctx=0x555555707410 <work_mem_int+3120>, dsc=0x7fffffffc950, coords=0x7fffffffc7d0) at src/draw/sw/lv_draw_sw_rect.c:75
#3  0x00005555555ba789 in lv_draw_rect (draw_ctx=0x555555707410 <work_mem_int+3120>, dsc=0x7fffffffc950, coords=0x7fffffffc7d0) at src/draw/lv_draw_rect.c:66
#4  0x000055555557c160 in lv_obj_draw (e=0x7fffffffcc80) at src/core/lv_obj.c:561
#5  0x000055555557dfd8 in lv_obj_event (class_p=0x5555556cd3c0 <lv_obj_class>, e=0x7fffffffcc80) at src/core/lv_obj.c:874
#6  0x000055555556722c in lv_obj_event_base (class_p=0x0, e=0x7fffffffcc80) at src/core/lv_event.c:98
#7  0x0000555555569758 in event_send_core (e=0x7fffffffcc80) at src/core/lv_event.c:458
#8  0x0000555555566fb9 in lv_event_send (obj=0x555555708d58 <work_mem_int+9592>, event_code=LV_EVENT_DRAW_MAIN, param=0x555555707410 <work_mem_int+3120>) at src/core/lv_event.c:75
#9  0x00005555555a132d in lv_obj_redraw (draw_ctx=0x555555707410 <work_mem_int+3120>, obj=0x555555708d58 <work_mem_int+9592>) at src/core/lv_refr.c:148
#10 0x00005555555a651b in refr_obj (draw_ctx=0x555555707410 <work_mem_int+3120>, obj=0x555555708d58 <work_mem_int+9592>) at src/core/lv_refr.c:974
#11 0x00005555555a154b in lv_obj_redraw (draw_ctx=0x555555707410 <work_mem_int+3120>, obj=0x555555708c40 <work_mem_int+9312>) at src/core/lv_refr.c:182
#12 0x00005555555a651b in refr_obj (draw_ctx=0x555555707410 <work_mem_int+3120>, obj=0x555555708c40 <work_mem_int+9312>) at src/core/lv_refr.c:974
#13 0x00005555555a154b in lv_obj_redraw (draw_ctx=0x555555707410 <work_mem_int+3120>, obj=0x5555557086d8 <work_mem_int+7928>) at src/core/lv_refr.c:182
#14 0x00005555555a651b in refr_obj (draw_ctx=0x555555707410 <work_mem_int+3120>, obj=0x5555557086d8 <work_mem_int+7928>) at src/core/lv_refr.c:974
#15 0x00005555555a59a1 in refr_obj_and_children (draw_ctx=0x555555707410 <work_mem_int+3120>, top_obj=0x5555557086d8 <work_mem_int+7928>) at src/core/lv_refr.c:856
#16 0x00005555555a5492 in refr_area_part (draw_ctx=0x555555707410 <work_mem_int+3120>) at src/core/lv_refr.c:791
#17 0x00005555555a42e3 in refr_area (area_p=0x5555557072b0 <work_mem_int+2768>) at src/core/lv_refr.c:680
#18 0x00005555555a373e in refr_invalid_areas () at src/core/lv_refr.c:618
#19 0x00005555555a20b8 in _lv_disp_refr_timer (tmr=0x5555557074c8 <work_mem_int+3304>) at src/core/lv_refr.c:325
#20 0x000055555561b03a in lv_timer_exec (timer=0x5555557074c8 <work_mem_int+3304>) at src/misc/lv_timer.c:313
#21 0x000055555561a7e8 in lv_timer_handler () at src/misc/lv_timer.c:109
#22 0x0000555555562c4e in main (argc=1, argv=0x7fffffffd9a8) at /home/liam/lvgl/lvgl_workspace_v8/main.c:179
```

Staok, 2024-03-19:
I set 'LV_GRAD_CACHE_DEF_SIZE' to 0 and finally the result is correct.

kisvegabor, 2024-03-20:
@liamHowatt, can we resolve the issue by hashing the gradient parameters (by value) instead of the pointers?

kisvegabor, 2024-04-06:
https://github.com/lvgl/lvgl/pull/5930 is merged so I close this issue.

PR #5930 (2024-03-25): fix(cache): gradient key computing
### Description of the feature or fix

Fixes #5866

Fix a bug where there are hash collisions for cached gradients due to the key being computed using the memory address of a stack value. See the linked issue for more info.

Compute the key using the descriptor fields instead of the descriptor memory address.

I feel like there might be a way to compute the key such that horizontal and vertical gradients of the same size resolve to the same key? That way the same cached gradient map could be used to draw identical horizontal and vertical gradients because the [... cut]
closed 2024-04-06 by kisvegabor