lvgl/lvgl #5870: reverse mode for lv_arc

IB-AST, 2024-03-14:
### LVGL version

V8.3.11

### What happened?

Touch handling in reverse arc mode does not seem to work correctly.

Setting any value through code is okay, but when touching it shows odd behaviour.

Video taken from SLS, but behaviour is same on real display.

https://github.com/lvgl/lvgl/assets/148654555/fd0a70cd-0d05-4d4b-a916-04a314f3a125

 

### How to reproduce?

just use arc in reverse mode and touch

kisvegabor, 2024-03-20:
Hi,

It was already fixed in v9 and I've cherry-picked the fix to `release/v8.4` [here](https://github.com/lvgl/lvgl/commit/c512eccb7bba2ea5439bb1310356debfb86506b8).

IB-AST, 2024-03-22:
Thanks a lot for tackling this kisvegabor!

Flickering is fixed, but unfortunately it will not allow for full range touch now.

Touch always stops same max. angle at about 11o'Clock.

At arc range of 0-1400 (Bg start 90°/ Bg end 315°)  it allows a max. value of ~840 on touch.

Different range or start/end angle gets same results.

kisvegabor, 2024-03-22:
Please send a code snippet and a video to better understand the issue.

kisvegabor, 2024-03-25:
Thank you, but please also send a code snippet to create and initialize the arc. This way I can see if I could fix it. :slightly_smiling_face:

IB-AST, 2024-03-25:
This is genereated by SLS:

    ui_Instrument2 = lv_arc_create(ui_InstrumentPanel);
    lv_obj_set_width(ui_Instrument2, 200);
    lv_obj_set_height(ui_Instrument2, 200);
    lv_obj_set_x(ui_Instrument2, 110);
    lv_obj_set_y(ui_Instrument2, 15);
    lv_obj_set_align(ui_Instrument2, LV_ALIGN_CENTER);
    lv_obj_add_flag(ui_Instrument2, LV_OBJ_FLAG_OVERFLOW_VISIBLE);     /// Flags
    lv_obj_clear_flag(ui_Instrument2, LV_OBJ_FLAG_CLICKABLE | LV_OBJ_FLAG_PRESS_LOCK | LV_OBJ_FLAG_CLICK_FOCUSABLE |
                      LV_OBJ_FLAG_GESTURE_BUBBLE | LV_OBJ_FLAG_SNAPPABLE | LV_OBJ_FLAG_SCROLLABLE | LV_OBJ_FLAG_SCROLL_ELASTIC |
                      LV_OBJ_FLAG_SCROLL_MOMENTUM | LV_OBJ_FLAG_SCROLL_CHAIN);     /// Flags
    lv_arc_set_range(ui_Instrument2, 0, 1400);
    lv_arc_set_value(ui_Instrument2, 300);
    lv_arc_set_bg_angles(ui_Instrument2, 90, 315);
    lv_arc_set_mode(ui_Instrument2, LV_ARC_MODE_REVERSE);
    lv_arc_set_rotation(ui_Instrument2, 45);
    lv_obj_set_style_arc_color(ui_Instrument2, lv_color_hex(0x313031), LV_PART_MAIN | LV_STATE_DEFAULT);
    lv_obj_set_style_arc_opa(ui_Instrument2, 255, LV_PART_MAIN | LV_STATE_DEFAULT);
    lv_obj_set_style_arc_width(ui_Instrument2, 20, LV_PART_MAIN | LV_STATE_DEFAULT);
    lv_obj_set_style_arc_rounded(ui_Instrument2, false, LV_PART_MAIN | LV_STATE_DEFAULT);

    ui_object_set_themeable_style_property(ui_Instrument2, LV_PART_INDICATOR | LV_STATE_DEFAULT, LV_STYLE_ARC_COLOR,
                                           _ui_theme_color_AccentColor);
    ui_object_set_themeable_style_property(ui_Instrument2, LV_PART_INDICATOR | LV_STATE_DEFAULT, LV_STYLE_ARC_OPA,
                                           _ui_theme_alpha_AccentColor);
    lv_obj_set_style_arc_width(ui_Instrument2, 20, LV_PART_INDICATOR | LV_STATE_DEFAULT);
    lv_obj_set_style_arc_rounded(ui_Instrument2, false, LV_PART_INDICATOR | LV_STATE_DEFAULT);

    lv_obj_set_style_bg_color(ui_Instrument2, lv_color_hex(0x000000), LV_PART_KNOB | LV_STATE_DEFAULT);
    lv_obj_set_style_bg_opa(ui_Instrument2, 0, LV_PART_KNOB | LV_STATE_DEFAULT);

Project settings:
[image]

kisvegabor, 2024-04-03:
I've just tested this code, and it doesn't work in either v8 or v9.
 
@C47D could take look at it? This is a simplified code to reproduce the issue:

```c
lv_obj_t * ui_Instrument2 = lv_arc_create(lv_scr_act());
lv_obj_set_size(ui_Instrument2, 200, 200);
lv_obj_set_align(ui_Instrument2, LV_ALIGN_CENTER);
lv_arc_set_range(ui_Instrument2, 0, 1400);
lv_arc_set_value(ui_Instrument2, 300);
lv_arc_set_bg_angles(ui_Instrument2, 90, 315);
lv_arc_set_mode(ui_Instrument2, LV_ARC_MODE_REVERSE);
lv_arc_set_rotation(ui_Instrument2, 45);
```

IB-AST, 2024-04-28:
Not stale

kisvegabor, 2024-04-29:
I haven't hear from @C47D  from a while. I hope is okay, and just busy with something. 

@liamHowatt can you add it to your todo list after the API retirement?

liamHowatt, 2024-05-06:
PR opened! #6188

PR #6188 (2024-05-12): fix(arc): arc pressing bounds detection
### Description of the feature or fix

Fixes #5870

### 1

Fix an issue where the start and end angle of the arc are not processed correctly if the start angle is smaller than the end angle (and the end angle was <=360 when it was set with the setter).

This is the whole fix
```diff
diff --git a/src/widgets/arc/lv_arc.c b/src/widgets/arc/lv_arc.c
index 6d4383971..aa9349a56 100644
--- a/src/widgets/arc/lv_arc.c
+++ b/src/widgets/arc/lv_arc.c
@@ -937,12 +937,12 @@ static bool lv_arc_angle_within_bg_bounds(lv_obj_t * obj, const lv_value_precise
     lv_valu [... cut]
closed 2024-05-12 by FASTSHIFT