lvgl/lvgl #5875: Multiple definition of "emoji_F617"

1saeed, 2024-03-14:
### LVGL version

v9.0.0

### What happened?

emoji_F617 is defined in more than one source file:
`test_img_emoji_F617.c`
`lv_example_imgfont_1.c`
So, by calling` LV_IMAGE_DECLARE(emoji_F617); `you will encounter a "multiple definition" linker error.

C47D, 2024-03-15:
Hi,

It was fixed after the release, let me search the pull request so you can decide to patch your local repo or update lvgl.

1saeed, 2024-03-30:
It's fixed in v9.1 release

closed 2024-03-30 by 1saeed

vishvanathdoijad, 2024-08-06:
I am getting the same error in LVGL 9.1.1
multiple definations:
lvgl/examples/assets/emoji/img_emoji_F617.c
lvgl/tests/src/test_assets/img_emoji_F617.c

kisvegabor, 2024-08-07:
It was fixed by adding [this](https://github.com/lvgl/lvgl/blob/master/tests/src/test_assets/test_img_emoji_F617.c#L2).

vishvanathdoijad, 2024-08-08:
> It was fixed by adding [this](https://github.com/lvgl/lvgl/blob/master/tests/src/test_assets/test_img_emoji_F617.c#L2).

Thnx @kisvegabor 

Now I am able to compile LVGL 9.1.1 on Linux machine (VItis IDE)

My question is, for 90 rotation in LVGL 9.1.1 does we required to set the "sw_rotate" which is define in "lvgl/scr/display/lv_display_private.h"?
If Yes, Please tell me the steps to set that bit in main code.

kisvegabor, 2024-08-13:
[Here](https://docs.lvgl.io/master/porting/display.html#rotation) you can see how rotation works in v9.

vishvanathdoijad, 2024-08-13:
> [Here](https://docs.lvgl.io/master/porting/display.html#rotation) you can see how rotation works in v9.

Can you send me a simple example of sw rotation,
And how should I modify my_flush_cb() function using "lv_draw_sw_rotate" & "lv_display_rotate_area"

kisvegabor, 2024-08-15:
Take look at [this](https://github.com/lvgl/lvgl/blob/master/src/drivers/sdl/lv_sdl_window.c#L225-L257) as an example.

 > Is my understanding correct, "lv_display_set_rotation" means hardware rotation and "lv_draw_sw_rotate" & "lv_display_rotate_area" in software rotation.

You need to use `lv_display_set_rotation` and either
- make your hardware rotate the screen
- rotate the rendered image with software rotation