lvgl/lvgl #5877: Outstanding bugs! CAN’T set different panels to show different gradient colors!

Staok, 2024-03-15:
### LVGL version

8.3.11

### What happened?

LVGL 8.3.11
squareLine 1.4.0

use several grad to set different panel, but they result the same show
I put this bug issue to LVGL github repo:
https://github.com/lvgl/lvgl/issues/5866

but I try in squareLine result the same issue:
[image]

the running result is:
[image]

so this is a LVGL bug??

I urgently need to use this feature.
That different panels to be able to set different gradient colors

Staok, 2024-03-18:
@kisvegabor looking forward for reply, I waited for days, is this happened because the wrong usage I did or the LVGL bug, can it be fixed, waiting for help

closed 2024-03-18 by Staok

kisvegabor, 2024-03-18:
Hi,

I've replied in #5866.