lvgl/lvgl #5879: Color in text

xaowang96, 2024-03-15:
### Introduce the problem

In V8, we can use #color to set color of text, like this:
lv_label_set_text(label, "#FFFFF white ##FF0000 red #");
It is very convenient to be used to set different color in a text. But in v9, there is no this feature.

### Proposal

@kisvegabor Are there any plans to add this feature back?

W-Mai, 2024-03-19:
In V9, we can use lv_span to achieve this goal.

xaowang96, 2024-03-19:
> In V9, we can use lv_span to achieve this goal.

lv_span is not as easy to use as b. At the same time, we have many strings which contain color text like `#FFFFF white ##FF0000 red #` that are saved to files. In this case, if we use span to show these texts, we need to parse color in text, there is a lot of work to do.

kisvegabor, 2024-03-21:
In an issue and interesting idea came up: we could parsing a HTML like string by `lv_span` and create the spans automatically. E.g.
```c
lv_span_create_from_string(parent, "<color=#fff> white </> normal <span font=some_font>bold</>")
```

The syntax needs to be defined of course. 

I wouldn't like to add back the old recoloring-only solution as it's not so flexible.

kisvegabor, 2024-04-06:
Let's continue in https://github.com/lvgl/lvgl/issues/6019

closed 2024-04-06 by kisvegabor

nikunj8780, 2024-10-25:
in lvgl 9.2.1 using like this 
` lv_obj_clean(lv_scr_act());
  lv_obj_t *scr = lv_scr_act();
  lv_obj_set_style_bg_color(scr, lv_color_hex(0x000000), 0);
  lv_obj_set_style_bg_opa(scr, LV_OPA_COVER, 0);

  LV_FONT_DECLARE(over_load);
  label = lv_label_create(scr);
  lv_label_set_text_fmt(label, "OVERLOAD");
  lv_obj_align(label, LV_ALIGN_CENTER, 0, 0);
  lv_obj_set_style_text_font(label, &over_load, 0);
  lv_obj_set_style_text_color(label, lv_color_hex(0xCC0000), 0);`