lvgl/lvgl #5882: Check for create ONLY ONE object on the screen.

AleksandrBraun, 2024-03-15:
### Problem to solve

I can’t figure out how to detect the presence of an object on the screen and prevent the creation of the same object again.

### Success criteria

It would be nice to know for sure that an object already exists on the screen and not create a duplicate of it.

### Solution outline

I'm currently trying to track the deletion of an object when it is closed. But this doesn't always work.
After all, an object is not always deleted by its own closing event. Sometimes it gets deleted when moving to another screen, removing the previous screen and its children's milestones.

### Rabbit holes

I see a solution to check in the button click event handler for the presence of a previously created object and ignore the creation of its duplicate.

### Testing

In my design, clicking a button creates a panel with different settings.
But I would like the panel to “hang in the air” without blocking the entire lower screen.
But then there remains the possibility of accidentally missing and pressing the button again. Which will lead to the unwanted creation of another panel on top of the previous one.

[image]

### Teaching

I tried doing this in the handler, but it doesn't work.

```
void button_Press_event(  lv_event_t* e ){
    if (  ui_MessageBox != NULL ) return;
    createMessageBox( lv_screen_active() );
}
```
I think this is due to the fact that when the button is clicked, a new object is created with a new unique number and cannot be tracked by the object name.

PGNetHun, 2024-03-15:
hello!
what you show is basically a modal/popup window (in web terminology).
to prevent the buttons behind/below it to be pressed accidentally, add a gray overlay behind the modal.
1. create an obj for the whole screen (width, height:  100%): this will be the container
2. create another obj that will be the gray overlay, again from position 0,0 to full width and height, background color black, 50% opacity
3. create your modal in the container obj, center it, and you can add your buttons, tabs there.
4.

AleksandrBraun, 2024-03-15:
[PGNetHun](https://github.com/PGNetHun)
Thanks for the quick tip.
Yes, I am familiar with understanding layers in the web
And creating a shading layer is also an option.
But I think that at the level of checking the condition in the event handler code, there is a more elegant solution than adding a whole block that creates a one-time object with a bunch of styles and loads memory.

kisvegabor, 2024-03-16:
Usually I also create a global variable or something similar. But instead of a `bool` variable I have an `lv_obj_t *` variable. If it's set, the widget is there. On delete I set it to `NULL`. This way I have access to the widget too.   

[Here](https://github.com/lvgl/lvgl/issues/5722) we were talking about allowing to set names for the widgets. Once we have names, the widgets can be found by name too. I think it can help in your case as well.

AleksandrBraun, 2024-03-16:
[kisvegabor](https://github.com/kisvegabor)
Thanks for the info.
Yes, I found this discussion recently. I also saw [this article](https://docs.lvgl.io/master/others/obj_id.html#obj-id) from the documentation. 
But I didn’t understand how it would be useful to me.
The additional ID generation method adds another field to the lv_obj_t object structure. But how is this fundamentally different from simply storing a reference to an existing object in a variable like lv_obj_t*? As you described above.
I tested calling the lv_obj_stringify_id method to get the current object ID and it simply outputs the string "objX" where X is the increasing value. While a simple request for an object by its name during creation gives a 32-bit integer, which is essentially also unique on the current screen.
```
ui_READY_Panel_soundPanel = lv_obj_create( target );
char buf_ID[ 32 ];
lv_obj_stringify_id( ui_READY_Panel_soundPanel, buf_ID, 32 );
DPRINTF( "Obj ID:\t%s\r\n", buf_ID );
DPRINTF( "Obj name ID:\t%d\r\n", ui_READY_Panel_soundPanel );
```
result:
[image]

Maybe I misunderstood something, but how, without intermediately storing a reference to an existing object, can I understand that it exists on the screen?
It seemed to me that being able to assign a custom ID to an object, such as `lv_obj_set_id( lv_obj_t* obj, char* custom_id),` (as [XuNeo](https://github.com/lvgl/lvgl/issues/5722#issuecomment-1963459310) suggested) similar to javascript, would help the user access the object directly by its ID.

kisvegabor, 2024-03-21:
> The additional ID generation method adds another field to the lv_obj_t object structure. But how is this fundamentally different from simply storing a reference to an existing object in a variable like lv_obj_t*? As you described above.

The main difference is that you don't need to store and maintain a global variable. You know that you named the widget as "my_popup". So by using `lv_obj_get_by_name("my_popup")` you can get widget. If it exists it will be returned. If not you get `NULL`.

AleksandrBraun, 2024-03-26:
> You know that you named the widget as "my_popup". So by using `lv_obj_get_by_name("my_popup")` you can get widget.

I'm using version 9 and didn't find this method `lv_obj_get_by_name`
Where do you get it from?

kisvegabor, 2024-03-28:
`lv_obj_get_by_name` is not implemented yet. It could be a new feature.

closed 2024-04-28 by lvgl-bot