lvgl/lvgl #5889: bar new value set issue

1saeed, 2024-03-15:
### LVGL version

v9.0.0

### What happened?

If you create a simple bar and set the value it will be rendered and flushed as expected, but when you change the value again something strange happens and it got disappeared.

### How to reproduce?

	lv_obj_t * bar = lv_bar_create(lv_screen_active());
	lv_obj_set_size(bar, 200, 20);
	lv_obj_center(bar);

	lv_bar_set_value(bar, 10, LV_ANIM_OFF);
	lv_refr_now(NULL);

	lv_bar_set_value(bar, 50, LV_ANIM_OFF);
	lv_refr_now(NULL);

	while(1){};

kisvegabor, 2024-03-19:
Hi,

can you send a photo about the corrupted screen?

1saeed, 2024-03-20:
[image]

[image]

liamHowatt, 2024-03-23:
I was not able to reproduce with the mono theme or otherwise. Are you able to send more code to reproduce, or try to reproduce it in the simulator?

kisvegabor, 2024-03-25:
A few questions:
1. How do you initialize your `lv_display_t`? (flush_cb, buffers)
2. Please share your flush_cb 
3. I suggest updating to v9.1

1saeed, 2024-03-25:
@liamHowatt @kisvegabor. Thanks for your replies. I've updated to 9.1 and this time, there's no more corrupted object. The progress bar shows the first set successfully after the first `lv_refr_now(NULL);` call but in the second refresh call it gets stuck in `refr_sync_areas();` and tracing inside that function, it gets stuck in `lv_draw_buf_copy(off_screen, sync_area, on_screen, sync_area);`. Finally, in `lv_draw_buf_copy` it gets asserted by this:

`    /*Source and dest color format must be same. Color conversion is not supported yet.*/
    LV_ASSERT_FORMAT_MSG(dest->header.cf == src->header.cf, "Color format mismatch: %d != %d",
                         dest->header.cf, src->header.cf);`
                         
dest->header.cf is 17 and src->header.cf is 16

I set the color format as `LV_COLOR_FORMAT_ARGB8888` which is I believe is 16 but it expects `LV_COLOR_FORMAT_XRGB8888` which is 17. In the `lv_conf.h` I have `#define LV_COLOR_DEPTH 32` the same as I had in v8.3. From the comment `/*Color depth: 8 (A8), 16 (RGB565), 24 (RGB888), 32 (XRGB8888)*/` I realised that there's no more ARGB888 format exist. 

Here's how the initialization look like:

`static lv_disp_t *display = NULL;

void LCD_init() {

   display = lv_display_create(LCD_WIDTH, LCD_HEIGHT);
   lv_display_set_buffers(display, (void*)lv1_frame_buff, (void*)lv2_frame_buff, LCD_WIDTH * LCD_HEIGHT * BPP_FACTOR, 
   LV_DISPLAY_RENDER_MODE_DIRECT);
   lv_display_set_color_format(display, LV_COLOR_FORMAT_ARGB8888);
   lv_display_set_flush_cb(display, disp_flush);
}

static void disp_flush(lv_display_t * disp, const lv_area_t * area, uint8_t * px_map)
{
		/*Return if the area is out the screen*/
		if (area->x2 < 0)
			return;
		if (area->y2 < 0)
			return;
		if (area->x1 > LCD_WIDTH - 1)
			return;
		if (area->y1 > LCD_HEIGHT - 1)
			return;

		uint32_t address = LCD_BUFFER_ADDR_AT_SRAM + (((LCD_WIDTH * area->y1) + area->x1) * BPP_FACTOR);

		CopyImageToLcdFrameBuffer((void*) px_map, (void*) address, lv_area_get_width(area), lv_area_get_height(area));

		return;
}

static void CopyImageToLcdFrameBuffer(void *pSrc, void *pDst, uint32_t xSize, uint32_t ySize)
{
	for(uint32_t y = 0 ; y < ySize ; y++)
	{
		lv_memcpy(pDst, pSrc, xSize * BPP_FACTOR);
		pDst += LCD_WIDTH * BPP_FACTOR;
		pSrc += xSize * BPP_FACTOR;
	}
	lv_disp_flush_ready(display);
}
`
All codes run smoothly in v8.3 but we all have to migrate to v9 eventually. Except for the CHANGELOG I couldn't find any documentation to fully address this migration. So, I would be grateful for any help on this very first step to test v9 out.

kisvegabor, 2024-03-26:
Ah, I think it's a bug in LVGL. Please try updating `lv_display_set_color_format` (in `lv_display.c`) like this:
```c
void lv_display_set_color_format(lv_display_t * disp, lv_color_format_t color_format)
{
    if(disp == NULL) disp = lv_display_get_default();
    if(disp == NULL) return;

    disp->color_format = color_format;
    disp->layer_head->color_format = color_format;
    disp->buf_1->header.cf = color_format;                 //ADDED
    if(disp->buf_2) disp->buf_2->header.cf = color_format;    //ADDED

    lv_display_send_event(disp, LV_EVENT_COLOR_FORMAT_CHANGED, NULL);
}
```

1saeed, 2024-03-26:
I added those lines and it doesn't get asserted anymore. However, the render is corrupted in direct mode. If I set either `LV_DISPLAY_RENDER_MODE_PARTIAL` or `LV_DISPLAY_RENDER_MODE_FULL` instead of `LV_DISPLAY_RENDER_MODE_DIRECT`, it works as expected.

[image]

kisvegabor, 2024-03-27:
Great, thank you for the feedback! I've opened #5973 to fix it.

PR #5973 (2024-04-11): fix(display): update the color format of the draw buffers on color format change
### Description of the feature or fix

fixes #5889

@XuNeo 
We [reshape](https://github.com/lvgl/lvgl/blob/master/src/core/lv_refr.c#L578) the draw buffer before rendering, so I think the changes are ok, but please confirm it.

### Notes
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.
- Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant.
- Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if applicable.
- If you added new options [... cut]
closed 2024-04-11 by FASTSHIFT