lvgl/lvgl #5891: encoder indev

kdschlosser, 2024-03-16:
### Introduce the problem

It appears there is no way to set the direction of an encoder. whether you want to move along the x axis or the right axis. For purposes of a mouse wheel it would be scrolling up and down (y axis) or scrolling left and right (x axis). It is assumed only a single axis.

### Proposal

IDK if there is a way to alter this defined behavior but I would suggest adding a new enumeration to set the direction.

kisvegabor, 2024-03-16:
Encoders are just jumping to the next item in the group and usually they are focused automatically. So there is no such a thing as direction in case of the encoders. Only the order of widget in the group matters.

kdschlosser, 2024-03-16:
An encoder is something like a mouse wheel yes? wheel a mouse wheel has 2 axis, one that controls the vertical scrolling and the other that controls the horizontal. So how does one go about doing this using the current design?

C47D, 2024-03-16:
I've never seen a mouse wheel that goes in 2 directions. It only goes forward and backwards, do you mean the click you can do on the mouse wheel?

kdschlosser, 2024-03-17:
Try pushing your wheel side to side. You will be in for a really neat surprise

kisvegabor, 2024-03-21:
The encoder is not for scrolling. It hardware like this
[image]

You can use it on an instrument where you don't have touch. The mousewheel is just to emulate the emcoder.

However we really have to add support for real scrolling but it's an other story.

kdschlosser, 2024-03-22:
An encoder is exactly what a mouse wheel is it just instructs direction instead a hardware set number for its position. This is because the encoder doesn't have a stop and start point. it is able to rotate an indefinite amount. There are many rotary encoders that function in this manner, take a volume knob on an audio video receiver for example. A lot of them you can turn and turn and turn. This is done by using direction only and not number of steps taken. That doesn't change what the hardware is, which is a rotary encoder. I now there is a way to collect the object that has focus, what I do not know how to do is instruct an object how to scroll. 

There is an enumeration for telling a gradient to render horizontally or vertically

```c
enum _lv_grad_dir_t {
    LV_GRAD_DIR_NONE, /**< No gradient (the `grad_color` property is ignored)*/
    LV_GRAD_DIR_VER,  /**< Vertical (top to bottom) gradient*/
    LV_GRAD_DIR_HOR,  /**< Horizontal (left to right) gradient*/
};
```

This enumeration could be changed to

```c
enum _lv_axis_t {
    LV_AXIS_NONE,
    LV_AXIS_VERTICAL,
    LV_AXIS_HORIZONTAL
};

typedef uint8_t lv_axis_t;
```

so not the enumeration could be used for the gradient and also for an added function 

```c
lv_obj_scroll(lv_obj_t *obj, lv_axis_t axis, int16_t steps);
```

Use the same style Cartesian coordinate system that is used for computer monitors where left is negative and right is positive and up is negative and down is positive. The steps parameter is supplied by the user or in this case it would be handled inside of the indev callback. By setting the `continue_reading` flag in the `lv_indev_data_t` structure would cause lvgl to redraw the display and then call the callback function again. If there is another change then once again the `continue_reading flag would get set to `true` If there is no change then the `continue_reading` flag would get set to `false` which would then allow the indev processing to return so normal operation would continue.

I believe that is the purpose of what the `continue_reading` flag was intended to do but it doesn't work. It seems like most of the mechanics are in place to make this work and I do not believe that a new indev type needs to be added to lvgl for the purpose of handling the scrolling. there only needs to be a function added that can be called from inside the indev callback function and that continue_reading flag needs to be fixed so it will update the display as needed and then call the indev callback again.

another way this can also be handled is to change the callback function so it return a boolean instead of having the continue_reading flag inside of the structure. That return value can be used to instruct the processing function if it should loop and update the display between each loop.

This is actually a simple thing to do. It looks like there has been some work that has been started in v9.0 with the added indev_scroll functions.

kdschlosser, 2024-03-22:
I noticed there is something done that is kind of to this effect in the `indev_encoder_proc` function

if the indev type is set to encoder and the `lv_indev_data_t` structure gets populated using `LV_KEY_LEFT` and `LV_KEY_RIGHT` using the `key` field in the data structure. doing that seems to do something with groups.

kdschlosser, 2024-03-22:
The loop is still in there for the continue_reading field but it doesn't appear that the display gets updated at all when each loop is made.

kisvegabor, 2024-03-25:
It's okay that the wheel of the mouse is an encoder too. 

If you use it as an LVGL encoder indev it will work with groups to emulate an encoder for a laboratory power supply you might developing. 

However, it's a completely different story if you are developing a desktop app and want to use the mouse wheel to scroll the hovered container. I agree that it's important, but it's not implemented yet.

kdschlosser, 2024-03-25:
It shouldn't be that hard to implement in the actual indev callback function. We are able to get the window that is being hovered over by performing a hit test.

`
void lv_obj_scroll_by(lv_obj_t * obj, int32_t x, int32_t y, lv_anim_enable_t anim_en);
`

can be used once the hovered object is collected using a hit test. there would need be a recursive parent lookup to determine if the scroll bars are shown. If  the needed bar is not shown then the parent would be collected and tested. It would work all the ay down the tree until the parent was NULL to locate what can be scrolled.

What is going to take time is doing a recursive iteration from the bottom of the tree all the way up performing a hit test on each object to see what is the top most item that is good with the hit_test. The only thing IDK how to do is to check if there is anything that is covering that item or not. 

I have to look into what `lv_obj_t * lv_indev_get_scroll_obj(const lv_indev_t * indev);` does.

I am a little bit confused because you has said that scrolling is not implemented but then I see this function above and this function as well `lv_dir_t lv_indev_get_scroll_dir(const lv_indev_t * indev);`.. why would there be no scrolling support and those functions exist?

kdschlosser, 2024-03-25:
This is what I have done up for SDL in python code.

```python
from micropython import const
import pointer_framework
import lvgl as lv  # NOQA

SDL_BUTTON_LEFT = const(0x01)
SDL_BUTTON_MIDDLE = const(0x02)
SDL_BUTTON_RIGHT = const(0x03)
SDL_BUTTON_X1 = const(0x04)
SDL_BUTTON_X2 = const(0x05)
SDL_BUTTON_LMASK = const(0x01)
SDL_BUTTON_MMASK = const(0x02)
SDL_BUTTON_RMASK = const(0x04)
SDL_BUTTON_X1MASK = const(0x08)
SDL_BUTTON_X2MASK = const(0x10)

class SDLMouse(pointer_framework.PointerDriver):
    def __init__(self):
        super().__init__(None)

        self.__current_state = self.RELEASED
        self.__x = -1
        self.__y = -1
        self.__wheel_x = 0
        self.__wheel_y = 0
        self.__scroll_obj = None
        self.__button_state = self.RELEASED

        self._py_disp_drv._data_bus.register_mouse_button_callback(self._button_cb)  # NOQA
        self._py_disp_drv._data_bus.register_mouse_wheel_callback(self._wheel_cb)  # NOQA
        self._py_disp_drv._data_bus.register_mouse_motion_callback(self._motion_cb)  # NOQA

    def _get_coords(self):
        if self.__scroll_obj is not None:
            self.__scroll_obj.scroll_by_bounded(self.__wheel_x, self.__wheel_y, 1)  # NOQA
            self.__scroll_obj = None
            return None

        return self.__button_state, self.__x, self.__y

    def _button_cb(self, kwargs):
        button = kwargs['button']

        if button | SDL_BUTTON_LMASK == button:
            if kwargs['state']:
                self.__button_state = self.PRESSED
            else:
                self.__button_state = self.RELEASED

            self.__x = kwargs['x']
            self.__y = kwargs['y']

    def _motion_cb(self, kwargs):
        self.__x = kwargs['x']
        self.__y = kwargs['y']

        state = kwargs['state']

        if state | SDL_BUTTON_LMASK == state:
            self.__button_state = self.PRESSED
        else:
            self.__button_state = self.RELEASED

    def _wheel_cb(self, kwargs):
        x = kwargs['x']
        y = kwargs['y']

        point = lv.point_t({'x': kwargs['mouseX'], 'y': kwargs['mousey']})

        obj = lv.indev_search_obj(self._disp_drv.get_layer_sys(), point)
        if not obj:
            obj = lv.indev_search_obj(self._disp_drv.get_layer_top(), point)
        if not obj:
            obj = lv.indev_search_obj(self._disp_drv.get_screen_active(), point)
        if not obj:
            obj = lv.indev_search_obj(self._disp_drv.get_layer_bottom(), point)

        if obj:
            dir_flags = obj.get_scroll_dir()

            if x and dir_flags | lv.DIR_HOR == dir_flags:
                self.__wheel_x = x
            else:
                self.__wheel_x = 0

            if y and dir_flags | lv.DIR_VER == dir_flags:
                self.__wheel_y = y
            else:
                self.__wheel_y = 0

            if self.__wheel_x or self.__wheel_y:
                self.__scroll_obj = obj
            else:
                self.__scroll_obj = None
        else:
            self.__wheel_x = 0
            self.__wheel_y = 0
            self.__scroll_obj = None
```

closed 2024-04-16 by lvgl-bot