lvgl/lvgl #5899: Text processing methods

sparkles43, 2024-03-18:
### Introduce the problem

Hi,

I am currently trying to do some UTF-8 text processing in my application. Due to the fact that I only require a minimal set of functions to process UTF-8 character I don't want to pull in another library but rather use the text processing capabilities lvgl offers on its own.

I have found the file [lv_text.h](https://github.com/lvgl/lvgl/blob/master/src/misc/lv_text.h#L259) (and [lv_txt.h](https://github.com/lvgl/lvgl/blob/release/v8.3/src/misc/lv_txt.h#L187) respectively) to contain global function pointers to useful text processing functions. Unfortunately all function pointers are labeled as private (prefixed with `_`) but are widely used in lvgl.

Is there any obstacle to making these function pointers part of the official API by removing the prefix? (I could imagine that it could increase maintenance efforts :thinking: )

Best regards,
sparkles

### Proposal

- Make function pointers part of the official API by removing the prefix `_` (for both lvgl v8 and lvgl v9)

kisvegabor, 2024-03-18:
The original purpose of adding the `_` was to indicate that a function is not part of the public API and to lower the size of the MicroPython binding. In v9 we started to establish the patter on having private headers instead (e.g. `lv_display_private.h`).

I think we can slowly transition to having private header files instead of `_` functions. 

cc @liamHowatt Could you open a PR for adding  `lv_text_private.h`? Note that `lv_text_private.h` shouldn't be included in `lvgl.h`, but only in the C files where needed.

sparkles43, 2024-03-19:
> The original purpose of adding the _ was to indicate that a function is not part of the public API ...

I guess my understanding was correct then.

> I think we can slowly transition to having private header files instead of _ functions.

Personally, I think private header files are a good way of preventing (more or less) accidental use of non-public APIs. :)

Regarding the original idea, can you give some feedback on making text processing part of the public API? I think it would be reasonable to handle strings in my UI application the same way lvgl does, i.e. have a common understanding of the text representations.

My initial proposal (to just drop the `_` prefix) is not a very good idea. The files should obviously be refactored to expose functions.

Anyways, I wonder why the text processing functions were exposed as extern function pointers. Was that to reduce calling overheads or was it rather used as a convenient way to switch between ASCII functions and UTF-8 functions in the lv_txt.c?