lvgl/lvgl #5902: Missing function to bind switch state (lv_switch_bind_checked)

xamix, 2024-03-18:
### LVGL version

v9.0.0

### What happened?

I can find the `lv_button_bind_checked` for buttons but not the equivalent for switch `lv_switch_bind_checked`, is this function missing or do I need to use another function?
I tryed `lv_button_bind_checked` on a switch and it seem to work...

Regards

kisvegabor, 2024-03-18:
Actually `lv_button_bind_checked` should work for buttons, switches and checkboxes too. It just binds the `LV_STATE_CHECKED` state of the widget. I think we should rather call this function `lv_obj_bind_checked`. What do you think?

cc @liamHowatt

liamHowatt, 2024-03-18:
Switch and checkbox are checkable by default. `lv_obj_add_flag(btn, LV_OBJ_FLAG_CHECKABLE)` is needed before `lv_button_bind_checked` on a button. Meanwhile `lv_button_bind_checked` works "out of the box" on checkboxes. Since there's nothing about the implementation of `lv_button_bind_checked` that relies on anything button-specific, and it's even suited to some other widgets even better than button, yes I think it's a good idea to rename it to `lv_obj_bind_checked`.

kisvegabor, 2024-03-21:
@liamHowatt It seems all good. Could you send a PR?

kisvegabor, 2024-04-05:
The Pull request is merged. Closing this issue.

closed 2024-04-05 by kisvegabor