lvgl/lvgl #5904: Add CI test for Windows build

kisvegabor, 2024-03-18:
### Introduce the problem

We have some warnings shown only [MSVC](https://github.com/lvgl/lvgl/pull/5873/files).

### Proposal

It's would be nice to set up a Window CI as well.

kisvegabor, 2024-04-02:
Not stale

liamHowatt, 2024-04-03:
I have opened a PR for building on Windows with GCC which supports `windows.h` and contains exceptions to the standard such as `_aligned_malloc` found in MSVC. #6015. This can be a starting point for using MSVC. It may be simple to use MSVC in CMake, but I do not know for certain, and wanted to establish some progress in the right direction in case MSVC is not trivial to use in CMake. `cl.exe` is the compiler that'd be invoked at each build step.

Since we just want to build for now, it might be simpler to maintain a build script (?) separate from CMake for MSVC.

kisvegabor, 2024-04-03:
Great! Let's continue in the PR! :slightly_smiling_face:

closed 2024-04-03 by kisvegabor